espressif/esp-matter #987: Enable to add esp32c6 to Google home

mukeshbauskar, 2024-06-28:
**Describe the bug**
While adding esp32c6 to Google as matter device I am getting error saying
Unable to add Accessory( You may need to restart your accessory before you can add it to your home)

Both devices are on same 2.4GHz network.
I tried enabling IGMP on AP.

**Environment**
- ESP-Matter Commit Id: lastest
- ESP-IDF Commit Id: main
- SoC : ESP32C6
- Device Logs (Please attach the log file):
- Host Machine OS: iPad OS
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

jonsmirl, 2024-06-28:
My first guess would be something is wrong with your certificates. This is what I use for testing....
~~~
esp-matter-mfg-tool -cn "Test" -v 0xFFF2 -p 0x8001 --pai \
-k $ESP_MATTER_PATH/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
-c $ESP_MATTER_PATH/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
-cd $ESP_MATTER_PATH/connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der \
--vendor-name "Lowpan" --product-name "Dimmer" --serial-num "12345678" \
--hw-ver-str "1.0.0" \
--passcode 20202021 --discriminator 3840 
~~~

mukeshbauskar, 2024-06-28:
@jonsmirl Sorry for my limited knowledge. I am very new to this. 
Should I run this before flashing or after flashing?

jonsmirl, 2024-06-28:
When the Espressif people wake up they should be able to point you at detailed instructions. 

If you compile and run the demos without modifying them, they should work.

mukeshbauskar, 2024-06-28:
@jonsmirl I really appreciate you trying to help.

dhrishi, 2024-06-28:
>executed wifi esp wifi connect

@mukeshbauskar This isn't required. Is there a specific reason why this is done? Ideally, the commissioning process involves onboarding the device to the Wi-Fi (IP) network too.

mukeshbauskar, 2024-06-28:
@dhrishi 

I followed following steps

1. Erased flash
2. Flashed
3. chip-tool interactive start
4. pairing code-wifi 0x7283 <wifi_ssid> <wifi_password> 34970112332
5. Google home
6. used 34970112332 as Id and tried joining

Still no luck

dhrishi, 2024-06-28:
@mukeshbauskar If you are using chip-tool and did not connect the device to Wi-Fi, you should use `pairing ble-wifi` command.

mukeshbauskar, 2024-06-28:
@dhrishi I tried that. [Here] (https://github.com/user-attachments/files/16037537/matter.txt)
 is the log from that. It still did not work.

jonsmirl, 2024-06-29:
> @dhrishi
> 
> I followed following steps
> 
> 1. Erased flash
> 2. Flashed
> 3. chip-tool interactive start
> 4. pairing code-wifi 0x7283 <wifi_ssid> <wifi_password> 34970112332 ([Here](https://github.com/user-attachments/files/16037467/matter.txt) logs)

You need to open a commissioning window from chip-tool and get the pairing code.

> 5. Google home

 6. used <the pairing code from the above command> as Id and tried joining

> 
> Still no luck

mukeshbauskar, 2024-06-29:
@jonsmirl can you please let me know command to get the pairing code. I used default one (34970112332) to pair.

mukeshbauskar, 2024-06-29:
@jonsmirl @dhrishi I am using mac os to flash the dev kit. I found that before I do anything I need to install the developer profile for chip tool to work. So I cleaned my repo and installed everything again. Then tried pairing using
chip-tool pairing ble-wifi 0x7283 <wifi-ssid> <wifi-password> 20202021 3840

The I got attached output. 
[matter.txt](https://github.com/user-attachments/files/16039686/matter.txt)

dhrishi, 2024-07-01:
@mukeshbauskar Can you please also upload the device side logs? The chip-tool logs indicate that it was unable to find the device on the operational (IP) network. Can you verify if you can ping the device from the host machine running the chip-tool. Additionally, you can also find the device using `dns-sd` for `_matter._tcp`

mukeshbauskar, 2024-07-01:
@dhrishi Here is the link to complete output of building and flashing. 
[matter.txt](https://github.com/user-attachments/files/16048710/matter.txt)

If you are looking for something, please let me know how to generate those logs.

mukeshbauskar, 2024-07-01:
@dhrishi Can we please have conference call to go over the problem, if that helps. Please let me know your availability.

dhrishi, 2024-07-01:
Please share the device (ESP32) side console logs. Also, did you get a chance to try this out -

>Can you verify if you can ping the device from the host machine running the chip-tool. Additionally, you can also find the device using dns-sd for _matter._tcp

mukeshbauskar, 2024-07-01:
@dhrishi Can you please let me how to get those logs.

Also I tried pinging it using (ping <IP address>) but did not get any response. But on the router side I could see that the device is connected and IP address was assigned.

jonsmirl, 2024-07-01:
idf.py monitor

mukeshbauskar, 2024-07-01:
@jonsmirl Thanks. I should monitor while I am trying to add it to Google Home, right??

jonsmirl, 2024-07-01:
yes, always use that command. That command will keep the log in the file matter.log and also show it on the screen.

jonsmirl, 2024-07-01:
I am on Linux, you are on a Mac.  You need to use the correct port from your system.

mukeshbauskar, 2024-07-02:
Here is the output for the command.

[matter.txt](https://github.com/user-attachments/files/16060154/matter.txt)

jonsmirl, 2024-07-02:
It is waiting for a BLE connection. Does not show you trying to connect to it in the log.

Run the Apple Home or Google Home app and scan this QR code.
https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning-and-control

dhrishi, 2024-07-02:
@mukeshbauskar The ESP32 console log here does not correspond to the chip-tool side logs you've been sharing before. As @jonsmirl mentioned the current log shows that the ESP32 is not being discovered by the commissioner phone app or chip-tool. Please use the QR code / manual code from the link above.

mukeshbauskar, 2024-07-02:
@dhrishi @jonsmirl Here are the logs for build, flash, chip-tool and Google Home connect.

[matter.txt](https://github.com/user-attachments/files/16061011/matter.txt)

jonsmirl, 2024-07-02:
Your device's IP is 192.168.50.172. Is it on the same network as the Google Home device? I don't see the Google Home device connecting to it.

mukeshbauskar, 2024-07-02:
@jonsmirl  Yes, it is on the same network. Steps I take on Google Home side.

1. Launch google home app
2. Go to Devices
3. click on Add
4. Matter-enabled device
5. Scan the QR code provide by previous link
6. Agree to terms and condition
7. On popup "Accessory" click on 'Add to "Google Home"'
8. Wait for "Accessory" Connection...
9. Error Unable to Add Accessory, Pairing Failed.

jonsmirl, 2024-07-02:
Here's a copy of my log with a successful pair of my device. My device is more complex so ignore all of the extra endpoints when comparing.

[matter.log](https://github.com/user-attachments/files/16065882/matter.log)

mukeshbauskar, 2024-07-02:
@jonsmirl @dhrishi [Here](https://github.com/user-attachments/files/16069729/matter.txt)
 is the logs which shows that Google tried to connect to the device.

The previous problem occurred because I was using iPad to add device. If I use android it tries to ping the device.

VaishaliAvhale, 2024-07-02:
Hi @mukeshbauskar,

Which hub are you using? Nest Hub 2nd generation, Nest Hub Max, etc.?

mukeshbauskar, 2024-07-02:
> Hi @mukeshbauskar,
> 
> Which hub are you using? Nest Hub 2nd generation, Nest Hub Max, etc.?

I am using Nest mini.

VaishaliAvhale, 2024-07-02:
Okay and you are using latest Google Home App right?
Just once try this and share me the logs
1. idf.py erase-flash flash monitor
2. Commission device by scanning QR code: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/_images/matter_qrcode_20202021_3840.png. Commssioning will start... follow the screen instruction like selecting room, giving name to the device and adding it into the room.
If you still face an issue share complete log file will look into it and get back to you.
Just I have build, flashed and commissioned latest esp-matter light used esp32c6 chip and it is working at my end. I'm also using Nest Hub 2nd Generation[Nest Hub mini]

mukeshbauskar, 2024-07-02:
@VaishaliAvhale 

I am doing exactly what you mentioned. And I have attached logs with previously. [Here](https://github.com/user-attachments/files/16069729/matter.txt) are the those logs.

VaishaliAvhale, 2024-07-02:
@mukeshbauskar sorry it's my bad here is the logs:
[esp-matter-light-esp32c6-dut.log](https://github.com/user-attachments/files/16072999/esp-matter-light-esp32c6-dut.log)

mukeshbauskar, 2024-07-02:
@VaishaliAvhale Can you please let me know the exact commands you execute on your system and steps your perform to pair it with Google home.

VaishaliAvhale, 2024-07-02:
1.path-to-esp-matter/example/light$ rm -rf build sdkconfig sdkconfig.old managed_components/ dependencies.lock   ---clean build light example
2. idf.py set-target esp32c6
3. idf.py build 
4. idf.py erase-flash flash monitor
5. scanning QR code using google Home App[after clicking on Add Matter  device will get QR scanning option]: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/_images/matter_qrcode_20202021_3840.png
Immediately commissioning started....once commissioning complete will get action [Done button on Google Home App click on that then select Room then give name to the device] 
** All set you can see device once above step complete with success

mukeshbauskar, 2024-07-02:
@VaishaliAvhale how are you connecting the esp32c6 to the wifi network. I thought I am suppose to use following command for that.
chip-tool pairing ble-wifi 0x7283 <ssid> <passphrase> 20202021 3840

dhrishi, 2024-07-02:
@mukeshbauskar You either use the host-based chip-tool or Google Home setup. The Matter commissioning process involves providing IP (Wi-Fi) connectivity to the ESP32-C6, so you do not need to do it explicitly.

jonsmirl, 2024-07-02:
Explain fabrics, he may not know the ecosystems can't see each other.

mukeshbauskar, 2024-07-02:
> @mukeshbauskar You either use the host-based chip-tool or Google Home setup. The Matter commissioning process involves providing IP (Wi-Fi) connectivity to the ESP32-C6, so you do not need to do it explicitly.

And to communicate the wifi details to esp32c6, the commission device needs blue tooth connectivity with esp32c6, right?

dhrishi, 2024-07-02:
>And to communicate the wifi details to esp32c6, the commission device needs blue tooth connectivity with esp32c6, right?

That's right. But it is all a part of the Matter protocol. You just need to start the process by scanning the QR code through Google Home app and the rest should just happen.
So, if you just follow the steps [here](https://github.com/espressif/esp-matter/issues/987#issuecomment-2204094146), you should be able to commission the ESP32-C6

mukeshbauskar, 2024-07-02:
> Explain fabrics, he may not know the ecosystems can't see each other.

@jonsmirl I might be diverging here. I have Nest mini and esp32c6. And this is my first matter device. Now when I am trying to commission esp32c6, I am adding a node to my matter fabric which does not have nodes yet.
Assume I am successful in adding esp32c6 then will the fabric contain two nodes one for Nest mini and another esp32c6? And when I am not home  and if I try to manipulate the device using google  home app on my phone, it will talk to nest mini which will communicate with esp32c6, right??

mukeshbauskar, 2024-07-02:
> That's right. But it is all a part of the Matter protocol. You just need to start the process by scanning the QR code through Google Home app and the rest should just happen. So, if you just follow [...]

@dhrishi I will cleanup my system and try this again...

mukeshbauskar, 2024-07-03:
@jonsmirl @dhrishi @VaishaliAvhale I found the issue. The issue was that I was connecting to a wifi network which was configure to not to have access to local intranet.

closed 2024-07-03 by mukeshbauskar