espressif/esp-matter #988: Unable to build project - Error : unable to load pigweed_environment.gni

Aryan-Tamboli, 2024-06-28:
**Describe the bug**
A clear and concise description of what the bug is.

**Environment**
- ESP-Matter Commit Id:
- ESP-IDF Commit Id:
- SoC (eg: ESP32 or ESP32-C3):
- Device Logs (Please attach the log file):
- Host Machine OS:
- Commissioner app and versions if present:
- Commissioner's logs if present:

Error Logs 
Performing configure step for 'chip_gn'FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure /home/aryan/esp-matter/examples/light/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure
cd /home/aryan/esp-matter/examples/light/build/esp-idf/chip && /usr/local/bin/gn --root=/home/aryan/esp-matter/connectedhomeip/connectedhomeip/config/esp32 gen --check --fail-on-unused-args /home/aryan/esp-matter/examples/light/build/esp-idf/chip && /usr/bin/cmake -E touch /home/aryan/esp-matter/examples/light/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure
ERROR at //build_overrides/pigweed_environment.gni:20:1: Unable to load "/home/aryan/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/build_overrides/pigweed_environment.gni".
import("${_bootstrap_root}/build_overrides/pigweed_environment.gni")
^------------------------------------------------------------------
See //third_party/connectedhomeip/third_party/pigweed/repo/pw_protobuf_compiler/proto.gni:16:1: whence it was imported.
import("//build_overrides/pigweed_environment.gni")
^-------------------------------------------------
See //third_party/connectedhomeip/third_party/pigweed/repo/pw_log/BUILD.gn:22:1: whence it was imported.
import("$dir_pw_protobuf_compiler/proto.gni")
^-------------------------------------------
See //third_party/connectedhomeip/src/lib/support/BUILD.gn:351:5: which caused the file to be included.
    "$dir_pw_log:impl",
    ^-----------------

sayondeep, 2024-07-01:
@Aryan-Tamboli 

> Describe the bug
A clear and concise description of what the bug is.

To diagnose the issue further, please give clear description of what you did...on running which command it failed 

> Environment
> 
> ESP-Matter Commit Id:
> ESP-IDF Commit Id:
> SoC (eg: ESP32 or ESP32-C3):
> Device Logs (Please attach the log file):
> Host Machine OS:
> Commissioner app and versions if present:
> Commissioner's logs if present:

Please provide the requested details

Aryan-Tamboli, 2024-07-01:
ESP-Matter Commit Id: https://github.com/espressif/esp-matter/commit/91b8cb1596cdcf09bf379fdefa2cd11e46a83009
ESP-IDF Commit Id: 
SoC (eg: ESP32 or ESP32-C3): ESP32C3
Host Machine OS: Windows 11 (WSL2 - Ubuntu 22.04)

I tried following commands in steps

cd esp-idf; source ./export.sh; cd ..
cd esp-matter; source ./export.sh; cd ..

cd esp-matter/example/room_air_conditioner

idf.py set-target esp32c3

idf.py build

[build_ac_example.txt](https://github.com/user-attachments/files/16049733/build_ac_example.txt)

Aryan-Tamboli, 2024-07-01:
I have attached the build log file above , the errors are : 
[940/2046] Performing configure step for 'chip_gn'
FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure //matter/light/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure 
cd //matter/light/build/esp-idf/chip && /usr/bin/gn --root=//.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32 gen --check --fail-on-unused-args //matter/light/build/esp-idf/chip && //.espressif/tools/cmake/3.24.0/bin/cmake -E touch //matter/light/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure
ERROR at //build_overrides/pigweed_environment.gni:20:1: Unable to load "//.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/build_overrides/pigweed_environment.gni".
import("${_bootstrap_root}/build_overrides/pigweed_environment.gni")
^------------------------------------------------------------------
See //third_party/connectedhomeip/third_party/pigweed/repo/pw_protobuf_compiler/proto.gni:16:1: whence it was imported.
import("//build_overrides/pigweed_environment.gni")
^-------------------------------------------------
See //third_party/connectedhomeip/third_party/pigweed/repo/pw_log/BUILD.gn:22:1: whence it was imported.
import("$dir_pw_protobuf_compiler/proto.gni")
^-------------------------------------------
See //third_party/connectedhomeip/src/lib/support/BUILD.gn:351:5: which caused the file to be included.
    "$dir_pw_log:impl",
    ^-----------------
[951/2046] Building CXX object esp-idf/openthread/CMakeFiles/__idf_openthread.dir/openthread/src/core/net/srp_client.cpp.obj
ninja: build stopped: subcommand failed.

sayondeep, 2024-07-01:
@AFellmett @Aryan-Tamboli 
For both of you, I can see that esp-matter is being referred form espressif tool directory i.e., ` .espressif/esp-matter`
Please move the esp-matter outside tools directory to somewhere else.

If using vs-code extension to install esp-matter,
please try to install esp-matter in different directory and not inside esp-idf tools directory [do not select (use current esp-idf tools directory) while installing from the extension]. You can do this instead,
choose existing repository --> github --> select a directory to install esp-matter.

If installation fails via vs-code extension please follow the instructions [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#esp-matter-setup) to install using terminal instead.

If using vs-code is necessary, first clone esp-matter to some other directory and then select that directory to install from vs-code as mentioned above.

sayondeep, 2024-07-01:
@Aryan-Tamboli are you using terminal or vs-code extension to build the project?
Can you please try using terminal if not tried using it.

Was the installation of esp-matter successfull? i.e., running `./install.sh`
Please remove the environment by running ` rm -rf path/to/esp-matter/connectedhomeip/connectedhomeip/.environment` and try to install esp-matter.

Aryan-Tamboli, 2024-07-01:
I have WSL2 setup, I build codes from the wsl terminal itself. I am able to flash CHIP Matter sdk examples but ESP-Matter sdk cause these errors.

sayondeep, 2024-07-01:
@Aryan-Tamboli 
> Was the installation of esp-matter successfull? i.e., running `./install.sh` Please remove the environment by running ` rm -rf path/to/esp-matter/connectedhomeip/connectedhomeip/.environment` and try to install esp-matter.

did you try these?

Aryan-Tamboli, 2024-07-02:
Thanks @sayondeep , now the I am able to flash the code successfully

closed 2024-07-02 by dhrishi