espressif/esp-matter #989: How to build all the layers in the repository

pavel808, 2024-06-28:
I am adding a new Matter cluster server to the light example in examples/light .This involves making changes not only to the example application code, but also to the internal _components_ and _connectedhomeip_ .

How do I execute a build that builds all changes made to the _components_ and _connectedhomeip_ etc. ?

By running **idf.py build** from the example/light folder, it doesn't detect changes made to _components_, _connectedhomeip_ or anything else in the lower layers. It only builds the application. 

Thanks in advance

dhrishi, 2024-06-28:
>This involves making changes not only to the example application code, but also to the internal components and connectedhomeip 

@pavel808 Ideally, any functionality you want to achieve should be exposed through APIs/events in the SDK and you should not have to modify the core components. Can you please let us know what are you trying to do?

pavel808, 2024-06-28:
@dhrishi Thank you for your reply.

Ok, so what I am trying to do is adapt the light example to have new 64 bit value attributes, and to be able to receive such values via commands from a client such as chip-tool.

I tried simply re-using attributes for saturation and hue etc., and the commands to change them, but I need to be able to deal with 64 bit values.   

I realized then that it probably makes most sense to create a separate cluster server for my purpose. I took the levelcontrol cluster server as a reference and duplicated the functionality everywhere I saw it(in components and connectedhomeip etc.) and renamed it to my own, with the new IDs attributes commands etc.

Of course, now I am unable to build these changes. 

In the end I want to be able to use chip-tool to send commands to update the new attributes via my new cluster server. So i'm not sure how to achieve this any other way.

dhrishi, 2024-06-28:
@pavel808 Okay. Basically you are looking to create a custom cluster and handling the commands. Can you please refer [this](https://github.com/espressif/esp-matter/blob/91b8cb1596cdcf09bf379fdefa2cd11e46a83009/components/esp_matter_rainmaker/esp_matter_rainmaker.cpp#L362) and see if it helps.
All your changes can just be in your application and need not even be in the esp-matter component.

pavel808, 2024-06-28:
@dhrishi Thanks. I will look into this and let you know.

dhrishi, 2024-07-10:
Closing this as the question is answered. Please raise a new issue in case of any problems

closed 2024-07-10 by dhrishi