espressif/esp-matter #990: No Enough heap size for me.

ronny-antoon, 2024-06-30:
**Describe the question/query that you have**
the light example, without modification esp_matter 1.3.0 esp-idf 5.1.2

only the on_off_light,
49000 Only heap size left for me. and that not enough 

any ideas

jonsmirl, 2024-06-30:
If you need to add significant code to esp-matter you will need to use an ESP32 version with PSRAM.  I find even the smallest 2MB PSRAM to be plenty.  From my system "Free internal 34,192 PSRAM free 1,474,332" so I am using 610KB PSRAM. I am probably wasting half of that 610KB with unimportant things.

dhrishi, 2024-07-01:
@ronny-antoon Please check if [this](https://docs.espressif.com/projects/esp-matter/en/latest/esp32c3/optimizations.html) section of the documentation is helpful

jonsmirl, 2024-07-01:
I would recommend for development to just use the PSRAM. Initially tried without using it and I spent way too much time fighting with RAM usage.  And then only after everything is working see if you can squeeze things down enough to avoid the PSRAM. Doing it this way will give you more RAM to turn on things like heap tracking too.

ronny-antoon, 2024-07-06:
thank you alot

closed 2024-07-06 by ronny-antoon

stefano055415, 2024-07-25:
I'm having the same problem too:

With esp-matter version 1.1 free heap size = 104700

With esp-matter version 1.3 free heap size = 49000

Obviously with the same firmware.

the difference is really big...

I also tried applying the changes suggested by @dhrishi, but the improvement is only 6000kb reaching 55000

Is there any optimization planned?

jonsmirl, 2024-07-25:
Matter is only going to keep growing.  I spent months fighting this, then gave up and added PSRAM. You might get it to work for 1.3 but then 1.4 is going to grow and you won't be able to fit.   We also switched to 8MB flash for the same reason. Our image is right at 2MB and if it grows it won't fit into 4MB flash.  Using the ESP32-S3-PICO-N8R2 with have plenty of flash and PSRAM for expansion.

ronny-antoon, 2024-07-25:
> I'm having the same problem too:
>
>
>
> With esp-matter version 1.1 free heap size = 104700
>
>
>
> With esp-matter version 1.3 free heap size = 49000
>
>
>
> Obviously with the same firmware.
>
>
>
> the difference [...]

55000 free after all application. Its so good. Dont need more

ronny-antoon, 2024-07-25:
> Matter is only going to keep growing.  I spent months fighting this, then gave up and added PSRAM. You might get it to work for 1.3 but then 1.4 is going to grow and you won't be able to fit.   We [...]

I think the sdk should fetch the esp32 standard without any modification.

stefano055415, 2024-07-25:
@ronny-antoon
55000 is enough for now, but if it continues to grow like this, we will soon be forced to change hardware.

50000 lost in only 2 versions... between another two we will no longer be able to compile

jonsmirl, 2024-07-25:
A lot of the grow is coming from the CHIP core, there isn't much Espressif can do about that.

stefano055415, 2024-07-25:
@jonsmirl 
Let's hope for an optimization...

jonsmirl, 2024-07-25:
No PSRAM should be fine for light bulbs. But anything more complex is going to need PSRAM. Don't expect this code to get any smaller. Code never gets smaller, it only grows.

You can drop out all of the esp-matter code if you want, see this example: https://github.com/espressif/esp-matter/tree/main/examples/zap_light  Zap is much harder to use for complex cases but for something like a light bulb it is not that hard to use.  That example has 118KB free while the esp-matter version has 105KB free.  Need to check if those are current numbers.

stefano055415, 2024-07-25:
I think you're right, I've already tried using CHIP without esp, but the difference remains minimal.

jonsmirl, 2024-07-25:
I just built this example for the S3 https://github.com/espressif/esp-matter/tree/main/examples/light  using current CHIP.  It has a lot of free memory. The S3 will have more free than the C3.
~~~
I (20609) app_main: Dump #1 Free internal 181672 biggest block 61424 total 181672 PSRAM free 0
~~~

stefano055415, 2024-07-25:
@jonsmirl 

Have you also tried to compile it with release 1.3?

to understand if there is a difference compared to the current one?

jonsmirl, 2024-07-25:
That is with head of esp-matter so it is beyond 1.3.

Maybe you have turned things on in your sdkconfig which are eating a lot of memory. Try using the sdkconfig from the lighting example.