espressif/esp-matter #991: How do I get the MAC address of commissioned devices in a Matter controller example code?

vishnuprasadvp12345, 2024-07-01:
1)   In the Matter controller example code, I used a commissioned Matter light example code. How do I get the MAC address of that                 commissioned controller (light example) from the Matter controller code?
2) I am using the BLE-WiFi method to commission devices. I want to commission a third-party Matter device using the BLE-WiFi method and need to pass parameters like the setup code and discriminator. How do I decode the third-party device's QR code?

wqx6, 2024-07-01:
1. You can try to read NetworkInterfaces attribute of GeneralDiagnostics cluster with the read_command of our controller.
2. You can use the [QRCodeSetupPayloadParser](https://github.com/project-chip/connectedhomeip/blob/master/src/setup_payload/QRCodeSetupPayloadParser.h) to parse the QR code.

vishnuprasadvp12345, 2024-07-01:
matter esp controller read-attr 4 2 0x0033 0x0006

read_command: Response Failure: Error IM:0x000005C3<0x1b>[0m
<0x1b>[0;32mI (191787) read_command: read done<0x1b>[0m

wqx6, 2024-07-01:
`Error IM:0x000005C3` is `UNSUPPORTED_CLUSTER`. The GeneralDiagnostics cluster should be on the Endpoint 0, but you are reading the cluster on Endpoint 2.

vishnuprasadvp12345, 2024-07-01:
matter esp controller read-attr 4 0 0x0033 0x0006 Same error using 0

wqx6, 2024-07-02:
The NetworkInterfaces attribute Id is 0x0000 instead of 0x0006.

vishnuprasadvp12345, 2024-07-02:
how to devices mac address

wqx6, 2024-07-02:
After you read the NetworkInterfaces attribute, you can get the `HardwareAddress` values of the interfaces, which are the MAC addresses for those network interfaces.

wqx6, 2024-07-02:
You can refer to [this document](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html?highlight=mac#mac-address).

dhrishi, 2024-07-26:
Closing as the question is answered

closed 2024-07-26 by dhrishi