espressif/esp-matter #992: Test Execution Failed in Test Harness [TC-G-1.1]

FMistrorigo, 2024-07-01:
**Description**

I'm trying to run all the preliminary tests to certify my matter device, I encountered a problem in Test Case TC-G-1.1 when the Semi-automatic test checks the DUT FeatureMap. The expected value is 0, while in my case it is 1, as shown in the attached image. 
I'm not using the ZAP Tool, but I created the root node and my endpoint with the functions provided by the Matter API.

[image]

**Additional details**

I noticed that in the file esp_matter_cluster.cpp in the esp-matter repository, when the groups cluster is created, if the cluster is a server, it uses this function: 

global::attribute::create_feature_map(cluster, 1); 

Every single cluster creates the feature_map attribute, but this is the only one with value=1 and i don't know why.

**Environment**

- ESP-Matter Commit Id: 8b7d90a5bde32be08ded595855ab98b9efa0ea10
- ESP-IDF Commit Id: 482a8fb2d78e3b58eb21b26da8a5bedf90623213
- SoC: ESP32-S3
- Device Logs: [TC-G-1.1_Error_Log.txt](https://github.com/user-attachments/files/16050625/TC-G-1.1_Error_Log.txt)
- Host Machine OS: Windows 11
- Test Harness Logs: [TH_TC_G-1.1_Logs.txt](https://github.com/user-attachments/files/16050636/TH_TC_G-1.1_Logs.txt)
[image]
[image]

FMistrorigo, 2024-07-01:
A similar problem happens in the Test Case TC-G-2.1 Step 1b, where the value of NameSupport is checked and should be equal to 0, but in my case is equal to 128.
I followed all the steps in the creation of the attribute NameSupport but the only thing that I found is the use of the function 

create_group_name_support(cluster, config->group_name_support);

but in my case inside config i didn't specify the value of group_name_support, so i don't know where the value 128 came from.

VaishaliAvhale, 2024-07-03:
Hi @FMistrorigo,

For TC-G-1.1, the feature map is 1 because the GroupNames feature of the Groups Cluster is enabled, which is mandatory. So, while generating PICS, choose the GroupNames feature and upload updated PICS then execute the test case.

For TC-G-2.1, the NameSupport value is 128 because the GroupNames feature is supported. Below you can see the expected outcome of step 1b:

[image]

FMistrorigo, 2024-07-04:
Hi @VaishaliAvhale,

Thanks for the response, I checked the PICS as you suggested and found out that the GroupNames feature was probably the problem, but my "Groups Cluster Test Plan.xml" file, once uploaded on the PICS tool, told me the GroupNames feature was optional (O) as show in the picture: 
[image]

During the tests i noticed that i was working in a wrong connectehomeip brach (main) and not with the one recomended in the documentation, so the real problem might be that i was using esp-idf 5.1.2 and esp-matter 1.2 but with the latest commit of connectedhomeip and using the test for matter 1.2, causing a lot of problems.

Now i'm having troubbles with the version of connectedhomeip and can't check if the problem is solved. I'll let you know if I figure something out.

Thanks again.

VaishaliAvhale, 2024-07-05:
If you want to execute a test case for Matter V1.2 then you are using the correct TH.For resolving connectedhomeip-related problem just checkout to the esp-matter release/v1.2 [https://github.com/espressif/esp-matter/tree/release/v1.2] branch and update submodule below steps
1. path-to-esp-matter$ git submodule update --init --depth 1
2. path-toesp-matter/conectedhomeip/connectedhomeip$ git submodule update --init --depth 1
clean build example 
3. path-to-esp-matter/example/light$ rm -rf build sdkconfig sdkconfig.old managed_components/ dependencies.lock
4. idf.py set-target esp32s3
5. idf.py build flash monitor
All set start testing 
About GroupNames feature: We are looking into it will get back to you soon

shubhamdp, 2024-07-05:
Hi @FMistrorigo as you correctly pointed out, NameSupport feature is optional and should not be set to `1`.
https://github.com/espressif/esp-matter/blob/ab58e2993d268844e9ab64af586681e59b659ffd/components/esp_matter/esp_matter_cluster.cpp#L1233

If possible, can you try changing it to `0` and run with default PICS i.e. (name support disabled pics).

We will fix this in the SDK.

FMistrorigo, 2024-07-05:
Hi @VaishaliAvhale and @shubhamdp,

I replicated all the steps, and i'm now working with this environment:
ESP-IDF Commit Id: a322e6bdad4b6675d4597fb2722eea2851ba88cb (HEAD, tag: v5.2.1)
ESP-Matter Commit Id: 1a35c738dff33e5e988d92d0028ed7a60d4336f6 (HEAD -> release/v1.2, origin/release/v1.2)
connectedhomeip Commit Id: 46e1ad8478c55ec20273da9d40a9c6fecf5c6b38
SoC: ESP32-S3
Host Machine OS: Ubuntu 22.04.3 (WSL)

I'm facing other issues that I already reported in a different ticket (https://github.com/espressif/esp-matter/issues/994)
I solved temporarly the issue by changing the esp-matter code.
In addition to this problem, my device is rebooting after a Guru Meditation Error which happens sporadically, so there's no point in me running tests until I fix this problem.
[GuruMeditationError esp_matter_start.txt](https://github.com/user-attachments/files/16108643/GuruMeditationError.esp_matter_start.txt)

PS: I'm searching for a solution to this problem but i'm probably gonna open a new ticket just for it, providing all te needed informations.

VaishaliAvhale, 2024-07-05:
Hi @FMistrorigo, Look's like esp-idf not setup correctely and connectedhomeip not on required commit please follow below step's and try again
1. path-esp-idf$ git reset --hard a322e6bdad4b6675d4597fb2722eea2851ba88cb
2. git submodule update --init --recursive
3. ./install.sh
4. source ./export.sh
5. path-to-esp-matter$ ./install.sh
6. source ./export.sh
7. path-to-esp-matter/connectedhomeip/connectedhomeip$ git reset --hard ba43f9d8a83821aa5724282b64d2e6a6c07e06f6

follow above step's and if you face any issue share complete log's here.

FMistrorigo, 2024-07-05:
Ok, If the written steps are all and in the correct order, I'm sure to be on the right commits and I'm sure i have esp-idf and esp-matter setup correctly because i followed every single step.
ESP_IDF: a322e6bdad4b6675d4597fb2722eea2851ba88cb (HEAD, tag: v5.2.1)
ESP_MATTER: 1a35c738dff33e5e988d92d0028ed7a60d4336f6 (HEAD -> release/v1.2, origin/release/v1.2)
connectedhomeip: ba43f9d8a83821aa5724282b64d2e6a6c07e06f6 (HEAD, origin/v1.2-branch, v1.2-branch)

LOGS of the version upgrade and setup: [update_version_log.txt](https://github.com/user-attachments/files/16110300/update_version_log.txt)

I'm still facing the issue i linked before, and this other error:
[image][image]

Guru Meditation Error Logs: 
[GuruMeditationError.txt](https://github.com/user-attachments/files/16110339/GuruMeditationError.txt)

NOTE: If i can't solve the problem I'll try to remove and do a clean installation of esp-idf and esp-matter, even if with the steps you provided it should be the same.

FMistrorigo, 2024-07-09:
Hi,

I’m still encountering issues with the “Guru Meditation Error.” but I tried to run the test again to see if it works with your suggestions.

Running TC-G-1.1 still results in the same problem. Initially, I followed @shubhamdp’s advice and modified the code, which allowed the test to pass. However, I’m concerned that when I submit my product for testing, it might not pass this test.

I also enabled the GroupNames feature and executed the test case as recommended by @VaishaliAvhale. The test succeeded using this approach. I’m leaning toward keeping the PICS with the feature to avoid any potential issues.

The confusion arises from conflicting information: while the code assumes the GroupNames feature is mandatory, the PICS file states that it’s optional.

VaishaliAvhale, 2024-07-09:
Hi @FMistrorigo,

The GroupNames feature is optional and has now been removed from the SDK. While testing, please follow this [link](https://github.com/espressif/esp-matter/issues/992#issuecomment-2210607749) and update the PICS by unselecting the GroupNames feature. Then, test the TC. The test case will pass for sure.

VaishaliAvhale, 2024-07-09:
Hi @FMistrorigo,

I have tried to build the esp-matter release/v1.2 light example for the [esp32s3], and it was successfully built on my end. I followed the same steps that I provided to you. I have attached the complete process logs below.

[Uploading release-v1-2-light-esp32s3-build.log…](https://github.com/espressif/esp-matter/issues/992)

FMistrorigo, 2024-07-09:
Hi @VaishaliAvhale,

I changed the SDK and the TC-G-1.1 is now ok, the problem persist in the 2.1, because as you said GroupNames feature is supported, so it's expecting 0 and not 128.
So, again, the code implements the GroupName attribute, but the Test thinks it's not supported because the PICS say so.

[image]
As you can see, inside the namespace groups of the file esp_matter_providers.cpp, the function "create_group_name_support" is called, even if in the documentation this attribute is optional.

I will activate the feature in the PICS and move on to other tests.

VaishaliAvhale, 2024-07-09:
If you make FeatureMap attribute 0 that means no one feature is supported i.e GroupNames Not Supported because it is an optional feature. It was our side bad that we enabled GroupNames by mistake now we have removed that but as the changes are not yet live hence shubham suggested this change https://github.com/espressif/esp-matter/issues/992#issuecomment-2210607749. 

To solve TC-G-2.1 issue you have to **unselect GroupNames Feature** from the PICS and re-execute the test case

 If GroupNames is not supported then NameSupport will be 0. and this is expected behavior.

Sorry for the confusion please ignore my first comment GroupNames Feature not mandatory and you don't need to select it in the PICS.[https://github.com/espressif/esp-matter/issues/992#issuecomment-2205625064]

closed 2024-07-11 by espressif-bot