espressif/esp-matter #994: Problem in function get_dac_provider of esp_matter_providers.cpp

FMistrorigo, 2024-07-04:
**Description**

I recently updated esp-idf to version 5.2.1 as suggested by the esp-matter 1.2 documentation, doing so created compilation and linker problems.
I would like to point out that in my code I use a custom provider for the Device Atterestation Credentials (DAC), so CONFIG_CUSTOM_DAC_PROVIDER is active in my sdkconfig.

When I compile I have problems in the esp-matter-providers.cpp file in the get_dac_provider function, in which:

1. ESP_LOGE(TAG, "custom_dac_provider cannot be NULL"); gives an error during compilation because apparently #include <esp_log.h> is missing
2. return chip::Credentials::Examples::GetExampleDACProvider(); problems because I changed the value of CHIP_DEVICE_CONFIG_DEVICE_VENDOR_ID and CHIP_DEVICE_CONFIG_DEVICE_PRODUCT_ID, inserting the values ​​of my product. This causes a chain reaction that causes:
          a) In ExampleDACs.cpp the following are never defined:
              - kDacCert
              - kDacPrivateKey
              - kDacPublicKey
          b) In ExamplePAI.cpp the following are never defined:
              - kPaiCert
              - kPaiPrivateKey
              - kPaiPublicKey

**Additional details**
I tried to solve the problem by adding the #include <esp_log.h> and declaring the missing variables, it worked, but i'm not the only one that is working at the current project, and all my coworkers, with a clean build of esp_matter and connectedhomeip, may need to make the same changes in the code.

I Attached the Logs to let you know exactly what's the problem.

**Environment**
- ESP-Matter Commit Id: 1a35c738dff33e5e988d92d0028ed7a60d4336f6 (HEAD -> release/v1.2, origin/release/v1.2)
- connectedhomeip Commit Id: ba43f9d8a83821aa5724282b64d2e6a6c07e06f6 (HEAD -> v1.2-branch, origin/v1.2-branch)
- ESP-IDF Commit Id: a322e6bdad4b6675d4597fb2722eea2851ba88cb (HEAD, tag: v5.2.1)
- SoC: ESP32-S3
- Device Logs:
- Host Machine OS: Ubuntu 22.04.3 (WSL)
- Device Logs (with the ESP_LOGE problem): [Log1.txt](https://github.com/user-attachments/files/16096664/Log1.txt)
- Device Logs (with the GetExampleDACProvider):  [Log2.txt](https://github.com/user-attachments/files/16096669/Log2.txt)

wqx6, 2024-07-08:
For the first problem, we will add the missing header ASAP.
And for the second problem, the ExampleDACProvider only provides DAC with vendorID 0xFFF1/0xFFF2 and productID 0x8000-0x801F, and it is only for development test,  so you can not use that DAC Provider for your product.

FMistrorigo, 2024-07-08:
Hi @wqx6 

[image]
This is part of the code in esp_matter_providers.cpp. The function get_dac_provider() is used to set the correct DAC provider to use in the  function setup_providers();

I can't understand why, the "return chip::Credentials::Examples::GetExampleDACProvider();" is compiled every time, this causes compilation problem for everyone that uses a VENDOR_ID or a PRODUCT_ID different from the ones in the range 0x8000-0x801F.

To explain myself better, if I have a VENDOR_ID of 0x1349, the file DeviceAttestationCredsExample.cpp is compiled anyways.
Inside it's functions the variables DevelopmentCerts::kDacCert, DevelopmentCerts::kPaiCert, DevelopmentCerts::kDacPrivateKey and DevelopmentCerts::kDacPublicKey are used as you can see in the image:
[image]

Checking the way these variables are defined inside the files ExampleDACs.h and ExamplePAI.h, we can see this:
[image]

The variables are defined with the extern but ther's not a definition of these variables for the case VENDOR_ID=0x1349 because in the ExampleDACs.cpp and ExamplePAI.cpp the #if that checks the value of the VENDOR_ID and the PRODUCT_ID, prevent the definition of these variables.

The way it can be solved is to provide a default definition of the different variables if none of the previous #if is compiled.
For Example adding this at the end of the code:

[image]

It's not the bast way to solve this problem and I don't know if it creates other issues, but for now at least is compiling.

shripad621git, 2024-07-21:
@FMistrorigo , from the edited description it seems that your issue is addressed using `DAC Provider Options` [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#factory-data-providers) i.e setting the appropriate `custom_dac_provider` before `esp_matter::start()`. 
Please close the issue if it is solved.

FMistrorigo, 2024-07-24:
Hi, 
As I said before the code in esp_matter_providers.cpp is this:
[image]
And if you consider that CONFIG_CUSTOM_DAC_PROVIDER=y, the way it works is like this:
[image]

If you look at the code you will see that  _return chip::Credentials::Examples::GetExampleDACProvider();_ is **compiled** every time and this is the problem. Please read again [this post](https://github.com/espressif/esp-matter/issues/994#issuecomment-2213219315) and follow the steps where I explain where the problem is.

I'm sure that I'm using my custom DAC provider, and I'm sure it works fine because with my edit to the code everything is perfect.

The only thing I'm not sure about is that in my configuration file i define the VendorID and ProductID like this:
[image]
The problem might be **the name of these two variables**, because the same name is used for the test ones, but in this case they are not in the range vendorID 0xFFF1/0xFFF2 and productID 0x8000-0x801F.

shripad621git, 2024-07-26:
@FMistrorigo , can you please try the patch below. I have tried it by using set_custom_dac_provider and the below patch. The compilation went through without any linking errors and the code changes. 
[compilation_fix.txt](https://github.com/user-attachments/files/16388652/compilation_fix.txt)

shripad621git, 2024-07-29:
@FMistrorigo, any updates here. If you want to try out the change, you can try it out with latest esp-matter as the change is merged [here](https://github.com/espressif/esp-matter/commit/10d519e89c57e4937b3184c6dd4800b6371c2b97).

FMistrorigo, 2024-07-29:
I just tryed it and it works fine without problems, Thank you very much.

closed 2024-07-29 by FMistrorigo