espressif/esp-matter #995: unable to connect esp-matter device to home-assistant

Dimandragon, 2024-07-04:
I tryed to connect esp32-s3 to home-assistant using door_lock example from esp-matter repo using qr code from https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning-and-control and home-assistant mobile app; The mobile application successfully completes the stages of finding a device, connecting to it, creating matter credentials and checking the network connection, but it fails on "connecting device to Home Assistant" stage

**Environment**
- ESP-Matter release/v1.3
- ESP-IDF v5.2.1
- ep32-s3
- logs in file [log_matter.txt](https://github.com/user-attachments/files/16102728/log_matter.txt)
- EndeavourOS x86_64 and Android 14
- Home Assistant Supervised and Home Assistant Android App 

Before flashing i erased flash memory from device. I think that maybe problem in home assistant, or maybe i forgot to set some configure options before building project (i used default sdkconfig)

sayondeep, 2024-07-05:
@Dimandragon It is a problem related to google play services while discovering a device over mdns. Can you please try after restarting the phone or may be try on another network.

Dimandragon, 2024-07-07:
> @Dimandragon It is a problem related to google play services while discovering a device over mdns. Can you please try after restarting the phone or may be try on another network.

I tried reboot my phone and i'v got same result (log is hear 
[matter_log_reboot.txt](https://github.com/user-attachments/files/16118870/matter_log_reboot.txt)
)

also i've tryed to use home assistant from iphone13 with same result (log also is hear
[matter_log.txt](https://github.com/user-attachments/files/16118872/matter_log.txt)
)

sayondeep, 2024-07-12:
@Dimandragon any update on the issue?

closed 2024-10-08 by dhrishi