espressif/esp-matter #996: Actions cluster

jonsmirl, 2024-07-05:
I tried coding up the Actions cluster to see if Apple Home understands it.
~~~
    endpoint::add_device_type(aggregator, LOWPAN_DT_AGGREGATOR, LOWPAN_DT_VERSION);
    aggregator_endpoint_id = endpoint::get_id(aggregator);

    actions::config_t actions_config;
    cluster_t *actions = cluster::actions::create(aggregator, &actions_config, CLUSTER_FLAG_SERVER);
    attribute_t *endpointlists = attribute::get(actions, Actions::Attributes::EndpointLists::Id);
    EndpointId eps[] = { 3, 4, 5};
    chip::app::DataModel::List<chip::EndpointId> endpoints;
    endpoints = eps;
    Actions::Structs::EndpointListStruct::Type endpointListStruct = {
                1234, CharSpan::fromCharString("Yellow"), Actions::EndpointListTypeEnum::kZone,
                endpoints
            };
    uint8_t *value = (uint8_t *)&endpointListStruct;
    uint16_t length = sizeof(endpoints);
    uint16_t count = 1;
    esp_matter_attr_val_t val = esp_matter_array(value, length, count);
    attribute::set_val(endpointlists, &val);
~~~
That code gives this error:
~~~
I (453052) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000025's Attribute 0x00000000 is <invalid type: 4> **********
E (453053) chip[DMG]: Attribute type 0x48 not handled
I (453055) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000025's Attribute 0x00000001 is <invalid type: 4> **********
E (453056) chip[DMG]: Attribute type 0x48 not handled
~~~
So maybe Attribute 0 is not getting initialized to an empty list? I did not use Attribute 0;
Any idea what is wrong with my test code for Attribute 1?

jonsmirl, 2024-07-05:
Doesn't appear to have any effect in Apple Home. I thought the Room field would be pre-filled with 'Yellow'.

jadhavrohit924, 2024-07-12:
@jonsmirl Esp-Matter's SDK doesn’t support reading structure/list from Esp-Matters storage. There are no src/app/clusters for the `Actions` cluster which is why when the read request comes, the device doesn’t find that attribute in the Connectedhomeip, and when it comes to Esp-Matter, Esp-Matter doesn’t know how to encode the list so you get `Attribute type 0x48 not handled`

jonsmirl, 2024-07-12:
Look at the second comment, I built an AttributeAccessOverride to send the attribute and verified that is was correctly generated.

None of Google/Apple/Amazon appear to do anything with the Actions cluster. Can you ask around and see if anything is using it?