espressif/esp-matter #997: problem in TC-TSTAT-2.1

FMistrorigo, 2024-07-09:
**Description**
I'm trying to test my device that imlements a thermostat cluster, in the PICS i added the "Thermostat Cluster Test Plan.xml" file and this implemented the test case TSTAT.

The problem is in the step 18a in my case, the test tryes to "Read attribute MaxHeatSetpointLimit from the DUT" but i get the error "Test Failure: The response value (3500) should be lower or equal to the constraint but 3500 > 3000.".
[image]

I thougt that maybe I didn't initialized correctly the values of the attributes of the thermostat but it was not the case, in fact everything seems fine and I don't have any attribute with value 3000.

In the previous steps (Step 1) it reads all the attribute values and in both AbsMaxHeatSetpointLimit and MaxHeatSetpointLimit the value is 3500. 
[image]
I can't figure out why in the test is used 3000 and not the 3500 value readed before.

The problem is only in the Heating Attributes and i don't know why.

These are my values:
[image]

**Environment**
ESP-IDF Commit Id: a322e6bdad4b6675d4597fb2722eea2851ba88cb (HEAD, tag: v5.2.1)
ESP-Matter Commit Id: https://github.com/espressif/esp-matter/commit/1a35c738dff33e5e988d92d0028ed7a60d4336f6 (HEAD -> release/v1.2, origin/release/v1.2)
connectedhomeip Commit Id: 46e1ad8478c55ec20273da9d40a9c6fecf5c6b38
SoC: ESP32-S3
Host Machine OS: Ubuntu 22.04.3 (WSL)

FMistrorigo, 2024-07-10:
This problem affect the TC-TSTAT-2.2, in the step 7c "Writes the limit of MinHeatSetpointLimit to MaxHeatSetpointLimit attribute", in this test it's checked that MinHeatSetpointLimit <= MaxHeatSetpointLimit <= AbsMaxHeatSetpointLimit, but if the AbsmaxHeatSetpointLimit is wrong the test fails. (In my case 300<=3500<=3000)

It seeems that the test can't save the right value of AbsmaxHeatSetpointLimit or in the test is used 3000 as default value.

FMistrorigo, 2024-07-12:
Why is the test using 3000 as value to check if the MaxHeatSetpointLimit is valid?

VaishaliAvhale, 2024-07-15:
3000 is the default value of the MaxHeatSetpointLimit attribute. As I mentioned earlier, the certification test cases are designed based on the default value. If you want to test the changes you have made, please update the test suite, otherwise, set the attribute value to the default.
To update the existing YAML Test Script, follow the steps shown in the attached screenshot:
[image]

FMistrorigo, 2024-07-25:
Thanks, it solved the problem, I still can't understand why the test was created like this, it was possible to make it fully automatic and without hardcoded values, but maybe you have your reasnons.

closed 2024-07-25 by FMistrorigo