espressif/esp-matter #1000: Do i need to use  Access Control Cluster.xml in my PICS?

FMistrorigo, 2024-07-11:
**Description**
In my project i defined the root node in this way:
[image]

node::config_t, in my esp_matter version, is defined inside esp_matter_endpoint in this way:
[image]

endpoint::root_node::config_t root_node creates an endpoint with these clusters:
[image]

As you can see the cluster access_control is defined, do i have to implement the Access Control Cluster.xml in my PICS even if I don't use it?

The access control is defined only with the cluster _revision attribute:
So in the test, when it tryes to read the attribute AccessContorlEntryChanged that is obviously not implemented but is Mandatory in the PICS if the Access Control Cluster as Server(ACL.S) is defined.
[image]

VaishaliAvhale, 2024-07-12:
If you are performing Access Control or Access Control Enforcement Cluster's test case then you need to use Access Control Cluster.xml 

>>The access control is defined only with the cluster _revision attribute:

It is header file. 
You will get the actual implementation of the Access Control Cluster in [esp_matter_cluster.cpp ](https://github.com/espressif/esp-matter/blob/f0f94acb1071c6b4a707981ab31ddafdf2924012/components/esp_matter/esp_matter_cluster.cpp#L187)file.

FMistrorigo, 2024-07-12:
Ok, so if the Access Control Cluster is defined but I **don't** implement it, I don't need to do the test with the Access Control Cluster.xml right?

VaishaliAvhale, 2024-07-12:
The Root Node is mandatory for all device types, and the Access Control Cluster is a mandatory cluster of the Root Node. Therefore, it must be supported.

>> Root Node: This endpoint is akin to a "read me first" endpoint that describes itself and the other endpoints that make up the node.

FMistrorigo, 2024-07-12:
Do I have to do the test or not?

The test checks all the mandatory attributes, but the SDK doesn't implement all of them.
[image]

Di I need to implement all of them or can I skip the test?

VaishaliAvhale, 2024-07-15:
Hi @FMistrorigo, It is a header file. You will find the actual implementation of the Access Control Cluster in the [esp_matter_cluster.cpp](https://github.com/espressif/esp-matter/blob/f0f94acb1071c6b4a707981ab31ddafdf2924012/components/esp_matter/esp_matter_cluster.cpp#L187) file.

All mandatory clusters, attributes, features, events, and commands are implemented in esp-matter. If you create a device type, you don't need to implement these mandatory elements yourself.

dhrishi, 2024-07-26:
@FMistrorigo Please close this if your questions are answered

FMistrorigo, 2024-07-29:
Thanks, I found the implementation and wrote some code to see the attribute of every cluster of every endpoint to check if the mandatory ones are present and now is fine.

closed 2024-07-29 by FMistrorigo