espressif/esp-matter #1001: wifi:m f null

jonsmirl, 2024-07-11:
I am getting wifi error again. I have plenty of RAM free.
~~~
app_main: Dump #272 Free internal 97948 biggest block 32740 total 97948 PSRAM free 1896072
~~~
This looks like some kind of bug with retransmits not freeing a buffer.
[sdkconfig.defaults.txt](https://github.com/user-attachments/files/16184049/sdkconfig.defaults.txt)

jonsmirl@ares:~/esp/esp-idf$ git log
commit a322e6bdad4b6675d4597fb2722eea2851ba88cb (HEAD, tag: v5.2.1)

~~~
(5390310) HAL_Rocker: HAL_Rocker::processEvent 7
I (5390310) HAL_Dimmer: HAL_Dimmer::processEvent
I (5390315) HAL_Rocker: Pow 206 206 Bri 1 1
I (5395294) ble_cluster: BLE addr 34:f1:ae:d4:08:00
I (5395295) HAL_Rocker: HAL_Rocker::processEvent 7
I (5395297) HAL_Dimmer: HAL_Dimmer::processEvent
I (5395300) HAL_Rocker: Pow 206 206 Bri 1 1
I (5396342) chip[EM]: <<< [E:13725i S:28487 M:92478493] (S) Msg TX to 2:AB0DE00000000000 [2C76] [UDP:192.168.1.96:35800] --- Type 0001:05 (IM:ReportData)
I (5396402) chip[EM]: >>> [E:13725i S:28487 M:230503301 (Ack:92478493)] (S) Msg RX from 2:AB0DE00000000000 [2C76] --- Type 0001:01 (IM:StatusResponse)
I (5396403) chip[IM]: Received status response, status is 0x00
I (5396405) chip[EM]: <<< [E:13725i S:28487 M:92478494 (Ack:230503301)] (S) Msg TX to 2:AB0DE00000000000 [2C76] [UDP:192.168.1.96:35800] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (5400196) ble_cluster: BLE addr 34:f1:ae:d4:08:00
I (5400197) HAL_Rocker: HAL_Rocker::processEvent 7
I (5400197) HAL_Dimmer: HAL_Dimmer::processEvent
I (5400200) HAL_Rocker: Pow 206 206 Bri 1 1
I (5402430) ROUTE_HOOK: Received RIO
I (5402430) ROUTE_HOOK: prefix FD6F:48B9:FE1A:1:: lifetime 1800
I (5404885) ROUTE_HOOK: Received RIO
I (5404886) ROUTE_HOOK: prefix FD6F:48B9:FE1A:1:: lifetime 1800
I (5405182) ble_cluster: BLE addr 34:f1:ae:d4:08:00
I (5405183) HAL_Rocker: HAL_Rocker::processEvent 7
I (5405183) HAL_Dimmer: HAL_Dimmer::processEvent
I (5405187) HAL_Rocker: Pow 206 206 Bri 1 1
I (5405271) ble_cluster: BLE addr 34:f1:ae:d4:08:00
I (5405272) HAL_Rocker: HAL_Rocker::processEvent 7
I (5405272) HAL_Dimmer: HAL_Dimmer::processEvent
I (5405275) HAL_Rocker: Pow 206 206 Bri 1 1
I (5406060) app_main: Dump #270 Free internal 97948 biggest block 32740 total 97948 PSRAM free 1896072
Task: Pre-Scheduler allocs 0x0 -> MALLOC_CAP_INTERNAL: 3560 MALLOC_CAP_SPIRAM: 392 Highwater 0
Task: main 0x3fcb822c -> MALLOC_CAP_INTERNAL: 78972 MALLOC_CAP_SPIRAM: 40880 Highwater 4212
Task: lua 0x3fca4190 -> MALLOC_CAP_INTERNAL: 8372 MALLOC_CAP_SPIRAM: 37792 Highwater 35436
Task: tiT 0x3fccba90 -> MALLOC_CAP_INTERNAL: 520 MALLOC_CAP_SPIRAM: 452 Highwater 80
Task: wifi 0x3fcce748 -> MALLOC_CAP_INTERNAL: 18560 MALLOC_CAP_SPIRAM: 60144 Highwater 2824
Task: CHIP 0x3fcd4350 -> MALLOC_CAP_INTERNAL: 38492 MALLOC_CAP_SPIRAM: 22528 Highwater 448
Task: nimble_host 0x3fcdd738 -> MALLOC_CAP_INTERNAL: 36 MALLOC_CAP_SPIRAM: 448 Highwater 2320
Task: sys_evt 0x3fcca86c -> MALLOC_CAP_INTERNAL: 200 MALLOC_CAP_SPIRAM: 112 Highwater 1700
Invalid handle 0x580e
Task: ipc0 0x3fcb3ce0 -> MALLOC_CAP_INTERNAL: 24084 MALLOC_CAP_SPIRAM: 0 Highwater 736
Task: ipc1 0x3fcb445c -> MALLOC_CAP_INTERNAL: 32 MALLOC_CAP_SPIRAM: 0 Highwater 740
Task: btController 0x3fcdb22c -> MALLOC_CAP_INTERNAL: 7528 MALLOC_CAP_SPIRAM: 0 Highwater 1044
Task: esp_timer 0x3fcb5cb8 -> MALLOC_CAP_INTERNAL: 112 MALLOC_CAP_SPIRAM: 0 Highwater 2904
Task: mcast 0x3fca3f98 -> MALLOC_CAP_INTERNAL: 224 MALLOC_CAP_SPIRAM: 388 Highwater 844
Task: HAL 0x3fca3b54 -> MALLOC_CAP_INTERNAL: 112 MALLOC_CAP_SPIRAM: 4448 Highwater 856

I (5406115) ROUTE_HOOK: Received RIO
I (5406116) ROUTE_HOOK: prefix FD6F:48B9:FE1A:1:: lifetime 1800
I (5406347) chip[EM]: <<< [E:13726i S:28487 M:92478495] (S) Msg TX to 2:AB0DE00000000000 [2C76] [UDP:192.168.1.96:35800] --- Type 0001:05 (IM:ReportData)
I (5406425) chip[EM]: >>> [E:13726i S:28487 M:230503302 (Ack:92478495)] (S) Msg RX from 2:AB0DE00000000000 [2C76] --- Type 0001:01 (IM:StatusResponse)
I (5406426) chip[IM]: Received status response, status is 0x00
I (5406428) chip[EM]: <<< [E:13726i S:28487 M:92478496 (Ack:230503302)] (S) Msg TX to 2:AB0DE00000000000 [2C76] [UDP:192.168.1.96:35800] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (5407139) ROUTE_HOOK: Received RIO
I (5407140) ROUTE_HOOK: prefix FD6F:48B9:FE1A:1:: lifetime 1800
I (5410168) ble_cluster: BLE addr 34:f1:ae:d4:08:00
I (5410170) HAL_Rocker: HAL_Rocker::processEvent 7
I (5410170) HAL_Dimmer: HAL_Dimmer::processEvent
I (5410175) HAL_Rocker: Pow 206 206 Bri 1 1
I (5414308) ROUTE_HOOK: Received RIO
I (5414308) ROUTE_HOOK: prefix FD6F:48B9:FE1A:1:: lifetime 1800
I (5415155) ble_cluster: BLE addr 34:f1:ae:d4:08:00
[... 168 lines cut ...]

W (5465398) wifi:m f null

I (5465419) chip[SC]: SecureSession[0x3fca71c0, LSID:28487]: State change 'kActive' --> 'kDefunct'
E (5465419) chip[DMG]: Time out! failed to receive status response from Exchange: 13731i
I (5465423) chip[IM]: Schedule subscription resumption when failing to establish session, Retries: 3
W (5465454) wifi:m f null

W (5465500) wifi:m f null

W (5465557) wifi:m f null

W (5465603) wifi:m f null

W (5465659) wifi:m f null

W (5465705) wifi:m f null

W (5465761) wifi:m f null

W (5465807) wifi:m f null

W (5465864) wifi:m f null

~~~

jonsmirl, 2024-07-15:
This failure is intermittent and most of the time it resolves itself after a while.

nachiketkukade, 2024-07-16:
@jonsmirl , are you operating in a noisy environment? since there is no pattern to this issue, we'll need a some more information to debug this. Is it possible to capture a sniffer log for this issue using Wireshark?

jonsmirl, 2024-07-16:
I suspect I am in a noisy environment. The only pattern I consistently see is that there are always retransmit failures before getting into this condition.  Maybe the retransmit failures are exhausting the packet buffer pool? And then it recovers after some timer expires the packets are put back into the pool?

I have no idea how to trigger it. I typically have several windows open with debug logs scrolling by, and then I notice the wifi:m messages.

jonsmirl, 2024-07-16:
Maybe this is related? mdns allocs failed despite having a large amount of memory free. I am dynamically allocating my network buffers from PSRAM. 
~~~
I (5002) NimBLE: 

E (5007) mdns: Cannot allocate memory (line: 6186, free heap: 1987320 bytes)
E (5007) mdns: Cannot allocate memory (line: 6186, free heap: 1982984 bytes)
  PTR : AF67F99EAFC7087E-0294220C17BB322E._matter._tcp
  SRV : DC5475EFA5BC.local:5540
  PTR : 92BAA20D89A92C76-AB0DE00000000026._matter._tcp
  SRV : DC5475EFA5BC.local:5540
I (5009) device: Interface IP Address Changed

I (20842) app_main: Dump #1 Free internal 85460 biggest block 31724 total 85460 PSRAM free 1884696

~~~
this is where is failed:
[image]

jonsmirl, 2024-07-21:
It appears to me that there is some mechanism where allocations for wifi and mdns can fail despite having large amounts of RAM free. And whatever this condition is, it will clear itself after a while.

I don't know if this is a general failure in allocations since I do very few allocations in my code. Almost all of my memory is allocated at startup and doesn't change.

jonsmirl, 2024-07-22:
Could you make a CONFIG option to statically allocate the wifi buffers from PSRAM? I don't need dynamic wifi buffers, I am just using the dynamic option to force them into PSRAM.

jonsmirl, 2024-07-24:
I went right into the condition on boot....
then recovered
[matter.log](https://github.com/user-attachments/files/16368369/matter.log)

jonsmirl, 2024-07-26:
[matter.log](https://github.com/user-attachments/files/16395512/matter.log)
This failure had retransmits involved.  It was retransmitting because I restarted my Android app.
Same 3.5sec recovery period.

jonsmirl, 2024-07-26:
My theory is that packet buffer are being held in case of the need to retransmit. And then that results in me running out of packet buffers before the held buffers expire.