espressif/esp-matter #1002: How to add product_appearance attribute to the basic information cluster

FMistrorigo, 2024-07-12:
Hi, 
I wanted to add to my product the informations about color and finish, I found out this is possible in the attribute "product_appearance" of the cluster "basic information".
In my case the root node is created with the cluster basic information but the only attributes are node_label and cluster_revision, I know that using the function create_product_appearance it's possible to create the attribute.
[image]

I couldn't fins an example where this function is used, and in the documentation is not clear what parameters I have to provide in "value", "length" and "count".

What parameters do I have to provide if I want to display color: white and finish: satin when in the chip tool i use the command "./chip_tool basicinformation read product-appearance 1 0"?

Thanks

shripad621git, 2024-07-16:
@FMistrorigo , thanks for reporting the issue. The product appearance attribute can be set by creating a factory_partiton using `esp-matter-mfg-tool`. It is read using the DeviceInstanceInfoProvider implementation. We will add the support for the attribute soon.

shripad621git, 2024-07-29:
@FMistrorigo ,the change is now present in connectedhomeip [here](https://github.com/project-chip/connectedhomeip/pull/34363). 
The `product-appearanace` attribute is read using the platform specific [DeviceInstanceInfoProvider](https://github.com/project-chip/connectedhomeip/blob/5e3127f5ac61e13c572a968199280d90a9c19dce/src/platform/ESP32/ESP32FactoryDataProvider.cpp#L248)  implementation 
Till we update the submodule, you can apply do these changes accordingly in your connectedhomeip submodule and try out the product-appearance attribute in esp-matter.
For trying this out, you need to create factory partition containing the product-finish and product-color attributes.
For e.g:
```
  ./generate_esp32_chip_factory_bin.py -d 3434 -p 99663300 \
                                    --product-name ESP-lighting-app --product-id 0x8001 \
                                    --vendor-name Test-vendor --vendor-id 0xFFF2 \
                                    --hw-ver 1 --hw-ver-str DevKit  --product-finish polished --product-color navy
```
You can try out the factory script present [here](https://github.com/project-chip/connectedhomeip/blob/master/scripts/tools/generate_esp32_chip_factory_bin.py) with the changes in the above PR for generating the factory partition. 
You will have to enable the following options in the menuconfig for testing out the prodcut-appearance attribute.
```
CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER=y
CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER=y
```
For more information on factory data, you can refer this [link](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/factory_data.md).

shripad621git, 2024-08-07:
@FMistrorigo , any updates on the issue.

FMistrorigo, 2024-08-07:
Hi @shripad621git,

I didn't want to create a Factory partition only for these attributes, for the Device Instance Info Provider I use a custom one and I tried to implement the two functions in this way, so when the get functions were called the output was what I wanted, but it didn't work.
[image]

I decided to go back to the function create_product_appearance and did this:
[image]
(0x02 is Satin, and 0x0E is White)

Everything was good in the chip tool was working so I left it like this.

Tell me if ther's any problem in what I did.
Thanks.

shripad621git, 2024-08-28:
@FMistrorigo , the above changes LGTM if you don't want to use factory partition.

closed 2024-08-28 by FMistrorigo