espressif/esp-matter #1003: After generating the ./generate_esp32_chip_factory_bin.py  , there was no change in reading the product name using the chip tool

MaplestoryAlen, 2024-07-12:
Generate a new product name using the command (./generate_esp32_chichip_factory-bin. py). The read data still hasn't changed (ProductName: TEST PRODUCT)

According to the document, the operation is as follows：

./generate_esp32_chip_factory_bin.py -d 3026 -p 88661234 \
                                    --product-name ESPLighting --product-id 0x8000 \
                                    --vendor-name Testvendor --vendor-id 0xFFF1 \
                                    --hw-ver 1 --hw-ver-str DevKit \
                                    --dac-cert  /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-DAC-01.der  \ 
                                    --dac-key   /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-DAC-Key-01.der \
                                    --pai-cert   /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-PAI-Cert.der \
                                     --cd /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/esp_dac_fff1_8000.der

[image]

# Disable the DS Peripheral support
CONFIG_ESP_SECURE_CERT_DS_PERIPHERAL=n
# Use DAC Provider implementation which reads attestation data from secure cert partition
CONFIG_SEC_CERT_DAC_PROVIDER=y
# Enable some options which reads CD and other basic info from the factory partition
CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER=y
CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER=y
CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL="fctry"

esptool.py -p  /dev/ttyUSB0  write_flash 0xd000  /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/esp_secure_cert_data/esp_secure_cert.bin

esptool.py -p  /dev/ttyUSB0  write_flash   0xc20000   /home/alen/esp-matter/connectedhomeip/connectedhomeip/scripts/tools/bin/factory_partition.bin

[image]

 chip-tool  command:

/chip-tool basicinformation  read product-name 0x11 0

[image]

**Environment**
- ESP-Matter Commit Id: V1.3
- ESP-IDF Commit Id:V5.2.1
- SoC (eg: ESP32 or ESP32-C3): ESP32S3
- Device Logs (Please attach the log file):
- Host Machine OS:
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

shripad621git, 2024-07-12:
@MaplestoryAlen , just to confirm looking at the partition-table, are you using nvs_encryption feature.

MaplestoryAlen, 2024-07-15:
> @MaplestoryAlen , just to confirm looking at the partition-table, are you using nvs_encryption feature.

The NVS encryption function is turned off
[image]

shripad621git, 2024-07-16:
@MaplestoryAlen , can you please share the sdkconfig file. Seems like `CONFIG_DEVICE_INSTANCE_INFO_PROVIDER=y `is not set in sdkconfig. For information on factory data you can refer [here](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/factory_data.md).

MaplestoryAlen, 2024-07-16:
> CONFIG_DEVICE_INSTANCE_INFO_PROVIDER

The current sdk.config file has been overwritten by me, and I only found 2 matching options below. Do I want to choose custom?

CONFIG_EXAMPLE_DEVICE_INSTANCE_INFO_PROVIDER=y
# CONFIG_CUSTOM_DEVICE_INSTANCE_INFO_PROVIDER is not set

shripad621git, 2024-07-16:
Can you please enable ```CONFIG_FACTORY_DEVICE_INSTANCE_INFO_PROVIDER=y``` as done [here](https://github.com/espressif/esp-matter/blob/6833f8b087892c7ba9b16740a591db323c2d99a4/examples/generic_switch/sdkconfig.defaults#L46).

MaplestoryAlen, 2024-07-16:
> CONFIG_FACTORY_DEVICE_INSTANCE_INFO_PROVIDER
I have a new problem now, which is that I am unable to connect the network,this is the sdk.config file
[sdkconfig.txt](https://github.com/user-attachments/files/16244592/sdkconfig.txt)

chip tool log:

[chip-too-log.txt](https://github.com/user-attachments/files/16244681/chip-too-log.txt)

Generate factory_data_ partition command：

./generate_esp32_chip_factory_bin.py -d 3434 -p 99663300 \
    --product-name ESP-lighting-app --product-id 0x8000 \
    --vendor-name Test-vendor --vendor-id 0xFFF1 \
    --hw-ver 1 --hw-ver-str DevKit \
    --cd  /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/esp_dac_fff1_8000.der

../esp-matter/connectedhomeip/connectedhomeip/scripts/tools/bin （there are 2 files）

[factory_partition.bin.txt](https://github.com/user-attachments/files/16244709/factory_partition.bin.txt)

[onboarding_codes.csv](https://github.com/user-attachments/files/16244717/onboarding_codes.csv)
 

esp32s3 monitor log:

[image]

After writing factory data, the PINCODE in the log is still the default 20202021. Why isn't it 99663300 written by the command (generate_esp32uchip_factory-bin. py)?

shripad621git, 2024-07-16:
@MaplestoryAlen , if you are using `esp_secure_cert_partition` as I can see the `CONFIG_SEC_CERT_DAC_PROVIDER=y` in the sdkconfig, you will get a` AttestationVerification` error as it will fetch for `DAC and PAI certificates from esp_secure_cert partition.`
If you want to use the partiton, then you have to use the previous command
```
./generate_esp32_chip_factory_bin.py -d 3026 -p 88661234
--product-name ESPLighting --product-id 0x8000
--vendor-name Testvendor --vendor-id 0xFFF1
--hw-ver 1 --hw-ver-str DevKit
--dac-cert /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-DAC-01.der \
--dac-key /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-DAC-Key-01.der
--pai-cert /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-PAI-Cert.der
--cd /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/esp_dac_fff1_8000.der --dac-in-secure-cert --target esp32s3
```
with `--dac-in-secure-cert and --target `option at the end which will generate a `esp_secure_cert_parition. `
Please flash both the partitons i.e `factory and esp_secure_cert` at appropriate `addresses as specified in the partitions.csv` file.
If you want to use only  the `factory_partition`, please set the option `CONFIG_SEC_CERT_DAC_PROVIDER=n` using the command you specified without dac options and flash it at appropriate address specified in partitons.csv file.
For more information on providers, you can refer [here]( https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#factory-data-providers).

shripad621git, 2024-07-16:
> @MaplestoryAlen , if you are using `esp_secure_cert_partition` as I can see the `CONFIG_SEC_CERT_DAC_PROVIDER=y` in the sdkconfig, you will get a` AttestationVerification` error as it will fetch for [...]

If you have generated the `esp_secure_cert_partition` using the above command , please provide `--paa-trust-store-path` i,e " Path to directory holding PAA certificate information."  while commissioning with chip-tool. You will find this option
if you do  `./chip-tool pairing ble-wifi 0x12 ssid  password 88661234 3026 --help`
for e.g.: 
If you have generated the certs in the partition using chip-cert and if you have used the ```Chip-Development-PAA-Cert.pem```
In this case your paa-trust-store-path will be relatice path to the directory where this `PAA cert` is located.

MaplestoryAlen, 2024-07-17:
> Chip-Development-PAA-Cert.pem

I found the path to the Chip-Development-PAA-Cert. pem file (/home/alen/esp-matter/connectedhomeip/connectedhomeip/credentials/development/attestation/Chip-Development-PAA-Cert.pem)，
but when I send the chip-tool command, there are always errors, and I don't know where I went wrong

command:

./chip-tool pairing ble-wifi 0x12 Gerlsair_XizhenElec  AcSmartExp 88661234 3026  --paa-trust-store-path  /home/alen/esp-matter/connectedhomeip/connectedhomeip/credentials/development/attestation/Chip-Development-PAA-Cert.pem

or

./chip-tool pairing ble-wifi 0x12 Gerlsair_XizhenElec  AcSmartExp 88661234 3026  --paa-trust-store-path  paa/Chip-Development-PAA-Cert.pem/home/alen/esp-matter/connectedhomeip/connectedhomeip/credentials/development/attestation/

chip-tool log:

len@lee:~/esp-matter/connectedhomeip/connectedhomeip/out/host$ ./chip-tool pairing ble-wifi 0x12 Gerlsair_XizhenElec  AcSmartExp 88661234 3026  --paa-trust-store-path  paa/Chip-Development-PAA-Cert.pem/home/alen/esp-matter/connectedhomeip/connectedhomeip/credentials/development/attestation/
[1721176464.744833][297178:297178] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_tool_kvs
[1721176464.746644][297178:297178] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1721176464.746750][297178:297178] CHIP:DL: ChipLinuxStorage::Init: Attempt to re-initialize with KVS config file: /tmp/chip_kvs
[1721176464.747315][297178:297178] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1721176464.747569][297178:297178] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1721176464.747795][297178:297178] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1721176464.748187][297178:297178] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-VFIArr)
[1721176464.749161][297178:297178] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1721176464.749513][297178:297178] CHIP:DL: NVS set: chip-counters/reboot-count = 27 (0x1B)
[1721176464.750341][297178:297178] CHIP:DL: Got Ethernet interface: ens33
[1721176464.750933][297178:297178] CHIP:DL: Found the primary Ethernet interface:ens33
[1721176464.751789][297178:297178] CHIP:DL: Failed to get WiFi interface
[1721176464.751886][297178:297178] CHIP:DL: Failed to reset WiFi statistic counts
[1721176464.751974][297178:297178] CHIP:IN: UDP::Init bind&listen port=0
[1721176464.752184][297178:297178] CHIP:IN: UDP::Init bound to port=44347
[1721176464.752244][297178:297178] CHIP:IN: BLEBase::Init - setting/overriding transport
[1721176464.752304][297178:297178] CHIP:IN: TransportMgr initialized
[1721176464.752391][297178:297178] CHIP:FP: Initializing FabricTable from persistent storage
[1721176464.752590][297178:297178] CHIP:TS: Last Known Good Time: 2023-10-14T01:16:48
[1721176464.752946][297178:297178] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x7AA4A885B54CF480, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1721176464.755407][297178:297178] CHIP:ZCL: Using ZAP configuration...
[1721176464.756857][297178:297178] CHIP:CTL: System State Initialized...
[1721176464.756977][297178:297178] CHIP:TOO: No PAAs found in path: paa/Chip-Development-PAA-Cert.pem/home/alen/esp-matter/connectedhomeip/connectedhomeip/credentials/development/attestation/
[1721176464.757038][297178:297178] CHIP:TOO: Please specify a valid path containing trusted PAA certificates using the argument [--paa-trust-store-path paa/file/path] or environment variable [CHIPTOOL_PAA_TRUST_STORE_PATH=paa/file/path]
[1721176464.757119][297178:297178] CHIP:TOO: Run command failure: examples/chip-tool/commands/common/CHIPCommand.cpp:81: CHIP Error 0x0000002F: Invalid argument
[1721176464.764680][297178:297178] CHIP:SPT: VerifyOrDie failure at src/lib/support/Pool.h:392: Allocated() == 0
Aborted (core dumped)

PAA file path：
[image]

Operation method for generating document files：

1.2.2 Generating Certification Declaration
./chip-cert gen-cd -K ../../credentials/test/certification-declaration/Chip-Test-CD-Signing-Key.pem \
    -C ../../credentials/test/certification-declaration/Chip-Test-CD-Signing-Cert.pem \
    -O esp_dac_fff1_8000.der -f 1 \
    -V 0xfff1 -p 0x8000 -d 0x0016 -c "CSA00000SWC00000-01" -l 0 -i 0 -n 1 -t 0
1.2.3 Generating PAI
./chip-cert gen-att-cert -t i -c "ESP TEST PAI" -V 0xfff1 -P 0x8000 \
    -C ../../credentials/development/attestation/Chip-Development-PAA-Cert.pem \
    -K ../../credentials/development/attestation/Chip-Development-PAA-Key.pem \
    -o Esp-Development-PAI-Cert.pem -O Esp-Development-PAI-Key.pem -l 4294967295
1.2.4 Generating DAC
./chip-cert gen-att-cert -t d -c "ESP TEST DAC 01" -V 0xfff1 -P 0x8000 \
    -C Esp-Development-PAI-Cert.pem -K Esp-Development-PAI-Key.pem \
    -o Esp-Development-DAC-01.pem -O Esp-Development-DAC-Key-01.pem -l 4294967295
1.2.5 Change format for the certificates and key (.pem to .der format)
Convert DAC key from .pem to .der format.
openssl ec -in Esp-Development-DAC-Key-01.pem -out Esp-Development-DAC-Key-01.der -inform pem -outform der
Convert DAC and PAI cert from .pem to .der format
openssl x509 -in Esp-Development-DAC-01.pem -out Esp-Development-DAC-01.der -inform pem -outform der
openssl x509 -in Esp-Development-PAI-Cert.pem -out Esp-Development-PAI-Cert.der -inform pem -outform der

shripad621git, 2024-07-17:
@MaplestoryAlen , please give the path till directory containing the PAA certs.
```
./chip-tool pairing ble-wifi 0x12 Gerlsair_XizhenElec AcSmartExp 88661234 3026 --paa-trust-store-path /home/alen/esp-matter/connectedhomeip/connectedhomeip/credentials/development/attestation/
```
Give this path itself. No need to give the file name.
In the later case, 
```
./chip-tool pairing ble-wifi 0x12 Gerlsair_XizhenElec AcSmartExp 88661234 3026 --paa-trust-store-path /home/alen/esp-matter/connectedhomeip/connectedhomeip/credentials/development/attestation/Chip-Development-PAA-Cert.pem
```
you need to provide the path of the PAA cert directory relative to the current directory path.

You can try either of the two solutions to get the device commissioned.

MaplestoryAlen, 2024-07-18:
> `CONFIG_FACTORY_DEVICE_INSTANCE_INFO_PROVIDER=y`

The issue has been resolved and I will close it. Thank you very much for your guidance these past few days

closed 2024-07-18 by MaplestoryAlen