espressif/esp-matter #1007: Memory Leak Question

ronny-antoon, 2024-07-12:
**Describe the question/query that you have**
I made my own program using esp_matter, and separated it to components for better maintained and testing.
I tested all my component expect the component using esp_matter and their is no memory leak, the component using esp_matter i dont know how to test it.

Look at this analytics and tell me is this called memory leak?
and how to test the component that using esp_matter?

[image]

shubhamdp, 2024-07-16:
Hi @ronny-antoon could you please tell us more about that setup? 
- What commissioner(s) were used?
- Does the controller performs wild card read?
- Since how long the tests are running (I see these are a  day's samples)?
- How are you sampling free heap? Is it interval based (how long is the sampling interval) or triggered by specific event

ronny-antoon, 2024-07-16:
esp-idf 5.2.1
esp_matter 1.3.0 component
Apple Homepod and iphone

bluetooth commision and then wifi

`Does the controller performs wild card read?` i dont know what it is.

the test more than 3 days i can give you more data

```cpp
    addMetricToBuffer("freeHeap", (int) heap_caps_get_free_size(MALLOC_CAP_DEFAULT));
    addMetricToBuffer("minFreeHeap", (int) heap_caps_get_minimum_free_size(MALLOC_CAP_DEFAULT));
    addMetricToBuffer("largestFreeBlock", (int) heap_caps_get_largest_free_block(MALLOC_CAP_DEFAULT));
```

DejinChen, 2024-07-29:
Hi @ronny-antoon. Could you test the memory leak on idf v5.3? There was a fix for memory leak before on idf.

ronny-antoon, 2024-07-29:
> Hi @ronny-antoon. Could you test the memory leak on idf v5.3? There was a fix for memory leak before on idf.

I cant compile with idf v5.3

DejinChen, 2024-07-31:
> I cant compile with idf v5.3

Could you offer the compiling log?

ronny-antoon, 2024-07-31:
ESP_IDF 5.2.1 compiling good.

but ESP-IDF 5.3 i got error

```
 *  Executing task: C:\Users\RonnyPC\esptest53\.espressif\tools\cmake\3.24.0\bin\cmake.exe -G=Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B=c:\Users\RonnyPC\Documents\MatterApplication\build -S=c:\Users\RonnyPC\Documents\MatterApplication 

-- Found Git: C:/Users/RonnyPC/esptest53/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/RonnyPC/esptest53/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/RonnyPC/esptest53/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features       
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/RonnyPC/esptest53/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Dependencies lock doesn't exist, solving dependencies.
............................Updating lock file at C:\Users\RonnyPC\Documents\MatterApplication\dependencies.lock
Processing 20 dependencies:
[1/20] bblanchon/arduinojson (7.0.4)
[2/20] espressif/button (3.2.1)
[3/20] espressif/cbor (0.6.0~1)
[4/20] espressif/cmake_utilities (0.5.3)
[5/20] espressif/esp_diag_data_store (1.0.1)
[6/20] espressif/esp_diagnostics (1.0.2)
[7/20] espressif/esp_encrypted_img (2.1.0)
[8/20] espressif/esp_insights (1.0.1)
[9/20] espressif/esp_matter (1.3.0)
[10/20] espressif/esp_secure_cert_mgr (2.4.1)
[11/20] espressif/jsmn (1.1.0)
[12/20] espressif/json_generator (1.1.2)
[13/20] espressif/json_parser (1.0.3)
[14/20] espressif/mdns (1.3.2)
[15/20] espressif/rmaker_common (1.4.6)
[16/20] idf (5.3.0)
[17/20] ronny-antoon/accessorymodule (4.9.3)
[18/20] ronny-antoon/devicemodule (4.9.0)
[19/20] ronny-antoon/metricsmodule (5.0.0)
[20/20] ronny-antoon/relaybuttonmodule (4.5.0)
-- Found Python3: C:/Users/RonnyPC/AppData/Local/Programs/Python/Python312/python.exe (found version "3.12.4") found components: Interpreter 
-- Project sdkconfig file C:/Users/RonnyPC/Documents/MatterApplication/sdkconfig
warning: default on the choice symbol SEC_CERT_DAC_PROVIDER (defined at C:/Users/RonnyPC/Documents/MatterApplication/managed_components/espressif__esp_matter\connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:929, C:/Users/RonnyPC/Documents/MatterApplication/managed_components/espressif__esp_matter\components/esp_matter/Kconfig:48) will have no effect, as defaults do not affect choice symbols
warning: the choice symbol SEC_CERT_DAC_PROVIDER (defined at C:/Users/RonnyPC/Documents/MatterApplication/managed_components/espressif__esp_matter\connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:929, C:/Users/RonnyPC/Documents/MatterApplication/managed_components/espressif__esp_matter\components/esp_matter/Kconfig:48) is defined with a prompt outside the choice
Loading defaults file C:/Users/RonnyPC/Documents/MatterApplication/sdkconfig.defaults...
C:/Users/RonnyPC/Documents/MatterApplication/sdkconfig.defaults:56 CONFIG_NIMBLE_MAX_CONNECTIONS was replaced with CONFIG_BT_NIMBLE_MAX_CONNECTIONS
warning: unknown kconfig symbol 'RONNY_ANTOON' assigned to 'y' in C:/Users/RonnyPC/Documents/MatterApplication/sdkconfig.defaults
warning: unknown kconfig symbol 'ESPTOOLPY_BAUD_921600B' assigned to 'y' in C:/Users/RonnyPC/Documents/MatterApplication/sdkconfig.defaults
warning: unknown kconfig symbol 'ESPTOOLPY_BAUD' assigned to '921600' in C:/Users/RonnyPC/Documents/MatterApplication/sdkconfig.defaults
warning: unknown kconfig symbol 'ESPTOOLPY_COMPRESSED' assigned to 'y' in C:/Users/RonnyPC/Documents/MatterApplication/sdkconfig.defaults
warning: unknown kconfig symbol 'ESPTOOLPY_MONITOR_BAUD_151200B' assigned to 'y' in C:/Users/RonnyPC/Documents/MatterApplication/sdkconfig.defaults
-- Compiler supported targets: xtensa-esp-elf
-- Found Python3: C:/Users/RonnyPC/esptest53/.espressif/python_env/idf5.3_py3.11_env/Scripts/python.exe (found version "3.11.2") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "MatterApplication" version: 1.0
-- Adding linker script C:/Users/RonnyPC/Documents/MatterApplication/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/RonnyPC/Documents/MatterApplication/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script C:/Users/RonnyPC/esptest53/v5.3/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/RonnyPC/esptest53/v5.3/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Users/RonnyPC/esptest53/v5.3/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
[... 48 lines cut ...]
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
[... cut ...] /partition_table C:/Users/RonnyPC/esptest53/v5.3/esp-idf/components/soc C:/Users/RonnyPC/esptest53/v5.3/esp-idf/components/spi_flash C:/Users/RonnyPC/esptest53/v5.3/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/RonnyPC/Documents/MatterApplication/build/bootloader
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\Users\RonnyPC\esptest53\.espressif\tools\ninja\1.11.1\ninja.exe" terminated with exit code: 1. 
 ```

DejinChen, 2024-08-01:
@ronny-antoon  First you can try to use idf v5.3 to compile projects without esp_matter component like examples in idf. If this error still occurs, you can try to modify [line](https://github.com/espressif/esp-idf/blob/e0991facf5ecb362af6aac1fae972139eb38d2e4/components/newlib/platform_include/sys/lock.h#L26) to `int reserved[23];`. If you can compile other projects, but the project with esp_matter component fails, please let me know.

ronny-antoon, 2024-08-01:
Thank you, solved.

the problem occurs when adding this component 
https://github.com/ronny-antoon/DeviceModule

and the solution was to add CONFIG_FREERTOS_USE_TRACE_FACILITY=y in sdkconfig.defaults 

how i can add this "CONFIG_FREERTOS_USE_TRACE_FACILITY=y" in the component it self and not from the sdkconfig.defaults ??

ronny-antoon, 2024-08-01:
the compile solved not the Memory.

DejinChen, 2024-08-01:
> how i can add this "CONFIG_FREERTOS_USE_TRACE_FACILITY=y" in the component it self and not from the sdkconfig.defaults ??

Maybe you can add `select FREERTOS_USE_TRACE_FACILITY` in a config of which the default value is y.

ronny-antoon, 2024-08-12:
[image]

After some test, I think the main problem for memory leak is the WiFi. 
after a WiFi disconnect and then reconnet "at least one time" the memory leak starting but i dont know where and why.

DejinChen, 2024-08-14:
Could you offer the device logs and the reproduction method?

wqx6, 2024-09-13:
Could you check whether your IDF include this commit https://github.com/espressif/esp-idf/commit/88b300d0646c9e87df52377351a94e5e2da1fa9b which fixes a possible memory leak issue? If no, you can try to cherry-pick that commit or move your IDF to a branch or tag that includes that fix and try again? 

And one more question, are you using minimal mdns or platform mdns? If you are using platform mdns, please also check whether you mdns component include this fix https://github.com/espressif/esp-protocols/commit/042533af90ebb2d8f6cfde411eb365859536f57b which might also cause memory leak.

dhrishi, 2024-11-13:
@ronny-antoon Please check if the inputs help and close the issue if resolved

dhrishi, 2025-01-13:
Closing this now

closed 2025-01-13 by dhrishi