espressif/esp-matter #1008: Looking for the source code for the Launchpad example "thread_matter_light"

baylf2000, 2024-07-13:
**Describe the question/query that you have**
Trying to find the source code for the binary examples that are available on launchpad and here: [https://espressif.github.io/esp-matter/](url), specifically the "thread_matter_light" example.

Any other examples of matter over thread would also be great.

dhrishi, 2024-07-13:
@baylf2000 These are standard esp-matter examples. The `thread_matter_light` is light example compiled for ESP32-H2 and ESP32-C6 (choosing Matter over Thread from the config options)

closed 2024-07-13 by dhrishi

baylf2000, 2024-07-21:
@dhrishi Thank you for the answer, but it's not entirely helpful. There are hundreds of configuration options for each example, and it's obvious that the `thread_matter_light` example changes more than just the single option for thread. At the very least it would be greatly helpful if the config file for that example could be supplied somewhere.

dhrishi, 2024-07-21:
Please see [here](https://github.com/espressif/esp-matter/blob/main/examples/light/sdkconfig.defaults.c6_thread) the sdkconfig file present in the light example. Let me know if this helps

baylf2000, 2024-07-25:
@dhrishi Thank you. Is there some information on how to use this sdkconfig file? When selecting the "esp32c6" device target the extension uses the "sdkconfig.defaults.esp32c6" config file. What is the correct way to work around this?

baylf2000, 2024-07-25:
@dhrishi I replaced the contents of `sdkconfig.defaults.esp32c6` with the contents of `sdkconfig.defaults.c6_thread`, but the light example now fails to build under both 5.2 and 5.3, with the following message:

```/home/lf/Documents/esp_projects/light/build/esp-idf/chip/../../../../../../.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLS.cpp:299:(.text._ZN4chip6Crypto8HKDF_sha11HKDF_SHA256EPKhjS3_jS3_jPhj+0x68): undefined reference to `mbedtls_hkdf'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.```

shubhamdp, 2024-07-25:
Please find details about custom sdkconfig defaults usage here: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html#custom-sdkconfig-defaults

In-a-nutshell, you can use it like `idf.py -DSDKCONFIG_DEFAULTS=<your sdkconfig files separated by ;> build`

For error, `undefined reference to `mbedtls_hkdf'` , you are missing the following config from your sdkconfig.defaults.
```
CONFIG_MBEDTLS_HKDF_C=y
```

baylf2000, 2024-07-26:
@shubhamdp Thank you, however when using that command with the matter `light` example, the build process fails as follows:

```/home/lf/Documents/esp_projects_5.3/light/managed_components/espressif__esp_bsp_generic/src/esp_bsp_generic.c:251:41: error: 'bsp_leds_rgb_config' undeclared here (not in a function); did you mean 'bsp_leds_config'?
  251 |         .led_indicator_strips_config = &bsp_leds_rgb_config,
      |                                         ^~~~~~~~~~~~~~~~~~~
      |                                         bsp_leds_config
/home/lf/Documents/esp_projects_5.3/light/managed_components/espressif__esp_bsp_generic/src/esp_bsp_generic.c:249:5: warning: excess elements in array initializer
  249 |     {
      |     ^
/home/lf/Documents/esp_projects_5.3/light/managed_components/espressif__esp_bsp_generic/src/esp_bsp_generic.c:249:5: note: (near initialization for 'bsp_leds_config')
[186/477] c++ obj/third_party/connectedhomeip/src/platform/ESP32/ESP32.DiagnosticDataProviderImpl.cpp.o
../../../../../../.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/DiagnosticDataProviderImpl.cpp: In member function 'virtual CHIP_ERROR chip::DeviceLayer::DiagnosticDataProviderImpl::GetNetworkInterfaces(chip::DeviceLayer::NetworkInterface**)':
../../../../../../.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/DiagnosticDataProviderImpl.cpp:205:45: warning: 'esp_netif_t* esp_netif_next(esp_netif_t*)' is deprecated: use esp_netif_next_unsafe() either directly or via esp_netif_tcpip_exec [-Wdeprecated-declarations]
  205 |     esp_netif_t * netif     = esp_netif_next(NULL);
      |                               ~~~~~~~~~~~~~~^~~~~~
In file included from ../../../../../../.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/DiagnosticDataProviderImpl.cpp:35:
/home/lf/esp/v5.3/esp-idf/components/esp_netif/include/esp_netif.h:1020:14: note: declared here
 1020 | esp_netif_t *esp_netif_next(esp_netif_t *esp_netif)
      |              ^~~~~~~~~~~~~~
../../../../../../.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/DiagnosticDataProviderImpl.cpp:215:74: warning: 'esp_netif_t* esp_netif_next(esp_netif_t*)' is deprecated: use esp_netif_next_unsafe() either directly or via esp_netif_tcpip_exec [-Wdeprecated-declarations]
  215 |         for (esp_netif_t * ifa = netif; ifa != NULL; ifa = esp_netif_next(ifa))
      |                                                            ~~~~~~~~~~~~~~^~~~~
/home/lf/esp/v5.3/esp-idf/components/esp_netif/include/esp_netif.h:1020:14: note: declared here
 1020 | esp_netif_t *esp_netif_next(esp_netif_t *esp_netif)
      |              ^~~~~~~~~~~~~~
[193/477] c++ obj/third_party/connectedhomeip/src/platform/ESP32/ESP32.OpenthreadLauncher.cpp.o
../../../../../../.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/OpenthreadLauncher.cpp:106:18: warning: 'esp_err_t cli_command_transmit_task()' defined but not used [-Wunused-function]
  106 | static esp_err_t cli_command_transmit_task(void)
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~
../../../../../../.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/OpenthreadLauncher.cpp:71:13: warning: 'void esp_openthread_matter_cli_init()' defined but not used [-Wunused-function]
   71 | static void esp_openthread_matter_cli_init(void)
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[477/477] stamp obj/esp32.stamp
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/lf/Documents/esp_projects_5.3/light/build/log/idf_py_stderr_output_44123 and /home/lf/Documents/esp_projects_5.3/light/build/log/idf_py_stdout_output_44123```

baylf2000, 2024-07-26:
By changing then following in `sdkconfig.defaults.c6_thread` I was able to compile and flash the example.

```
# LEDs
CONFIG_BSP_LEDS_NUM=0
CONFIG_BSP_LED_TYPE_RGB=n
# CONFIG_BSP_LED_RGB_GPIO=8
# CONFIG_BSP_LED_RGB_BACKEND_RMT=y
```

However I am not able to commission the device. This is a different issue however. I'll open a separate bug report. Thank you for your help.