espressif/esp-matter #1009: Running out of space in nvs partition corrupts flash.

jonsmirl, 2024-07-13:
I had a bug in my code which caused me to write too many keys into the nvs partition. That should have just failed with an out-of-space error, but it didn't it corrupted my partition instead.

shripad621git, 2024-11-13:
@jonsmirl , an infinite lopp writing keys to the nvs_partition did not cause any corruption of the partition on my end.
I tested the light app as well as the ble_mesh_bridge app with bridged endppint with infinite loop writing too many keys to nvs with esp32c3. My idf version for this test is v5.2.1.
It gave the following error `4157` which stands for `ESP_ERR_NVS_NOT_ENOUGH_SPACE` which stands for ` no enough space in the underlying storage to save the value`.

jonsmirl, 2024-11-13:
Did you try to keep adding after it was full? I was not properly checking the error so I just kept adding until I rebooted.

shubhamdp, 2026-10-16:
Closing as stale, please re-open if still an issue

closed 2026-10-16 by shubhamdp