espressif/esp-matter #1010: Creating a Toggle Switch with On/Off Cluster and Binding for Group Control

mortezaMoezzifar, 2024-07-14:
#### Summary:
I am working on creating a toggle switch using the On/Off Cluster in Matter that supports binding and group functionality. The goal is to control a single light or a group of lights, with an LED on the switch that reflects the status of the light or group of lights.

#### Details:
- **Device**: Toggle switch with an LED indicator.
- **Functionality**: The switch should be able to turn a single light or a group of lights on and off. The LED indicator on the switch should reflect the current status (on/off) of the controlled light or group of lights.
- **Cluster**: On/Off Cluster
- **Command**: Using the Toggle command for control.

#### Question:
What is the best method to obtain the status of a light or a group of lights in Matter so that the LED indicator on my toggle switch can accurately reflect the on/off state of the controlled lights?

### Relevant Information:
- The On/Off Cluster documentation indicates that the status of a device can be obtained through attributes and commands. However, it is unclear what the most efficient approach is for a toggle switch that needs to update its LED based on the status of a potentially bound group of lights.

Thank you for your assistance!

jadhavrohit924, 2024-07-15:
@mortezaMoezzifar Please check out [light](https://github.com/espressif/esp-matter/tree/main/examples/light) and [light_switch](https://github.com/espressif/esp-matter/tree/main/examples/light_switch) where we have showed binding between light and switch. You can just group the lights and use the same switch to control it.

jonsmirl, 2024-07-15:
@jadhavrohit924  He wants an indicator light in the switch. Think about that and you'll see that it is a much larger problem than it appears. For example, if you group the lights in your app, how do the Apple/Amazon/Google apps control the grouped lights and keep them in sync?

jadhavrohit924, 2024-07-29:
@mortezaMoezzifar I understand your issue, what you can do is. You subscribe your generic switch to the light and whenever there is a change in your end accessory, the switch will know about it via matter subscription and you can then change the state of your LED on the switch.

jadhavrohit924, 2024-08-09:
@mortezaMoezzifar Have you tried the above suggestions? Please close the issue if it isresolved.

mortezaMoezzifar, 2024-08-21:
@jadhavrohit924 i need example code for subscribe on light to update state of led on my light switch

jadhavrohit924, 2024-08-27:
@mortezaMoezzifar Please checkout the code of [subscription](https://github.com/espressif/esp-matter/blob/main/components/esp_matter_controller/commands/esp_matter_controller_subscribe_command.cpp), here is the reference [example](https://github.com/espressif/esp-rainmaker/blob/52524f601b75fb11a4b89f1a87f1e950e9a9f153/examples/matter/matter_controller_with_touchscreen/main/app_matter_ctrl.cpp#L274)

closed 2024-09-02 by mortezaMoezzifar