espressif/esp-matter #1011: Setting the MSDS host name in CHIP

jonsmirl, 2024-07-15:
The MDNS host name is set here:
https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ESP32DnssdImpl.cpp#L164
~~~
    if (strcmp(service->mHostName, "") != 0)
    {
        VerifyOrExit(mdns_hostname_set(service->mHostName) == ESP_OK, error = CHIP_ERROR_INTERNAL);
    }
~~~
Is there some way I can pass in my own host names so that they don't get set to string like this? 1C53F9E795FE.local, DC5475EFA5BC.local, 98F4AB76D1E4.local, etc...

wqx6, 2024-07-15:
You can try to use the function `mdns_delegate_hostname_add` and `mdns_service_add_for_host` to add a new host and service. Note that you should obtain all the addresses of the network interface and pass the addresses when adding the host.

jonsmirl, 2024-07-16:
This seems to work...
~~~
        esp_netif_t *intf = esp_netif_get_default_netif();
        char *delegated_hostname;
        if (-1 == asprintf(&delegated_hostname, "lowpan-%d", nodeId)) {
            abort();
        }
        mdns_ip_addr_t addr4, addr6;
        addr4.addr.type = ESP_IPADDR_TYPE_V4;
        esp_netif_ip_info_t info;
        esp_netif_get_ip_info(intf, &info);
        addr4.addr.u_addr.ip4 = info.ip;
        addr6.addr.type = ESP_IPADDR_TYPE_V6;
        esp_netif_get_ip6_linklocal(intf, &addr6.addr.u_addr.ip6);
        addr4.next = &addr6;
        addr6.next = NULL;

        ESP_ERROR_CHECK( mdns_delegate_hostname_add(delegated_hostname, &addr4) );
        ESP_ERROR_CHECK( mdns_service_add_for_host(NULL, "_http", "_tcp", delegated_hostname, 80, NULL, 0) );
        free(delegated_hostname);
~~~

closed 2024-07-16 by jonsmirl

aggaddam, 2024-07-23:
@jonsmirl  I am also trying to add custom mdns without impacting the chip stack. I am not successful in setting up custom mdns services. Refer this issue #1022