espressif/esp-matter #1012: Missing function to create attribute TagList (TC-DESC-2.1)

FMistrorigo, 2024-07-15:
**Description**
I was doing the test case TC-DESC-2.1 and I noticed I was missing the attribute TagList of the cluster Descriptor in the root node.
I proceded to try to add the attribute with the usual function create but it didn't exist, in the file esp_matter_attribute the only functions aviable to add attributes to the cluster Descriptor are these:
[image]
None of the function add the attribute TagList with the id TagList::Id that's defined in Attributes.h
[image]

**Environment**
ESP-IDF Commit Id: a322e6bdad4b6675d4597fb2722eea2851ba88cb (HEAD, tag: v5.2.1)
ESP-Matter Commit Id: https://github.com/espressif/esp-matter/commit/1a35c738dff33e5e988d92d0028ed7a60d4336f6 (HEAD -> release/v1.2, origin/release/v1.2)
connectedhomeip Commit Id: 46e1ad8478c55ec20273da9d40a9c6fecf5c6b38
SoC: ESP32-S3
Host Machine OS: Ubuntu 22.04.3 (WSL)

shripad621git, 2024-07-16:
@FMistrorigo , the change is available on release/v1.3 [here.](https://github.com/espressif/esp-matter/commit/4c68c4e961468e08f5d2f22dd0776909d7746143) We will backport it soon to release/v1.2.

shripad621git, 2024-07-17:
@FMistrorigo, the `Taglist` feature is available on [release/v1.2](https://github.com/espressif/esp-matter/tree/release/v1.2) now.

closed 2024-07-17 by dhrishi