espressif/esp-matter #1014: is it possible to make a matter bridge with esp32 and control other devices with it ?

Jeodus, 2024-07-16:
for example: ı want to add my two light to a matter bridge and add others to another mather bridge and ı want to control them with bridges.

lights should be any company.

shubhamdp, 2024-07-23:
May be for general information about Matter bridge please check this [medium article](https://blog.espressif.com/matter-bridge-for-non-matter-devices-d3b7f003a004).

Back to your original question: Yes, it is possible to make Matter bridge with ESP32, few bridge examples here: [blemesh-bridge](https://github.com/espressif/esp-matter/tree/main/examples/blemesh_bridge), [esp-now-bridge](https://github.com/espressif/esp-matter/tree/main/examples/esp-now_bridge_light), and [zigbee-bridge](https://github.com/espressif/esp-matter/tree/main/examples/zigbee_bridge).

You will have to develop the firmware as per your Light, eg: if you have a blemesh light then you can start with blemesh-bridge, and start customising as per your need.

Or, not sure if you are looking for Matter Controller, which can control any Matter accessory(eg: any Matter enabled Light), then you can find that example here: https://github.com/espressif/esp-matter/tree/main/examples/controller

dhrishi, 2024-07-26:
Closing as the questions are answered and references provided

closed 2024-07-26 by dhrishi

Sarika1544, 2025-08-20:
@shubhamdp i have flased the blemesh_bridge code but after flashing that code in esp32s3 i am unable to do the commissioning with chip-tool can you please tell me what is issue ?

shubhamdp, 2025-08-20:
@Sarika1544 did you follow the steps outlined in https://github.com/espressif/esp-matter/blob/main/examples/bridge_apps/blemesh_bridge/README.md? Can you share the failures that you encountered? I'd suggest raising another issue with information that would help debug it.

Sarika1544, 2025-08-22:
> [@Sarika1544](https://github.com/Sarika1544) did you follow the steps outlined in https://github.com/espressif/esp-matter/blob/main/examples/bridge_apps/blemesh_bridge/README.md? Can you share the [...]

Hi @shubhamdp as you said i have raised another seperate issue [https://github.com/espressif/esp-matter/issues/1522](url) can you please help me to solve my issue?