espressif/esp-matter #1016: Error : "esp_matter_core: Node does not exist or esp_matter does not start"

Shivi-6, 2024-07-17:
After flashing the matter controller example to my ESP32-WROVER-E and monitoring, the following error appears 
E (2723) esp_matter_core: Node does not exist or esp_matter does not start
I (2733) chip[DL]: E (2733) app_main: Failed to start Matter, err:259
Done driving station state, nothing else to do...
I (2753) chip[DL]: HandlePlatformSpecificBLEEvent 49152
I (2763) chip[DL]: HandlePlatformSpecificBLEEvent 49152
I (2763) chip[DL]: WIFI_EVENT_STA_CONNECTED
I (2773) chip[DL]: WiFi station state change: Connecting -> Connecting_Succeeded
E (2783) chip[DMG]: Endpoint 0, Cluster 0x0000_0031 not found in IncreaseClusterDataVersion!
I (2783) wifi:E (2793) chip[DMG]: <ba-add>idx:0 (ifx:0, 9c:a2:f4:5c:a3:e7), tid:0, ssn:3, winSize:64Endpoint 0, Cluster 0x0000_0031 not found in IncreaseClusterDataVersion!

**Environment**
- ESP-Matter Commit Id: latest
- ESP-IDF Commit Id: Latest
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file):
[idf_py_stdout_output.pdf](https://github.com/user-attachments/files/16263544/idf_py_stdout_output.pdf)

- Host Machine OS: Ubuntu
- Commissioner app and versions if present: Alexa 2.2.575865.0

shripad621git, 2024-07-29:
Can you please give a try by setting `CONFIG_ESP_MATTER_ENABLE_MATTER_SERVER=n` set using `idf.py menuconfig`.

Shivi-6, 2024-07-30:
I tried the above solution with the latest IDF version and it generated some build error.  On changing to v5.3, this issue is resolved.

closed 2024-07-30 by Shivi-6