espressif/esp-matter #1018: API to get local fabricId/NodeId list?

jonsmirl, 2024-07-18:
It there an API to get the local fabricId/NodeId list?

wqx6, 2024-07-18:
You can try to iterate the Fabric Table of the server instance
```
FabricTable &fabricTable = Server::GetInstance().GetFabricTable();
for (auto iter = fabricTable.begin(); iter != fabricTable.end(); ++iter) {
       printf("FabricId 0x%llx NodeId 0x%llx", iter->GetFabricId(), iter->GetNodeId());
}
```

jonsmirl, 2024-07-21:
Example of how to use this....
~~~
void Cluster_WebServer::enable_host(bool enable)
{
    char host_name[32] = "";

    chip::FabricTable &fabricTable = chip::Server::GetInstance().GetFabricTable();
    for (auto iter = fabricTable.begin(); iter != fabricTable.end(); ++iter)
    {
        printf("FabricId 0x%llx NodeId 0x%llx", iter->GetFabricId(), iter->GetNodeId());
        uint64_t node_id = iter->GetNodeId();
        // find node ids which begin with 0xAB0DE
        // pick off the id number and use it for the websever MDNS name 'lowpan-xxx.local'
        if (iter->GetVendorId() == LOWPAN_VENDOR_ID)
        {
            uint64_t node_id = iter->GetNodeId() && 0xFFFFFFFF;
            snprintf(host_name, sizeof(host_name), "lowpan-%llx", node_id);
            esp_netif_t *intf = esp_netif_get_default_netif();
            mdns_ip_addr_t addr4, addr6;
            addr4.addr.type = ESP_IPADDR_TYPE_V4;
            esp_netif_ip_info_t info;
            esp_netif_get_ip_info(intf, &info);
            addr4.addr.u_addr.ip4 = info.ip;
            addr6.addr.type = ESP_IPADDR_TYPE_V6;
            esp_netif_get_ip6_linklocal(intf, &addr6.addr.u_addr.ip6);
            addr4.next = &addr6;
            addr6.next = NULL;

            // create MDNS record for the web server at 'lowpan-xxx.local'
            ESP_LOGI(TAG, "adding host_name");
            ESP_ERROR_CHECK(mdns_delegate_hostname_add(host_name, &addr4));
            ESP_ERROR_CHECK( mdns_service_add_for_host("web", "_http", "_tcp", host_name, 80, NULL, 0) );
            strncat(host_name, ".local", sizeof(host_name) - strlen(host_name) - 1);
            break;
        }
    }

    ESP_LOGI(TAG, "host name is %s", host_name);
    endpoint_t *root = endpoint::get(device.getNode(), 0);
    cluster_t *cluster = esp_matter::cluster::get(root, cluster_webserver::Id);
    /* Create custom hostname attribute */
    attribute::create(cluster, cluster_webserver::attribute::hostname::Id, ATTRIBUTE_FLAG_WRITABLE, esp_matter_char_str(host_name, strlen(host_name)));
}
~~~

closed 2024-07-21 by jonsmirl