espressif/esp-matter #1019: PICS Tool certification related information (matter V1.3)

MaplestoryAlen, 2024-07-19:
I am now in the certification stage and a member of the CSA Alliance, but I don't know how to make the SDK program generate files for PICS Tool to test. May I ask if there are any relevant processes or guidance documents for users to use PICS Tool for preliminary testing? I have read the Matter Test Harris User Manual on the CSA official website, which introduces that it is for Raspberry Pi and the chip is Nordic, but I am certified for ESP32S3, so many of them are not applicable. Can you give me some guidance and usage process?

leonardbay, 2024-07-21:
Same issue, ESP-IDF "Matter Certification" mentions [PICS files](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/certification.html#pics-files) but not how to generate or find them, and their link to sample files requires login

VaishaliAvhale, 2024-07-23:
Hi @MaplestoryAlen,

The PICS Tool is used for generating the PICS (Protocol Implementation Conformance Statement) as per the application. You can get this PICS from the CSA. Check this link: [CSA Matter CSG Documents](https://groups.csa-iot.org/wg/matter-csg/document/folder/2756).

Here you will find:

1. Verification Document: Contains Test Cases (TC). You need to execute all applicable TCs from the first sheet of the verification document.
2. PICS_XML_File:
  Steps to Update PICS using PICS tool:
  a. Download and unzip the PICS tool zip.
  b. Open the PICS tool in a browser.
  c. Upload the applicable cluster's PICS_XML file to the PICS tool. Select the present attributes, commands, events, and features for the specific cluster and click on the Generate button. Repeat this for all applicable clusters' PICS. You need to upload this PICS while executing TCs using the Test Harness (TH). For more information, check the Matter TH User Guide.
3. Matter TH User Guide: Refer to this for setting up the Raspberry Pi/TH, executing different types of Test Cases, and for uploading TCs execution logs on TEDS.

marcorussi, 2024-07-31:
Hello, I am in the process of Matter certification test and I am struggling to select the right PICS files. How do I know which option of each cluster should I include? I know about the clusters I use in my application such as ON/OFF and Level but I do not know about all the other ones as they were already implemented in the Espressif samples which I based my project on (for example commissioning and base information). Can anyone help please?

VaishaliAvhale, 2024-08-01:
There are two types of clusters present on device type - `Application Specific Clusters` and  `Core Clusters`. `Core clusters`  are present on `Root Node[endpoint 0]`. `Descriptor Cluster` is an exception because it is present on all endpoints. For selecting Core Cluster PICS please check `2.1 Root Node` Device Type in Device Library Specification and select all Mandatory cluster.

MaplestoryAlen, 2024-08-06:
> Hi @MaplestoryAlen,
>
> The PICS Tool is used for generating the PICS (Protocol Implementation Conformance Statement) as per the application. You can get this PICS from the CSA. Check this link: [CSA [...]

The problem has been resolved. Thank you very much

closed 2024-08-06 by MaplestoryAlen

Haerteleric, 2024-10-02:
could you clarify what the applicable cluster's PICS_XML are ? 

i can only find cluster test plan xml's inside the zip