espressif/esp-matter #1020: Matter Door Lock Server events

bilalmalik76, 2024-07-19:
How to implement the following Matter door lock mandatory server events using esp-matter sdk: 

- DoorLockAlarm even
- DoorStateChange event
- LockOperation event
- LockOperationError event
- LockUserChange event

According to Matter harness test above are the mandatory attributes.

shubhamdp, 2024-07-23:
@bilalmalik76 when you create a door lock cluster, then these events will be added to the data model. From your list only 3 are mandatory:
1. DoorLockAlarm
2. LockOperation
3. LockOperationError

DoorStateChange depends on door position change (DPS) feature. LockUserChange depends on user(USR) feature.

For these two you will have to add these features to your data model. Please check the [programming guide](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#features) for how to add features.

dhrishi, 2024-07-26:
@bilalmalik76 Will you please give it a try and provide an update?

dhrishi, 2024-08-08:
Closing it now. Re-open in case of any questions

closed 2024-08-08 by dhrishi