espressif/esp-matter #1021: Failure in ninja while installing SDK

Mr-Alirezaa, 2024-07-20:
**Describe the bug**
I have cloned esp-matter sdk with the following command and I'm sure that nothing went wrong during the cloning process:

```
git clone \
    --branch release/v1.2 \
    --depth 1 \
    --shallow-submodules \
    --recursive https://github.com/espressif/esp-matter.git
```

After that I tried to install the sdk with `./install.sh` and it failed with the following log. I can also confirm that I have the prerequisites installed as per [this document](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html#for-macos-users)

[esp-matter-installation.log](https://github.com/user-attachments/files/16322155/esp-matter-installation.log)

last few lines of the above log:
```
...

  File "/Users/a.asadi/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 573, in setup
    result = step(spin)
             ^^^^^^^^^^
  File "/Users/a.asadi/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 762, in virtualenv
    if not virtualenv_setup.install(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/a.asadi/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 443, in install
    install_packages(gn_target)
  File "/Users/a.asadi/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 424, in install_packages
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['ninja', '-C', '/Users/a.asadi/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.
```

**Environment**
- ESP-Matter Commit Id: d692395263fe6cd254b262366635d6b7e6cdad0c (release/v1.2)
- ESP-IDF Commit Id:  a322e6bdad4b6675d4597fb2722eea2851ba88cb (v5.1.2)
- SoC (eg: ESP32 or ESP32-C3): N/A
- Device Logs (Please attach the log file):
- Host Machine OS: MacOS Sonoma 14.5 (23F79)
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

shubhamdp, 2024-07-23:
@Mr-Alirezaa can you please share the python version details on you system.

Also, before running `esp-matter/install.sh` you should ideally source `esp-idf/export.sh`. That step is missing in the swift's documentation. Can you please try this?
Please remove the `esp-matter/connectedhomeip/connectedhomeip/.environment` directory before running `./install.sh`. More detailed steps can be found in esp-matter [programming guide](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#id1).

shubhamdp, 2024-07-23:
@Mr-Alirezaa Can you please try this patch, I see some problem with `pyyaml` and `construct` packages.
```
cd connectedhomeip/connectedhomeip
git apply change_python_pkg_version.patch
rm -rf .environment
cd ../../
source esp-idf/export.sh
./install.sh
```

[change_python_pkg_version.patch](https://github.com/user-attachments/files/16344358/change_python_pkg_version.patch)

Mr-Alirezaa, 2024-07-23:
Hi @shubhamdp, thank you for your response.
> @Mr-Alirezaa can you please share the python version details on you system.

```
$ python3 --version
Python 3.12.4

# Also

$ source ~/esp/esp-idf/export.sh
Setting IDF_PATH to '/Users/a.asadi/esp/esp-idf'
Detecting the Python interpreter
Checking "python3" ...
Python 3.12.4
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Constraint file: /Users/a.asadi/.espressif/espidf.constraints.v5.2.txt
Requirement files:
 - /Users/a.asadi/esp/esp-idf/tools/requirements/requirements.core.txt
Python being checked: /Users/a.asadi/.espressif/python_env/idf5.2_py3.12_env/bin/python
Python requirements are satisfied.
Updated PATH variable:
  /Users/a.asadi/esp/esp-idf/components/espcoredump:/Users/a.asadi/esp/esp-idf/components/partition_table:/Users/a.asadi/esp/esp-idf/components/app_update:/Users/a.asadi/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/Users/a.asadi/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/Users/a.asadi/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/Users/a.asadi/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/Users/a.asadi/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/Users/a.asadi/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/Users/a.asadi/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/Users/a.asadi/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/Users/a.asadi/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/Users/a.asadi/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/Users/a.asadi/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/Users/a.asadi/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/Users/a.asadi/.espressif/python_env/idf5.2_py3.12_env/bin:/Users/a.asadi/esp/esp-idf/tools:/Users/a.asadi/.sdkman/candidates/java/current/bin:/Users/a.asadi/.modular/pkg/packages.modular.com_mojo/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/Library/Apple/usr/bin:/Users/a.asadi/.mint/bin
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
```

I also have tested by uninstalling 3.12 and installing 3.11. both were installed via homebrew.

> Also, before running `esp-matter/install.sh` you should ideally source `esp-idf/export.sh`. That step is missing in the swift's documentation. Can you please try this? 

I have tested installation in both ways, once by using Swift's document and once using the [official espressif document](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#esp-matter-setup). I have tested both with and without sourcing the `esp-idf/export.sh`, result is no different.

> Please remove the `esp-matter/connectedhomeip/connectedhomeip/.environment` directory before running `./install.sh`. More detailed steps can be found in esp-matter [programming guide](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#id1).

In my first few tries, I did't delete the `.environment` folder but then I came across the section of the document you mentioned, then tried installing after removing `.environment` folder.

I have also tested passing --no-host-tool to the install script.

```sh
$ ./install.sh --no-host-tool
```

shubhamdp, 2024-07-23:
@Mr-Alirezaa have you tried with the patch?

Mr-Alirezaa, 2024-07-23:
> @Mr-Alirezaa Can you please try this patch, I see some problem with `pyyaml` and `construct` packages.
>
> ```
> cd connectedhomeip/connectedhomeip
> git apply change_python_pkg_version.patch
> rm -rf [...]

This WORKED! Thanks a lot! 🙏

> @Mr-Alirezaa have you tried with the patch?

Yes, now I tried and it passed the step. 

```
Detecting the Python interpreter
Checking "python3" ...
Python 3.12.4
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Constraint file: /Users/a.asadi/.espressif/espidf.constraints.v5.2.txt
Requirement files:
 - /Users/a.asadi/esp/esp-idf/tools/requirements/requirements.core.txt
Python being checked: /Users/a.asadi/.espressif/python_env/idf5.2_py3.12_env/bin/python
Python requirements are satisfied.
Updated PATH variable:
  /Users/a.asadi/esp/esp-idf/components/espcoredump:/Users/a.asadi/esp/esp-idf/components/partition_table:/Users/a.asadi/esp/esp-idf/components/app_update:/Users/a.asadi/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/Users/a.asadi/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/Users/a.asadi/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/Users/a.asadi/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/Users/a.asadi/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/Users/a.asadi/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/Users/a.asadi/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/Users/a.asadi/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/Users/a.asadi/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/Users/a.asadi/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/Users/a.asadi/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/Users/a.asadi/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/Users/a.asadi/esp/esp-idf/components/espcoredump:/Users/a.asadi/esp/esp-idf/components/partition_table:/Users/a.asadi/esp/esp-idf/components/app_update:/Users/a.asadi/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/Users/a.asadi/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/Users/a.asadi/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/Users/a.asadi/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/Users/a.asadi/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/Users/a.asadi/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/Users/a.asadi/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/Users/a.asadi/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/Users/a.asadi/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/Users/a.asadi/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/Users/a.asadi/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/Users/a.asadi/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/Users/a.asadi/.espressif/python_env/idf5.2_py3.12_env/bin:/Users/a.asadi/esp/esp-idf/tools:/Users/a.asadi/.sdkman/candidates/java/current/bin:/Users/a.asadi/.modular/pkg/packages.modular.com_mojo/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/Library/Apple/usr/bin:/Users/a.asadi/.mint/bin
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Running Matter Setup

Synchronizing submodule url for 'examples/common/QRCode/repo'
Synchronizing submodule url for 'examples/common/m5stack-tft/repo'
Synchronizing submodule url for 'third_party/asr/asr550x/asr_sdk'
Synchronizing submodule url for 'third_party/asr/asr582x/asr_sdk'
Synchronizing submodule url for 'third_party/asr/asr595x/asr_sdk'
Synchronizing submodule url for 'third_party/asr/components'
Synchronizing submodule url for 'third_party/boringssl/repo/src'
Synchronizing submodule url for 'third_party/bouffalolab/repo'
Synchronizing submodule url for 'third_party/bouffalolab/repo/components'
Synchronizing submodule url for 'third_party/bouffalolab/repo/customer_app'
Synchronizing submodule url for 'third_party/bouffalolab/repo/docs'
Synchronizing submodule url for 'third_party/bouffalolab/repo/tools'
Synchronizing submodule url for 'third_party/cirque/repo'
Synchronizing submodule url for 'third_party/editline/repo'
Synchronizing submodule url for 'third_party/freertos/repo'
Synchronizing submodule url for 'third_party/imgui/repo'
Synchronizing submodule url for 'third_party/infineon/cyw30739_sdk/repos/30739A0'
Synchronizing submodule url for 'third_party/infineon/cyw30739_sdk/repos/CYW930739M2EVB-01'
Synchronizing submodule url for 'third_party/infineon/cyw30739_sdk/repos/btsdk-include'
Synchronizing submodule url for 'third_party/infineon/cyw30739_sdk/repos/btsdk-tools'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/TARGET_CY8CKIT-062S2-43012'
[... cut ...]
  Using cached MarkupSafe-2.1.5-cp312-cp312-macosx_10_9_universal2.whl.metadata (3.0 kB)
Downloading lark-1.1.2-py2.py3-none-any.whl (104 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 104.7/104.7 kB 373.3 kB/s eta 0:00:00
Using cached Jinja2-3.0.1-py3-none-any.whl (133 kB)
Using cached MarkupSafe-2.1.5-cp312-cp312-macosx_10_9_universal2.whl (18 kB)
Installing collected packages: stringcase, lark, MarkupSafe, jinja2
Successfully installed MarkupSafe-2.1.5 jinja2-3.0.1 lark-1.1.2 stringcase-1.2.0
All done! You can now run:

  . ./export.sh
```

shubhamdp, 2024-07-23:
@Mr-Alirezaa Thanks for the confirmation, I'll raise the fix in connectedhomeip repo and then update the submodule for v1.2

Mr-Alirezaa, 2024-07-23:
@shubhamdp Never mind.

So I'll let you close the issue if you want to close it now, or after the fix is done.

shubhamdp, 2024-07-24:
But, I still wonder, why our CI's don't see the problem.

shubhamdp, 2024-07-24:
I have raised the PR https://github.com/project-chip/connectedhomeip/pull/34477

shubhamdp, 2024-07-29:
@Mr-Alirezaa Fix is merged in release/v1.2 and this issue can be closed

closed 2024-07-29 by dhrishi