espressif/esp-matter #1022: Unable to add custom mDNS services after esp-matter SDK connects to WiFi network

aggaddam, 2024-07-23:
I am unable to add custom mDNS services on top of the esp-matter SDK which is running on the device post connecting to a Wi-Fi network during the matter commissioning process. The Connected Home IP stack also inits mDNS, adds services for _matterc/_matter services for device discovery.

ESP's mDNS library is used by the Connected Home IP stack, which is accessible from the ESP32 port layer. Once the ESP-Matter SDK connects to Wi-Fi during the commissioning process, it initializes mDNS and starts its services _matterc/_matter for device discovery. I want to add custom mDNS services along with the _matterc/_matter service types. Is it possible to add custom mDNS services when the matter device is commissioned and connected to the home WiFi network from the application layer? I tried it but was unable to achieve it. Any suggestions for implementation?

Here is the sample code I am calling under the event "chip::DeviceLayer::DeviceEventType::kDnssdInitialized:"

```
            mdns_txt_item_t serviceTxtData[] = {
                                                    {"u","user"},
                                                    {"p","password"}
                                                };

            esp_err = mdns_service_add(srvc_name,
                                        srvc_type,
                                        srvc_proto,
                                        srvc_port_num,
                                        serviceTxtData, N_ELEMENTS(serviceTxtData));
            ESP_ERROR_CHECK(esp_err);

            if ((ESP_OK != esp_err) && (ESP_ERR_NO_MEM != esp_err))
            {
                // The mdns_service_add will return an error if we try to add an existing service
                esp_err = mdns_service_txt_set(srvc_type,
                                               srvc_proto,
                                               serviceTxtData, N_ELEMENTS(serviceTxtData));
                ESP_ERROR_CHECK(esp_err);
            }
```

Note: I was able to initialize mDNS and add services successfully without the Connected Home IP running on the device.

Additional context
...

jonsmirl, 2024-07-23:
There is working code in this issue: https://github.com/espressif/esp-matter/issues/1018

aggaddam, 2024-07-24:
@jonsmirl  I tried the logic you shared, but I am getting an error as an invalid state when mdns_delegate_hostname_add() is called

jonsmirl, 2024-07-24:
You can only run that code once in your app. If it runs a second time the MDNS entry already exists and it will error. So I incremented a static on the IPv4 and IPv6 names and then only called when it was 2.

jonsmirl, 2024-07-24:
Your device already has a MDNS name. I created a second one which is was more human friendly.
~~~
jonsmirl@ares:~$ avahi-browse -r -t  _matter._tcp 
= enp3s0f1 IPv6 C7944E937AA07DD5-0083161AE5399201             _matter._tcp         local
   hostname = [000000000000.local]
   address = [192.168.1.4]
   port = [5541]
   txt = ["T=1"]
= enp3s0f1 IPv6 C7944E937AA07DD5-008462EE7296AD71             _matter._tcp         local
   hostname = [000000000000.local]
   address = [192.168.1.4]
   port = [5541]
   txt = ["T=1"]
= enp3s0f1 IPv6 950545243CFE2D15-00000000FE0D396C             _matter._tcp         local
   hostname = [1C53F9E795FE.local]
   address = [192.168.1.53]
   port = [5540]
   txt = []
= enp3s0f1 IPv6 AF67F99EAFC7087E-57784A802FC2C15C             _matter._tcp         local
   hostname = [DC5475EFA5BC.local]
   address = [192.168.1.3]
   port = [5540]
   txt = []
= enp3s0f1 IPv6 C66F15BC7C50AF35-AB0DE00000000002             _matter._tcp         local
   hostname = [DC5475EFA5BC.local]
   address = [192.168.1.3]
   port = [5540]
   txt = []
~~~
I then made a second, more friendly name for 192.168.1.3. The delegate name is an alias.
~~~
jonsmirl@ares:~$ avahi-browse -r -t  _http._tcp
= enp3s0f1 IPv6 web                                           Web Site             local
   hostname = [lowpan-2.local]
   address = [192.168.1.3]
   port = [80]
   txt = []
= enp3s0f1 IPv4 web                                           Web Site             local
   hostname = [lowpan-2.local]
   address = [192.168.1.3]
   port = [80]
   txt = []
~~~
~~~

jonsmirl, 2024-07-24:
I should also mention that adding a HTTP server to a Matter device completely ruins the security model. In my case it is only used for debugging and it is not enabled by default.  You have to enable/disable it from the device settings page in my phone app. Unless you build an entire login system anyone can get to it.

aggaddam, 2024-07-24:
@jonsmirl  I am getting the same error as an invalid state when mdns_delegate_hostname_add() is called once post after fetching ipv4 and ipv6 addresses by the chip stack. I can see chip mdns services running ok when queried from Avahi.. I am using ESP-Matter SDK v1.2. Are you using the latest ESP_matter SDK ?

jonsmirl, 2024-07-24:
Those messages from chip can repeat later when IP address leases expire. So count the number of messages and only call your routine when the count == 2.  Or just add a static flag in the routine so that it can only be called once.

aggaddam, 2024-07-24:
@jonsmirl I am using the static flag to call the function only once. I verified from the debug logs as well it was called only once.

jonsmirl, 2024-07-24:
Note that the messages from CHIP can come in on multiple threads, and also mdns_add_service can block.
This will work:
~~~
static bool flag = true;
if (flag) {
  flag = false;
   mdns_add_service() /// can block
}
~~~
this will fail because two threads can get inside.
~~~
static bool flag = true;
if (flag) {
   mdns_add_service() /// can block
  flag = false;
}
~~~

aggaddam, 2024-07-25:
@jonsmirl I have already taken care of your suggestions. 
I am seeing the delegate hostname itself mdns_delegate_hostname_add()  is failing with an invalid state before adding any service using mdns_add_service()

albrownwood, 2024-08-17:
@aggaddam disabling USE_MINIMAL_MDNS in menuconfig solves the invalid state error in my case.