espressif/esp-matter #1023: CMake Error When Enabling ESP Insights in ESP-Rainmaker Matter Light Example Build

ShahilDholariya, 2024-07-24:
**Describe the bug**
A clear and concise description of what the bug is.

I use combine esp-rainmaker and esp-matter.  
When building the ESP-Rainmaker matter_light example, the project builds correctly. However, when enabling ESP Insights in menuconfig, the build fails with the following CMake error:

esp insights project commit: cfdd71e cmake error at /home/dnk067/esp/esp-idf/tools/cmake/component.cmake:381 (__component_get_property): __component_get_property function invoked with incorrect arguments for function named: __component_get_property call stack (most recent call first): /home/dnk067/esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/cmakelists.txt:471 (idf_component_get_property)

This is my Cmake file,
set(EXTRA_COMPONENT_DIRS
    "${MATTER_SDK_PATH}/config/esp32/components"
    "${ESP_MATTER_PATH}/components" 
    "${ESP_MATTER_PATH}/device_hal/device"
    "${ESP_MATTER_PATH}/examples/common"
    "${RMAKER_PATH}/examples/common/app_insights"
    "${RMAKER_PATH}/components"
    ${extra_components_dirs_append})

**Environment**
- ESP-Matter Commit Id:  release 1.0
- ESP-IDF Commit Id: 4.4.3
- SoC (eg: ESP32 or ESP32-C3): ESP32S3
- Device Logs (Please attach the log file):

- Host Machine OS: Ubuntu 22.04
- Host Machine Python version: 3.10

- Commissioner app and versions if present:
- Home hub app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

shubhamdp, 2024-07-24:
@ShahilDholariya Can you exactly specify the commit ID for esp-matter, I checked the release/v1.0 branch, which points to connectedhomeip submodule [4088a77f557e8571a39338fad51a1d8eb0131d79](https://github.com/espressif/connectedhomeip/blob/4088a77f557e8571a39338fad51a1d8eb0131d79), and do not have 471 lines in [CMakeLists.txt](https://github.com/espressif/connectedhomeip/blob/4088a77f557e8571a39338fad51a1d8eb0131d79/config/esp32/components/chip/CMakeLists.txt)

So, actual commit Ids would help debug the problem faster.

shubhamdp, 2024-07-24:
Also, Matter v1.0 has been obsolete, and you cannot certify the new devices for v1.0.

ShahilDholariya, 2024-07-24:
The commit IDs for my ESP-Matter setup are as follows:

ESP-Matter Commit ID: ab58e2993d268844e9ab64af586681e59b659ffd
connectedhomeip submodule Commit ID: ab58e2993d268844e9ab64af586681e59b659ffd
These are the commit IDs currently in use in my environment.

ShahilDholariya, 2024-07-24:
[image]

This is my Cmake file Screenshot that's help you to find error at 471 line.

shubhamdp, 2024-07-24:
@ShahilDholariya This is not v1.0, you are using latest main branch, the commit you are using 3 weeks back.

Anyways, try this patch and see if this works
```
diff --git a/config/esp32/components/chip/CMakeLists.txt b/config/esp32/components/chip/CMakeLists.txt
index d0c1fc4463..5a3274c464 100644
--- a/config/esp32/components/chip/CMakeLists.txt
+++ b/config/esp32/components/chip/CMakeLists.txt
@@ -472,7 +472,15 @@ if (CONFIG_SEC_CERT_DAC_PROVIDER)
 endif()

 if (CONFIG_ENABLE_ESP_INSIGHTS_TRACE)
-    idf_component_get_property(esp_insights_lib espressif__esp_insights COMPONENT_LIB)
+    idf_build_get_property(build_components BUILD_COMPONENTS)
+    # esp_insights can be used as an independent component or through component manager so,
+    # We should check and add the right component.
+    if("espressif__esp_insights" IN_LIST build_components)
+        idf_component_get_property(esp_insights_lib espressif__esp_insights COMPONENT_LIB)
+    elseif("mdns" IN_LIST build_components)
+        idf_component_get_property(esp_insights_lib esp_insights COMPONENT_LIB)
+    endif()
+
     list(APPEND chip_libraries $<TARGET_FILE:${esp_insights_lib}>)
 endif()
```

shubhamdp, 2024-07-24:
I have raised a PR to fix it, we will update the submodule once it gets merged. https://github.com/project-chip/connectedhomeip/pull/34472

shubhamdp, 2024-07-29:
Hi @ShahilDholariya we merged the fix in release/v1.3. Please try on that branch.

ShahilDholariya, 2024-08-05:
Thank you ,

I make this change and now my build work correctly with ESP-Insights but when i go to my esp insights dashboard and add package than it not shows error messages in correct form.
I tried custom error messages logged using ESP_LOGE in my ESP32 application are not displayed correctly on the ESP Insights dashboard. Instead of my custom messages, the dashboard shows messages indicating the location of a string in the read-only data section (ROData) along with the program counter (PC).

for (int i = 0; i < 10; ++i) {
        ESP_LOGE(TAG, "Error code 2: Custom error message.");
        vTaskDelay(1000 / portTICK_PERIOD_MS); // Wait 1 second between reports
    }
this is my custom error message for testing purpose but on dashboard it shows like,

05/08/24 11:25:27
app_main: string at ROData: 0x3c134e16 PC: 0x4200ae61

ERROR
05/08/24 11:25:28
app_main: string at ROData: 0x3c134e16 PC: 0x4200ae61

10 times So, why on dashboard it not print error into its real form.

This is My Firmware package details ,it might be helpful 

Name: MATTER_LIGHT

Package Id: 19de7450-4c18-4cc6-8441-cfffbfe8a15c

Type: l1

Model:

Firmware Version: e123e56-dirty

Firmware SHA: 586f58a28cb855f8

...

closed 2024-08-31 by ShahilDholariya