espressif/esp-matter #1024: Matter-Zigbee Bridge implementation using ESP Zigbee Gateway devkit(ESP32S3 + ESP32H2)

Shivi-6, 2024-07-24:
**Describe the question/query that you have**
Using USB-to-C cable to connect ESP32H2, flashed ot_rcp example from esp-idf/examples/openthread
Removed cable from ESP32H2 and connected to ESP32S3, flashed Zigbee_bridge example from esp-matter/examples

On monitoring while connected to ESP32S3 usb port using **idf.py -p /dev/ttyACM0 monitor**, I am getting the error **E (1552) ZB_ESP_PLATFORM: esp_zb_platform_config(55): host mode uart not supported**

I am a total newbie to this so I don't know if I am doing this right. Please let me know what is the issue and how can I resolve this and what are the next steps to configure this devkit as a zigbee bridge.

DejinChen, 2024-07-26:
Could you check the `host_connection_mode` in app_zboos.h? It should be `ZB_HOST_CONNECTION_MODE_NONE` in the latest esp-matter example.

Shivi-6, 2024-07-26:
Hi @DejinChen , I checked app_zboss.h. host_connection_mode is set to ZB_HOST_CONNECTION_MODE_NONE

DejinChen, 2024-07-29:
Hi @Shivi-6 Are your idf on master branch? If so, may checkout idf release/v5.3 or other branch before.

Shivi-6, 2024-07-29:
@DejinChen Thanks for your response. with v5.3, the error is now resolved.

See log :
I(7837) OPENTHREAD:[I] P-SpinelDrive-: co-processor reset: RESET_POWER_ON
E(7837) OPENTHREAD:[C] P-SpinelDrive-: Software reset co-processor successfully
[0W(7857) OPENTHREAD:n[W] P-RadioSpinel-: Error processing result: NotImplemented
W(7857) OPENTHREAD:[W] P-RadioSpinel-: Error waiting response: NotImplemented
I (7857) ZB_ESP_SPINEL: Radio spinel workflow register successfully
I (7877) esp_zboss: status: -1
I (7877) esp_zboss: Zigbee stack initialized
W(7887) OPENTHREAD:[W] P-RadioSpinel-: Error processing result: NotImplemented
W(7887) OPENTHREAD:[W] P-RadioSpinel-: Error waiting response: NotImplemented
I (7907) esp_zboss: status: 0
I (7907) esp_zboss: Start network formation
I (7907) esp_zboss: Formed network successfully
I (7907) esp_zboss: ieee extended address: 74:4d:bd:ff:fe:61:7c:3d, PAN ID: 0xbb0d)
I (8087) wifi:<ba-add>idx:0 (ifx:0, 9c:a2:f4:5c:a3:e7), tid:0, ssn:14, winSize:64
I (8497) esp_zboss: status: 0
I (8497) esp_zboss: Network steering started

Now I am not sure what the next step is.. In Readme, it directly talks about post commissioning but how do I commission the bridge?

DejinChen, 2024-07-29:
Follwing [commissioning](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning-and-control) with chip-tool, or you can commission it with Apple Home or other matter ecosystem.

Shivi-6, 2024-07-30:
Thanks to your help, I was able to realize matter bridge.

closed 2024-07-30 by Shivi-6

tsiura, 2024-10-08:
Have same issue with 5.3.1 and esp_zigbee_gateway examples. All code by default.
`SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0x403c8914
I (26) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (26) boot: compile time Oct  8 2024 17:17:35
I (26) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 0015e000
I (87) boot:  3 zb_storage       Unknown data     01 81 0016e000 00004000
I (94) boot:  4 zb_fct           Unknown data     01 81 00172000 00000400
I (101) boot:  5 rcp_fw           Unknown data     01 82 00173000 0007d000
I (109) boot: End of partition table
I (113) esp_image: segment 0: paddr=00010020 vaddr=3c0c0020 size=26aach (158380) map
I (150) esp_image: segment 1: paddr=00036ad4 vaddr=3fc9a000 size=04944h ( 18756) load
I (155) esp_image: segment 2: paddr=0003b420 vaddr=40374000 size=04bf8h ( 19448) load
I (160) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=bee88h (781960) map
I (305) esp_image: segment 4: paddr=000feeb0 vaddr=40378bf8 size=1134ch ( 70476) load
I (330) boot: Loaded app from partition at offset 0x10000
I (330) boot: Disabling RNG early entropy source...
I (342) cpu_start: Multicore app
I (351) cpu_start: Pro cpu start user code
I (351) cpu_start: cpu freq: 160000000 Hz
I (351) app_init: Application information:
I (354) app_init: Project name:     EspZbGateway
I (359) app_init: App version:      1
I (364) app_init: Compile time:     Oct  8 2024 17:17:29
I (370) app_init: ELF file SHA256:  466fa220c...
I (375) app_init: ESP-IDF:          v5.3.1-dirty
I (380) efuse_init: Min chip rev:     v0.0
I (385) efuse_init: Max chip rev:     v0.99 
I (390) efuse_init: Chip rev:         v0.1
I (395) heap_init: Initializing. RAM available for dynamic allocation:
I (402) heap_init: At 3FCAA650 len 0003F0C0 (252 KiB): RAM
I (408) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (414) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (420) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (428) spi_flash: detected chip: generic
I (431) spi_flash: flash io: dio
I (436) sleep: Configure to isolate all GPIO pins in sleep state
I (442) sleep: Enable automatic switching of GPIO sleep configuration
I (450) main_task: Started on CPU0
I (460) main_task: Calling app_main()
E (460) ZB_ESP_PLATFORM: esp_zb_platform_config(55): host mode uart not supported
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4200c69e
--- 0x4200c69e: app_main at /home/user/Projects/EspZbGateway/main/esp_zigbee_gateway.c:157 (discriminator 1)
`
Tried to set host_config.connection_mode to NONE - no success

DejinChen, 2024-10-10:
Could you test it with idf v5.3 tag?

DejinChen, 2024-10-10:
Could you test it with master branch?

tsiura, 2024-10-13:
So, i tried to pull latest release idf (recognized as 5.4.0), build both - zigbee_rcp and zigbee_gateway, flash H2 SoC with RCP firmware, i got such error:
SPIWP:0xee
mode:DIO, clock div:1
load:0x408460e0,len:0x1894
load:0x4083cad0,len:0xef4
load:0x4083efd0,len:0x2d7c
entry 0x4083cada
I (23) boot: ESP-IDF v5.4-dev-3489-g6e5a178b31 2nd stage bootloader
I (24) boot: compile time Oct 13 2024 11:21:02
I (25) boot: chip revision: v0.1
I (27) boot: efuse block revision: v0.0
I (32) boot.esp32h2: SPI Speed      : 64MHz
I (37) boot.esp32h2: SPI Mode       : DIO
I (41) boot.esp32h2: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00089000
I (85) boot:  3 zb_storage       Unknown data     01 81 0009a000 00004000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=42040020 size=0d218h ( 53784) map
I (119) esp_image: segment 1: paddr=0001d240 vaddr=40800000 size=02dd8h ( 11736) load
I (123) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=33300h (209664) map
I (177) esp_image: segment 3: paddr=00053328 vaddr=40802dd8 size=0a990h ( 43408) load
I (191) esp_image: segment 4: paddr=0005dcc0 vaddr=4080d770 size=01bb8h (  7096) load
I (198) boot: Loaded app from partition at offset 0x10000
I (199) boot: Disabling RNG early entropy source...
I (211) cpu_start: Unicore app
I (220) cpu_start: Pro cpu start user code
I (221) cpu_start: cpu freq: 96000000 Hz
I (221) app_init: Application information:
I (223) app_init: Project name:     EspZbRcp2
I (228) app_init: App version:      1
I (233) app_init: Compile time:     Oct 13 2024 11:20:34
I (239) app_init: ELF file SHA256:  26df2834d...
I (244) app_init: ESP-IDF:          v5.4-dev-3489-g6e5a178b31
I (250) efuse_init: Min chip rev:     v0.0
I (255) efuse_init: Max chip rev:     v0.99
I (260) efuse_init: Chip rev:         v0.1
I (265) heap_init: Initializing. RAM available for dynamic allocation:
I (272) heap_init: At 40815C00 len 00037780 (221 KiB): RAM
I (278) heap_init: At 4084D380 len 00002B60 (10 KiB): RAM
I (285) spi_flash: detected chip: generic
I (289) spi_flash: flash io: dio
I (293) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (300) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (308) main_task: Started on CPU0
I (308) main_task: Calling app_main()
E (318) ZB_ESP_MACSPLIT_UART: mode is choosen but which host connection mode is not choosen

abort() was called at PC 0x4200fa9f on core 0
--- 0x4200fa9f: zb_assert at ??:?
and bootloop.
then i tried to flash H2 with 5.3.0 firmware - looks like it works properly.
Then i flashed S3 chip with 5.4.0 zigbee_gateway firmware and tried to start, thats give me such error:
I (468) main_task: Started on CPU0
I (478) main_task: Calling app_main()
E (478) ZB_ESP_PLATFORM: esp_zb_platform_config(55): host mode uart not supported
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4200cf99
--- Error: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))
--- Waiting for the device to reconnect--- 0x4200cf99: app_main at C:/Projects/esp32/EspZbGateway/main/esp_zigbee_gateway.c:155 (discriminator 1)