espressif/esp-matter #1027: Unable to commission matter over thread light example.

baylf2000, 2024-07-26:
**Describe the bug**
I have not been able to commission the light example using matter over thread. I've tried multiple pathways, however I'm primarily using the Home Assistant OTBR and their Android commissioning app. I've also tried using the Android Home app directly.

The error I get in the commissioner app is `Can't connect to thread network xyz. Check that your device can work with this network type and try again.`

Firmware is built in vscode with the current esp-idf extension. ESP-IDF versions 5.2 and 5.3 were both tried. ESP-Matter was installed in vscode with the `Install ESP-MATTER` option in the ESP-IDF extension.

**Environment**
- ESP-Matter Commit Id: 
- ESP-IDF Commit Id:
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6
- Device Logs (Please attach the log file):

Device Log: (Boot to failure) - [esp32c6_matter_thread_light.txt](https://github.com/user-attachments/files/16390494/esp32c6_matter_thread_light.txt)

- Host Machine OS: Ubuntu 22.04
- Host Machine Python version: 3.10.12

- Commissioner app and versions if present: Home Assistant Android App, Android Home App
- Commissioner's logs if present:

Some of the commissioner logs: 

**Any additional details**
...

wqx6, 2024-08-27:
Looks like the esp32c6 was not attached to the thread network of the OTBR you were using. It became a leader at last. Could you enable the CONFIG_OPENTHREAD_CLI on esp32c6 and input this command in the esp32c6's console after the thread credential is set during commissioning to verify that the esp32c6 receives the same thread dataset as the OTBR was set?
```
matter esp ot_cli dataset active -x
```

dhrishi, 2024-11-13:
@baylf2000 Any update you can share? Please close the issue if resolved

dhrishi, 2024-12-23:
Closing this now

closed 2024-12-23 by dhrishi