espressif/esp-matter #1030: How can ı use aggregator endpoint to bind it 2 or more device

Jeodus, 2024-07-29:
I’m very new to the Matter protocol.
 I’m using the Google Home app to control devices. I want to create an aggregator node and bind two or more different devices to it 
ı want to see all the devices collectively under the aggregator.

can someone lead me the way like how can ı begin. thanks  

I am a student and I want to improve myself in this field.

jadhavrohit924, 2024-08-01:
@Jeodus Please look into the [blemesh bridge](https://github.com/espressif/esp-matter/tree/main/examples/blemesh_bridge) example.

Jeodus, 2024-08-01:
[image]
thank you but my devices list like this ı want to saw my light under sw named bridge.

Jeodus, 2024-08-01:
[image]
Thank you but my devices list like this. ı want to see my light device under sw named bridge.

jadhavrohit924, 2024-08-02:
It should be under sw named bridge. Which commissioner are you using?

Jeodus, 2024-08-02:
google home

shubhamdp, 2024-08-08:
blemesh_bridge in esp-matter only works with on_off blemesh model, If you'd like please check the [readme](https://github.com/espressif/esp-matter/tree/main/examples/blemesh_bridge#ble-mesh-bridge) and give a try with blemesh on-off example from esp-idf.

shubhamdp, 2024-12-10:
@Jeodus any update on this?

dhrishi, 2024-12-23:
Closing this now

closed 2024-12-23 by dhrishi