espressif/esp-matter #1031: How to perform commissioning of Controller

Shivi-6, 2024-07-30:
**Describe the question/query that you have**
After flashing and monitoring the controller example onto my esp32-wrover-e kit, I am trying to commission the controller and connect it to Alexa but the procedure in https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#matter-controller is not clear. How do I commission the controller and the end devices? 

**Additional context**
I tried the following :

**matter esp controller pairing onnetwork 0x7283 20202021**

but this is the response I get :
**Done
> E (263002) chip[DIS]: Timeout waiting for mDNS resolution.**

jonsmirl, 2024-07-31:
In general you don't commission controllers, you commission devices.   You are trying to hook two controllers together. Controllers send commands, devices receive them. It would make more sense if your controller commissioned a light bulb, and Alexa commissioned the same light bulb and then you both controlled it.

Shivi-6, 2024-08-06:
Thanks for this clarification. But in the rainmaker matter controller example, the readme says **_"The commissioning flow for Matter Controller would be the same as other Matter devices. The controller will get node NOC (Node Operational Certificate) during Matter commissioning process."_** 
Based on this, I tried commissioning the matter controller after removing amazon echo dot so that only one controller is added and not two.

Shivi-6, 2024-08-08:
@dhrishi This makes things a bit clear. 
I needed a controller so I tried it with rainmaker and I was able to commission it and I could also add end devices from Rainmaker app and then control them from the app as well as the console.
I also understood that the controller example in esp-matter is similar to chip-tool and can be used to commission end devices from the console. But the process of commissioning the end devices from the console is not clear. I am trying to commission a matter enabled TP-link smart plug using the manual pairing code but I am not able to do so. Can you please help me with that?

shubhamdp, 2024-08-08:
@Shivi-6 can you please share the logs from the esp-matter controller?

Shivi-6, 2024-08-09:
Hi @shubhamdp , logs of when the controller connects to Rainmaker?
I have three logs,
1. The controller is commissioned but cannot be connected to Rainmaker due to some timeout (this is sporadic) 
[Controller_Fail.pdf](https://github.com/user-attachments/files/16555984/Controller_Fail.pdf)

2. The controller is commissioned and connected successfully to Rainmaker but not able to retrieve end device information
[Controller_Success.pdf](https://github.com/user-attachments/files/16555990/Controller_Success.pdf)

3. The controller cannot retrieve node list but is able to control the nodes (two end devices - smart plug and dimmer) 
[Device_Control.pdf](https://github.com/user-attachments/files/16556026/Device_Control.pdf)

shubhamdp, 2024-08-17:
@Shivi-6, in `Device_Control.pdf` I only see you are trying invoke command, to commission the end device from there you will have to use the pairing command. Few examples are listed in our programming guide, PTAL: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#pairing-commands

PS: You will have to connect the commissioner to the WiFi before starting the process and that can be done using `matter esp wifi connect <ssid> <password>` command

dhrishi, 2024-08-27:
@Shivi-6 Following the above comment should help you commission the end device successfully. Please give it a try and close the issue if it works

Shivi-6, 2024-08-29:
I tried commissioning the end device but it shows "Timeout waiting for mDNS resolution."

```
[1724922684.284315][1865989:1865989] CHIP:TOO: Command: pairing onnetwork 0x1234 20202021 
[1724922684.302672][1865989:1865991] CHIP:CTL: Setting attestation nonce to random value
[1724922684.302732][1865989:1865991] CHIP:CTL: Setting CSR nonce to random value
[1724922684.314490][1865989:1865991] CHIP:CTL: Generating NOC
[1724922684.342115][1865989:1865991] CHIP:FP: Validating NOC chain
[1724922684.342586][1865989:1865991] CHIP:FP: NOC chain validation successful
[1724922684.342680][1865989:1865991] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1724922684.342687][1865989:1865991] CHIP:TS: Last Known Good Time: 2023-10-14T01:16:48
[1724922684.342709][1865989:1865991] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1724922684.342713][1865989:1865991] CHIP:TS: Retaining current Last Known Good Time
[1724922684.343646][1865989:1865991] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1724922684.344196][1865989:1865991] CHIP:TS: Committing Last Known Good Time to storage: 2023-10-14T01:16:48
[1724922684.344772][1865989:1865991] CHIP:CTL: Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: 765ADAD98A0FB07A)
[1724922684.344781][1865989:1865991] CHIP:DIS: Updating services using commissioning mode 0
[1724922684.345136][1865989:1865991] CHIP:DIS: CHIP minimal mDNS started advertising.
[1724922684.345737][1865989:1865991] CHIP:DL: Using WiFi MAC for hostname
[1724922684.345761][1865989:1865991] CHIP:DIS: Advertise operational node 765ADAD98A0FB07A-000000000001B669
[1724922684.345790][1865989:1865991] CHIP:DIS: Responding with _matter._tcp.local
[1724922684.345815][1865989:1865991] CHIP:DIS: Responding with 765ADAD98A0FB07A-000000000001B669._matter._tcp.local
[1724922684.345821][1865989:1865991] CHIP:DIS: Responding with 765ADAD98A0FB07A-000000000001B669._matter._tcp.local
[1724922684.345843][1865989:1865991] CHIP:DIS: Responding with 1866DA4686A50000.local
[1724922684.345865][1865989:1865991] CHIP:DIS: Responding with 1866DA4686A50000.local
[1724922684.345872][1865989:1865991] CHIP:DIS: Responding with _I765ADAD98A0FB07A._sub._matter._tcp.local
[1724922684.345895][1865989:1865991] CHIP:DIS: CHIP minimal mDNS configured as 'Operational device'; instance name: 765ADAD98A0FB07A-000000000001B669.
[1724922684.346365][1865989:1865991] CHIP:DIS: mDNS service published: _matter._tcp
[1724922715.388353][1865989:1865991] CHIP:DIS: Timeout waiting for mDNS resolution.
```

shubhamdp, 2024-12-10:
@Shivi-6 are you still seeing the commissioning failure? If yes, can you please share the complete logs.

dhrishi, 2025-01-13:
Closing this now

closed 2025-01-13 by dhrishi