espressif/esp-matter #1032: Error while commissioning Thread devices using ESP-Matter controller

vishnuprasadvp12345, 2024-07-31:
Hi, I am currently working with the ESP-Matter controller to commission Thread devices. I used an ESP32-S3 and an ESP32-C6 to create a Thread network and obtained the following dataset:
0e080000000000010000000300000d35060004001fffe0020825b17d758cb8bef50708fd3e01901fefdef40510af5f0dfd6832477b597f8c70401c9917030f4f70656e5468726561642d666163380102fac80410fb35542f8c4d36bb0758c47769606f710c0402a0f7f8
I used this dataset in the CHIP tool to successfully commission my end devices. However, when I tried to use it in my ESP controller code with the following command:
matter esp controller pairing ble-thread 0x02 0x0e080000000000010000000300000d35060004001fffe0020825b17d758cb8bef50708fd3e01901fefdef40510af5f0dfd6832477b597f8c70401c9917030f4f70656e5468726561642d666163380102fac80410fb35542f8c4d36bb0758c47769606f710c0402a0f7f8 20202021 3840
I encountered the following error:
Error: 83886338
I suspect the format of the data being sent through the console is incorrect. Can anyone help me arrange the command in the correct format?

wqx6, 2024-08-08:
Could you increase the console buffer length `CHIP_SHELL_CMD_LINE_BUF_MAX_LENGTH` in menuconfig and try again?

dhrishi, 2024-08-27:
@vishnuprasadvp12345 Any update you can share? Please close the issue if it works

vishnuprasadvp12345, 2024-08-27:
ok

vishnuprasadvp12345, 2024-08-29:
Error: dataset_tlvs data too large  i change  to 512 no use

closed 2024-08-29 by vishnuprasadvp12345