espressif/esp-matter #1033: The light example automatically changes brightness after being turned off.

shaohuayuan, 2024-08-01:
**Describe the bug**
A clear and concise description of what the bug is.

**Environment**
- ESP-Matter Commit Id:commit ba5b64b0f714e768154c27e8a579c47dc19bcdf0 
- ESP-IDF Commit Id:commit a322e6bdad4b6675d4597fb2722eea2851ba88cb (HEAD, tag: v5.2.1)
- SoC (eg: ESP32 or ESP32-C3):ESP32-C3
- Device Logs (Please attach the log file):
[I (30) boot: ESP-IDF v5.2.log](https://github.com/user-attachments/files/16449462/I.30.boot.ESP-IDF.v5.2.log)

- Host Machine OS:macos 14.5 
- Host Machine Python version:Python 3.9.6 

- Commissioner app and versions if present:axela
- Home hub app and versions if present:

https://github.com/user-attachments/assets/d94be729-6f40-46d1-bcb5-36fbb24fe0f1

- Commissioner's logs if present:

**Any additional details**
...
“I compiled and downloaded the light example to the development board. After setting the brightness to 50% and turning off the light, the brightness automatically changes to 100% when the light is turned on again after a few seconds.”

shaohuayuan, 2024-08-02:
I (83304) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000008's Attribute 0x00000000 is 254 **********

 Every time turning on the light automatically set Cluster id 0x00000008, Attribute id 0x00000000 to 254,How to fix it?

jadhavrohit924, 2024-08-02:
@shaohuayuan We are aware of this issue, and we will fix it soon.

eiSouthBoy, 2025-03-19:
Hello, has this problem been fixed?
I also encountered the same problem. When I turned off the Dimmable through the UI, the brightness on the UI was displayed as 1%, then it immediately became 30%, and the light was also on and did not turn off.

dhrishi, 2025-05-15:
@shaohuayuan Please close the issue if resolved

jadhavrohit924, 2025-07-25:
Can we close the issue?

closed 2025-08-13 by dhrishi