espressif/esp-matter #1034: How to make 2 accessories, one with "Single lightbulb" and one with "Three Plugins

ronny-antoon, 2024-08-01:
**Describe the question/query that you have**
A clear and concise description of what you want to ask.

**Additional context**
...

jadhavrohit924, 2024-08-06:
@ronny-antoon It's not possible to create more than one node on the same device. You have to create multiple endpoints on the same node only.

ronny-antoon, 2024-08-09:
its only in esp_matter or the Matter Spec says that only one node per physical device

jadhavrohit924, 2024-08-14:
This is an esp-matter's limitation: we cannot create more than one node per device.

ronny-antoon, 2024-08-16:
> @ronny-antoon It's not possible to create more than one node on the same device. You have to create multiple endpoints on the same node only.

Is that a Matter Specification limitation, or just esp_matter!?

ronny-antoon, 2024-08-16:
> This is an esp-matter's limitation: we cannot create more than one node per device.

Is there any plan to include support for multiple nodes per device in future updates?

jadhavrohit924, 2024-08-26:
Sorry, not as of now!!

closed 2026-04-07 by shubhamdp