espressif/esp-matter #1035: Thread Sleepy End Device help

pavel808, 2024-08-01:
Hello all,

I have an ESP32-C6 battery-powered device which works as Thread end device. 

I am now exploring the Thread Sleepy Device capabilities (SED or SSED) to preserve battery life. 

My device is quite simple. It receives commands over Matter / Thread from time to time from a client / user (non periodically - it could be once every few seconds, or once every few hours). The device on reception of a command then sends data out on a GPIO pin to control another device.

I believe SSED would suit the best in this case?

How can I simply enable this in my build? I don't see options for this via `idf.py menuconfig` . 

I see the flag in the code `CHIP_DEVICE_CONFIG_ENABLE_SSED` . Would it simply be enough to have this enabled? Or should other parameters be configured? Unfortunately I don't see a proper example on how to use this.

Thanks so much in advance for any help or guidelines.

wqx6, 2024-08-06:
You should call `ConnectivityMgr().SetThreadDeviceType(ConnectivityManager::kThreadDeviceType_SynchronizedSleepyEndDevice)` to make a Thread device be an SSED.

pavel808, 2024-08-07:
> You should call `ConnectivityMgr().SetThreadDeviceType(ConnectivityManager::kThreadDeviceType_SynchronizedSleepyEndDevice)` to make a Thread device be an SSED.

@wqx6 Thanks for your reply. Are there other configuration options involved? For polling times etc.

closed 2024-12-19 by pavel808