espressif/esp-matter #1037: Connect device to Zigbee gateway

NotClear, 2024-08-02:
I bought the Zigbee Gateway DevKit board and successfully installed the Zigbee Bridge example. I would like to test an IKEA tradfri lamp with Zigbee 3.0. Can I connect the lamp to the gateway if so how? The lamp does not connect to the gateway in pairing mode.

wqx6, 2024-08-06:
Firstly, you should figure out what pair method the IKEA lamp are using to join a Zigbee network. Are the lamp using steering/install-code/touch-link to join the network? Our Zigbee bridge are using steering to add a Zigbee device. And you should also change the pairing method on the Zigbee bridge. Please raise issue on https://github.com/espressif/esp-zigbee-sdk if you have any issues on adding Zigbee device to Zigbee gateway. once you have added the device to the network, you can read the data model with the Bridge and create the dynamic endpoint for the zigbee device.

dhrishi, 2024-08-27:
Closing as the question is answered.

closed 2024-08-27 by dhrishi