espressif/esp-matter #1038: Help needed to configure device samples for certification

leonardbay, 2024-08-03:
Our firmware is based on the [Matter ESP32 All Clusters Example](https://github.com/project-chip/connectedhomeip/blob/master/examples/all-clusters-app/esp32/README.md), and the device type is a Fan.

When the firmware is built with the default sample settings the device operates as expected, passes Test Harness tests, chip-tool can commission and control it, and it can be added to and controlled by the Apple Home app.

When following instructions in [3: Matter Certification](https://docs.espressif.com/projects/esp-matter/en/latest/esp32h2/certification.html#) **"For Matter Certification Test, vendors should generate their own test Product Attestation Authority (PAA) certificate"**, the device cannot be commissioned by the chip tool, and console commands like "matter config" execute with errors.

Please see below the step by step details of how CD, PAA, DAC, factory partition are created and the firmware is built and flashed, noting that our vendor id is replaced by ????, and other fake names are used throughout.

### partition file (unchanged from the sample)
```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: Firmware partition offset needs to be 64K aligned, initial 36K (9 sectors) are reserved for bootloader and partition table
esp_secure_cert,  0x3F, ,0xd000,    0x2000, encrypted
nvs,      data, nvs,     0x10000,   0xC000,
nvs_keys, data, nvs_keys,,          0x1000, encrypted
otadata,  data, ota,     ,          0x2000
phy_init, data, phy,     ,          0x1000,
ota_0,    app,  ota_0,   0x20000,   0x1E0000,
ota_1,    app,  ota_1,   0x200000,  0x1E0000,
fctry,    data, nvs,     0x3E0000,  0x6000
```
### added to sdkconfig.defaults
```
CONFIG_ESP_MATTER_NVS_PART_NAME="nvs"
CONFIG_ESP_MATTER_BRIDGE_INFO_PART_NAME="nvs"
CONFIG_ENABLE_CHIP_SHELL=y
CONFIG_BUILD_CHIP_TESTS=n
CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER=y
CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER=y
CONFIG_FACTORY_COMMISSIONABLE_DATA_PROVIDER=y
CONFIG_FACTORY_DEVICE_INSTANCE_INFO_PROVIDER=y
CONFIG_FACTORY_PARTITION_DAC_PROVIDER=y
CONFIG_FACTORY_DEVICE_INFO_PROVIDER=y
CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL="fctry"
CONFIG_CHIP_CONFIG_NAMESPACE_PARTITION_LABEL="fctry"
CONFIG_CHIP_COUNTERS_NAMESPACE_PARTITION_LABEL="fctry"
CONFIG_CHIP_KVS_NAMESPACE_PARTITION_LABEL="fctry"
```
### generate CD file
```
chip-cert gen-cd \
--vendor-id 0x???? \
--product-id 0x0001 \
--device-type-id 0x002B \
--certificate-id CSA00000SWC00000-01 \
--certification-type 1 \
--version-number 1 \
--format-version 1 \
--security-level 0 \
--security-info 0 \
--key ~/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-Signing-Key.pem \
--cert ~/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-Signing-Cert.pem \
--out ~/matter/test_CD.der
```
### generate test PAA certificate
```
chip-cert gen-att-cert \
--subject-vid 0x???? \
--subject-cn "Test PAA" \
--valid-from "2024-08-01 00:00:00" \
--lifetime 4294967295 \
--type a \
--out ~/matter/test_paa_cert.pem \
--out-key ~/matter/test_paa_key.pem
```
### generate factory partition binary files
```
esp-matter-mfg-tool \
-v 0x???? \
--vendor-name VendorName \
--product-name ProductName \
-cn TESTMFGCN \
--paa -c /media/sf_Matter_Shared/test_paa_cert.pem \
-k /media/sf_Matter_Shared/test_paa_key.pem \
-cd /media/sf_Matter_Shared/test_CD.der \
-n 1 \
-p 0x0001 \
--serial-num "12345678" \
--hw-ver 1 \
--hw-ver-str v1.0 \
--passcode 20202021 \
--discriminator 3840
```
### build and flash firmware
```
idf.py build flash
```
### flash Matter factory partition
```
esptool.py write_flash 0x3E0000  out/????_1/d3a04978-d27f-4191-b3cc-19c0cd6d81e8/d3a04978-d27f-4191-b3cc-19c0cd6d81e8-partition.bin
```
### monitor console error 45
```
idf.py monitor
>matter config
VendorId:        ???? (0x????) # <--- correct vendor id, product id, and hardware version
ProductId:       1 (0x1)
HardwareVersion: 1 (0x1)
Error: 45
```
### chip tool error
```
pairing ble-wifi 1234 <SSID> <wifi password> 20202021 3840 --paa-trust-store-path ./test_paa_cert.pem
...
[1722692717.190] [40096:40098] [CTL] Performing next commissioning step 'AttestationVerification'
[1722692717.190] [40096:40098] [CTL] Verifying attestation
--> [1722692717.193] [40096:40098] [CTL] Error on commissioning step 'AttestationVerification': 'src/controller/CHIPDeviceController.cpp:1209: CHIP Error 0x000000AC: Internal error'
--> [1722692717.193] [40096:40098] [CTL] Failed verifying attestation information. Now checking DAC chain revoked status.
[1722692717.193] [40096:40098] [CTL] Commissioning stage next step: 'AttestationVerification' -> 'AttestationRevocationCheck'
[1722692717.193] [40096:40098] [CTL] Performing next commissioning step 'AttestationRevocationCheck' with completion status = 'src/controller/CHIPDeviceController.cpp:1209: CHIP Error 0x000000AC: Internal error'
[1722692717.193] [40096:40098] [CTL] Verifying device's DAC chain revocation status
--> [1722692717.193] [40096:40098] [CTL] Failed in verifying 'Attestation Information' command received from the device: err 101. Look at AttestationVerificationResult enum to understand the errors
--> [1722692717.193] [40096:40098] [CTL] Error on commissioning step 'AttestationRevocationCheck': 'src/controller/CHIPDeviceController.cpp:1258: CHIP Error 0x000000AC: Internal error'
[1722692717.193] [40096:40098] [CTL] Going from commissioning step 'AttestationRevocationCheck' with lastErr = 'src/controller/CHIPDeviceController.cpp:1258: CHIP Error 0x000000AC: Internal error' -> 'Cleanup'
...
```

Thank you for your help! Hopefully other developers will also benefit from it.

leonardbay, 2024-08-03:
Additional information: building with test vendor and product ids seems to work, chip-tool commissions and controls our device.

Please see steps below:

### generate cd file
```
chip-cert gen-cd \
--vendor-id 0xFFF2 \
--product-id 0x8001 \
--device-type-id 0x002B \
--certificate-id CSA00000SWC00000-01 \
--certification-type 1 \
--version-number 1 \
--format-version 1 \
--security-level 0 \
--security-info 0 \
--key ~/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-Signing-Key.pem \
--cert ~/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-Signing-Cert.pem \
--out ~/matter/fff2_8001_CD.der
```
### generate binary partition
```
esp-matter-mfg-tool \
-cn "Fan" \
-v 0xFFF2 \
-p 0x8001 \
--pai \
-k ~/connectedhomeip/credentials/test/Chip-Test-PAI-FFF2-8001-Key.pem \
-c ~/connectedhomeip/credentials/test/Chip-Test-PAI-FFF2-8001-Cert.pem \
-cd ~/matter/fff2_8001_CD.der \
--vendor-name "Vendor" \
--product-name "Fan" \
-n 1 \
--hw-ver 1 \
--hw-ver-str "v1.0" \
--passcode 20202021 \
--discriminator 3840
```

shripad621git, 2024-08-06:
@leonardbay , you can keep the other namespaces without a change except the `CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL="fctry" ` i.e
```
CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL="fctry"
CONFIG_CHIP_CONFIG_NAMESPACE_PARTITION_LABEL="nvs"
CONFIG_CHIP_COUNTERS_NAMESPACE_PARTITION_LABEL="nvs"
CONFIG_CHIP_KVS_NAMESPACE_PARTITION_LABEL="nvs"
```
Also please note that the `[--paa-trust-store-path]` is the path to directory holding PAA certificate information which can be absolute or relative to the current working directory and not the PAA cert file itself.

For e.g. : If the PAA cert is located at credentials/test/attetsation, the command will be 

```
pairing ble-wifi 1234 <SSID> <wifi password> 20202021 3840 --paa-trust-store-path  /path/to/credentials/test/attestation
```

closed 2024-08-06 by leonardbay