espressif/esp-matter #1040: Issues with Error Messages Display on ESP Insights Dashboard and Firmware Package Naming

ShahilDholariya, 2024-08-06:
**Describe the bug**
when i go to my esp insights dashboard and add package than it not shows error messages in correct form.
I tried custom error messages logged using ESP_LOGE in my ESP32 application are not displayed correctly on the ESP Insights dashboard. Instead of my custom messages, the dashboard shows messages indicating the location of a string in the read-only data section (ROData) along with the program counter (PC).

for (int i = 0; i < 10; ++i) {
ESP_LOGE(TAG, "Error code 2: Custom error message.");
vTaskDelay(1000 / portTICK_PERIOD_MS); // Wait 1 second between reports
}
this is my custom error message for testing purpose but on dashboard it shows like,

05/08/24 11:25:27
app_main: string at ROData: 0x3c134e16 PC: 0x4200ae61

ERROR
05/08/24 11:25:28
app_main: string at ROData: 0x3c134e16 PC: 0x4200ae61
...(10 times)

This issue persists for all error messages and appears 10 times. I am unsure why the dashboard is not displaying the actual error message.
Additionally, I have noticed that my firmware package is named matter_light_vfd78129.zip instead of the expected matter_light_v1.0.zip. 

This is My Firmware package details ,it might be helpful

Name: MATTER_LIGHT
Package Id: 51fb0352-9663-439d-a8cd-6935c0c7a9c3
Type: LED
Model:
Firmware Version: fd78129
Firmware SHA: c0f3e80d26ab030d
...

shripad621git, 2024-08-07:
@ShahilDholariya , if you expand the error message, it will prompt you to upload the firmware package generated in build directory.
[image]
You need to upload the firmware package from here to be able to view the error logs.
[image]

ShahilDholariya, 2024-08-08:
Thank you for Response,
I already upload firmware package and include matter_light-vfd78129.zip file but i am not able to view error logs.
even i tried simple rainmaker example code and upload zip file inside firmware package but in that example also not able to view error logs.

Please suggest any solution.

shripad621git, 2024-08-08:
@ShahilDholariya , please ensure that the firmware package should be of the latest firmware package and the running firmware should be same. Like if you do something like idf.py flash or idf.py build, you have to upload the new firmware package as mentioned [here](https://rainmaker.espressif.com/docs/esp-insights/#upload-firmware-package).

dhrishi, 2024-08-27:
@ShahilDholariya Please close the issue if resolved

dhrishi, 2024-09-03:
Please reopen if you still see the issue, even after following the above steps

closed 2024-09-03 by dhrishi