espressif/esp-matter #1041: FAIL: ConnectDevice: GDBus.Error:org.bluez.Error.Failed

caojun97, 2024-08-06:
**Describe the bug**

when i run the command: `chip-tool pairing ble-wifi 0x7283 MY_SSID MY_PASSWD 20202021 3840`, i got a error.

The error is like that:
```
CHIP:DL: FAIL: ConnectDevice: GDBus.Error:org.bluez.Error.Failed: Software caused connection abort (36)
[1722950998.888235][62794:62795] CHIP:DL: ConnectDevice retry: 1 out of 4
[1722951001.407642][62794:62795] CHIP:DL: FAIL: ConnectDevice: GDBus.Error:org.bluez.Error.Failed: Software caused connection abort (36)
[1722951001.407680][62794:62795] CHIP:DL: ConnectDevice retry: 2 out of 4
[1722951002.937974][62794:62795] CHIP:DL: FAIL: ConnectDevice: GDBus.Error:org.bluez.Error.Failed: Software caused connection abort (36)
[1722951002.938006][62794:62795] CHIP:DL: ConnectDevice retry: 3 out of 4
[1722951003.927847][62794:62795] CHIP:DL: FAIL: ConnectDevice: GDBus.Error:org.bluez.Error.Failed: Software caused connection abort (36)
[1722951003.927895][62794:62795] CHIP:DL: ConnectDevice retry: 4 out of 4
[1722951003.959837][62794:62795] CHIP:BLE: Device connection failed: src/platform/Linux/bluez/BluezEndpoint.cpp:657: CHIP Error 0x000000AC: Internal error
...

CHIP:TOO: Run command failure: src/platform/Linux/bluez/BluezEndpoint.cpp:656: CHIP Error 0x000000AC: Internal error
```

But my bluetooth is OK, 
```
hci0:	Type: Primary  Bus: USB
	BD Address: 60:E9:AA:D3:81:94  ACL MTU: 1024:8  SCO MTU: 240:4
	UP RUNNING 
	RX bytes:548517 acl:99 sco:0 events:12643 errors:0
	TX bytes:30161 acl:38 sco:0 commands:1378 errors:0
	Features: 0xff 0xfe 0x8f 0xfe 0xd8 0x3f 0x5b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'localhost'
	Class: 0x1c010c
	Service Classes: Rendering, Capturing, Object Transfer
	Device Class: Computer, Laptop
	HCI Version: 5.0 (0x9)  Revision: 0x0
	LMP Version: 5.0 (0x9)  Subversion: 0x17f3
	Manufacturer: Qualcomm (29)

```

So i think the error is a bug?   I hope you can help me, thank you so much

**Environment**
- ESP-Matter Commit Id: v main
- ESP-IDF Commit Id: v5.2.2
- SoC (eg: ESP32 or ESP32-C3): ESP32-C3
- Device Logs (Please attach the log file):
[ble-wifi-error.log](https://github.com/user-attachments/files/16511036/ble-wifi-error.log)
[idf-moniot.log](https://github.com/user-attachments/files/16511044/idf-moniot.log)

- Host Machine OS:  Ubuntu 20.04
- Host Machine Python version: pybhon v3.8

- Commissioner app and versions if present:
- Home hub app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

shubhamdp, 2024-08-23:
@eiSouthBoy any updates on this issue?

dhrishi, 2024-08-27:
Closing this now. Plesae re-open if you still see issues

closed 2024-08-27 by dhrishi