espressif/esp-matter #1042: iot_button double_click

stefano055415, 2024-08-07:
Trying the generic_switch example, I noticed that the BUTTON_DOUBLE_CLICK event is not executed.

I believe the problem is related to version 2.5.0 of the espressiff/button dependency

Has anyone encountered the same problem?

shubhamdp, 2024-08-08:
`BUTTON_DOUBLE_CLICK` event is under `CONFIG_GENERIC_SWITCH_TYPE_LATCHING`, did you change the switch type to latching?

shubhamdp, 2024-08-09:
@stefano055415 I cut-pasted the code snippet that you provided and run the code on C3, with iot_button version 2.5.0, and I was able to see the double click event.

esp-idf: v5.2.2
SoC: ESP32-C3 with onboard push button (GPIO_9)

<details>
<summary>less dependencies.lock</summary>

```
dependencies:
  espressif/button:
    component_hash: 4d3ea5c1b7a4926c9d027e0bde0188207dadb0d1841d1cec2524f0cddd22f779
    source:
      service_url: https://api.components.espressif.com/
      type: service
    version: 2.5.0
```
</details>

[image]

Can you please help with more details (idf revision, chip, io number, any other details that would help debug this), or can you try with newer iot_button version?

stefano055415, 2024-08-09:
Hi @shubhamdp ,

I also tried setting GPIO 0 (active_level = false), it works correctly.

trying with GPIO 23 (active_level = true) the double click does not work

I am also using idf 5.2.2 and espressif/button.h 2.5.0

stefano055415, 2024-08-09:
I'm closing the issue. 

setting the time to 

CONFIG_BUTTON_SHORT_PRESS_TIME_MS=300

the double-click event is detected.

I probably add an external contact, I couldn't press the button twice without staying within 180 ms

closed 2024-08-09 by stefano055415