espressif/esp-matter #1043: Performing read requests

simonFloat, 2024-08-07:
I am trying to have my esp device perform a read request to get basic information of other devices on the matter network. I figured the correct call would be **group_request_send**, but looking at the implementation it only performs a custom callback?

```
esp_err_t group_request_send(uint8_t fabric_index, request_handle_t *req_handle)
{
    if (!req_handle) {
        ESP_LOGE(TAG, "command handle is null");
        return ESP_ERR_NO_MEM;
    }

    if (client_group_request_callback) {
        client_group_request_callback(fabric_index, req_handle, request_callback_priv_data);
    }

    return ESP_OK;
}
```

What would be the correct way for performing a read request?

simonFloat, 2024-08-08:
Figured out that the specification only supports unicast for sending read requests. But I am still unsure how to perform a read request from a device to a device, does my device have to be a controller to read basic information or can it be a basic node? If so It seems I need to use bindings but do they only bind clusters, like on-off server and client. How would one retrieve basic information?

jadhavrohit924, 2024-08-09:
> does my device have to be a controller to read basic information 

No.

Have you explored [clients](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_client.h) in esp-matter? Please check this we have provided API to send read, write and subscribe request from client to server.

simonFloat, 2024-08-09:
I have explored it but I am stuck at how to get a peer_device (deviceProxy)? What is the process of finding devices?

simonFloat, 2024-08-10:
Maybe my understanding of client and controller is a bit clouded? Is it correct that a client does communication with servers but gets told who to communicate with from a controller? And the controller role is more broad and also often acts like a client? or what is the distinction?

wqx6, 2024-08-12:
The reading attribute could only be unicast request, so you should send the read request to other devices one by one. The steps to perform a read request from a device to another device could be:
1. create a callback of client request, and set it by `esp_matter::client::set_request_callback()`.
2. create a response callback for the read request
```
void my_client _request_callback(peer_device_t *peer_device, request_handle_t *req_handle, void *priv_data)
{
    if (req_handle->type == READ_ATTR) {
        esp_matter::client::interaction::read::send_request(peer_device, &req_handle->attribute_path, 1, NULL, 0, response_callback);
    }
}
```
**Note:** the response callback is the callback to handle the response of read request.
3. call `esp_matter::client::connect()` to establish a CASE session and send the request.
You can refer to the controller component to know how to send the request https://github.com/espressif/esp-matter/blob/main/components/esp_matter_controller/commands/esp_matter_controller_read_command.cpp#L42-L44

wqx6, 2024-08-12:
You need to establish secure session with the device with FindOrEstablishSession and then you can get the peer_device in the success callback.

simonFloat, 2024-08-12:
Thank you so much, I will try this out! 

Can I also ask if you could explain the differences between client and controller? Looking at the code I can see that the controller also has utility for sending commands using the readClient, but also features from the commissioner role. My understanding is a controller is just a device that can control one/many other devices, and therefore makes use of a client cluster to do so. But what does the components/controller encapsulate?

wqx6, 2024-08-12:
> Can I also ask if you could explain the differences between client and controller? Looking at the code I can see that the controller also has utility for sending commands using the readClient, but [...]

A client is a device which can send command-invoke or attribute/event-read/subscribe or attribute-write requests to a Matter cluster server. It is based on cluster, for instance, a on-off cluster client can send toggle command to another on-off server. As for controller/commissioner, it is a device which includes a lot of client clusters.

The resources for a matter device could be managed by the two following classes:
 1. Server instance in `connectedhomeip/src/app/server/Server.h` which is used for most products that are not mainly controller/administrator role
 2. DeviceController in `connectedhomeip/src/controller/CHIPDeviceController.h` which is used for controller/commissioner.

Note: A controller could also use Server instance to manage its resources. but the controller with the server instance requires a Matter data model (root node and other endpoints).

The controller component provides APIs for the initialization of the `DeviceController` and the client operations(command-invoke, attribute-read/subscribe, event-read/subscribe, attribute-write).

simonFloat, 2024-08-12:
Thank you so much, that clears it up for me! 

Can I ask how node_ids are shared between devices. That is how does a device obtain node_ids that has been assigned after commissioning, if they are shared across the network or if they are found when doing operational discovery, and if it is something all devices can perform/request (does it require any privileges)?

wqx6, 2024-08-13:
Matter doesn't restrict how the node_ids are shared between devices. Normally the node_ids are shared with the cloud. After commissioner(Phone app) commissions a Matter commissionee, it will assign a node id for that end-device and upload the device information and matte node_id to the cloud server, and the controller will be able the fetch the node_id from the cloud then. Another way to share the node_ids is binding cluster, after the admin sends binding commands to the binding cluster server, the binding cluster server will not the node_ids of the bound devices. Of course the binding cluster server requires the privilege of the the bound devices to send request to them. so the admin might also need to the write the acl of the bound devices to allow the control of bound server.

simonFloat, 2024-08-13:
Is it then Correctly understood that it then depends on the commissioner and ecosystem the user uses? (google, apple etc..). Im wondering as I am interested in my device(controller) to initiate the retrieval of node_ids and perform the read requests. But would my device have to request them from someone with access to the cloud server where it is stored.

Also would multiple commissioners then have to use the same ecosystem or at least have access to the same cloud server?

wqx6, 2024-08-14:
> Is it then Correctly understood that it then depends on the commissioner and ecosystem the user uses?

Yes.

> would my device have to request them from someone with access to the cloud server where it is stored.

The device can also browse operational mdns service(_matter._tcp) to get the node-id. But it is much more complicated, as you have to parse the instance names of the browsed results and filter the nodes which are not in the Matter fabric of the controller. And the controller will also need to read the descriptor clusters to know the device type of the node with specific node-id. It will be much easier if there is a cloud server which can help the controller getting the node-id and device information.

> would multiple commissioners then have to use the same ecosystem or at least have access to the same cloud server?

Yes.

wqx6, 2024-08-14:
`GetDiscoveredNodes()` in `GetDiscoveredNodes()` will return commissionable nodes which are in the IP network but opening their commissioning window.

simonFloat, 2024-08-14:
> `GetDiscoveredNodes()` in `GetDiscoveredNodes()` will return commissionable nodes which are in the IP network but opening their commissioning window.

Is it Correctly understood that it will only find nodes that are still open for commissioning?

Is it possible to request the node-ids from a higher role? like an administrator or commissioner.

simonFloat, 2024-08-14:
I saw the the connectedhomeip submodule got updates yesterday, which included dns-sd utility for operational discovery. This makes use of the _DiscoveryImplPlatform_. My question is where this component gets initialised?

simonFloat, 2024-08-14:
> I saw the the connectedhomeip submodule got updates yesterday, which included dns-sd utility for operational discovery. This makes use of the _DiscoveryImplPlatform_. My question is where this component gets initialised?

When attempting to get the instance of the _DiscoveryImplPlatform_ I get the linker error: 
`undefined reference to chip::Dnssd::DiscoveryImplPlatform::GetInstance()`.

closed 2024-08-17 by simonFloat