espressif/esp-matter #1045: Commissioning Zigbee End Devices

Shivi-6, 2024-08-08:
**Describe the question/query that you have**
I have the zigbee gateway devkit. After following the steps in Zigbee Bridge example, I was able to flash and commission the zigbee bridge and it can now be added as a hub in Alexa as well as in Rainmaker App.
Even after adding the zigbee bridge, when I try to add zigbee devices in Alexa, it says "Your device needs a hub to connect to Alexa". I am not sure how can I add the zigbee end devices to work with the matter-zigbee_bridge example.

**Additional context**
`I (25792) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 0 **********
**E (25792) zigbee_bridge: Unable to Update Bridge Device, ep: 0, cluster: 48, att: 0**
I (25792) chip[EM]: <<< [E:46470r S:42275 M:76736135 (Ack:162606593)] (S) Msg TX to 1:038D91DEAB9EE1C1 [8F71] [UDP:192.168.1.116:5541] --- Type 0001:09 (IM:InvokeCommandResponse)
I (25802) chip[SVR]: Commissioning completed successfully
I (25802) chip[DIS]: Updating services using commissioning mode 0
I (25802) chip[DIS]: CHIP minimal mDNS started advertising.
I (25812) chip[DIS]: Advertise operational node 072A1D9FE6478F71-038E815F96662240
I (25822) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 072A1D9FE6478F71-038E815F96662240.
I (25822) chip[DIS]: mDNS service published: _matter._tcp
I (25822) chip[SC]: SecureSession[0x3fca43d8, LSID:42274]: State change 'kActive' --> 'kPendingEviction'
I (25832) chip[BLE]: Releasing end point's BLE connection back to application.
I (25832) app_main: Commissioning complete
I (25832) esp_matter_core: Commissioning Complete
I (25832) app_main: Commissioning window closed
I (25832) NimBLE: GAP procedure initiated: stop advertising.

I (25832) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

I (25852) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 1 ) indicate = 0
I (25852) chip[DL]: CHIPoBLE unsubscribe received
I (25852) chip[DL]: BLE GAP connection terminated (con 1 reason 0x216)
I (25862) chip[DL]: BLE deinit successful and memory reclaimed
**E (25862) chip[BLE]: no endpoint for unsub recvd**
I (25862) esp_matter_core: BLE Disconnected
I (25932) chip[EM]: >>> [E:46470r S:42275 M:162606594 (Ack:76736135)] (S) Msg RX from 1:038D91DEAB9EE1C1 [8F71] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (25942) chip[EM]: >>> [E:46471r S:42275 M:162606595] (S) Msg RX from 1:038D91DEAB9EE1C1 [8F71] --- Type 0001:08 (IM:InvokeCommandRequest)
I (25942) esp_matter_command: Received command 0x00000009 for endpoint 0x0000's cluster 0x0000003E
I (25942) chip[ZCL]: OpCreds: Received an UpdateFabricLabel command
I (25962) chip[FP]: Metadata for Fabric 0x1 persisted to storage.
I (25972) chip[EM]: <<< [E:46471r S:42275 M:76736136 (Ack:162606595)] (S) Msg TX to 1:038D91DEAB9EE1C1 [8F71] [UDP:192.168.1.116:5541] --- Type 0001:09 (IM:InvokeCommandResponse)
I (25982) chip[EM]: >>> [E:46471r S:42275 M:162606596 (Ack:76736136)] (S) Msg RX from 1:038D91DEAB9EE1C1 [8F71] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (25992) chip[EM]: >>> [E:46472r S:42275 M:162606597] (S) Msg RX from 1:038D91DEAB9EE1C1 [8F71] --- Type 0001:02 (IM:ReadRequest)
I (26002) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x00000028's Attribute 0x0000FFFC is 0 **********
I (26002) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x00000028's Attribute 0x0000FFFD is 3 **********
I (26002) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x00000028's Attribute 0x00000005 is (empty) **********
I (26012) chip[EM]: <<< [E:46472r S:42275 M:76736137 (Ack:162606597)] (S) Msg TX to 1:038D91DEAB9EE1C1 [8F71] [UDP:192.168.1.116:5541] --- Type 0001:05 (IM:ReportData)
I (26052) chip[EM]: >>> [E:46472r S:42275 M:162606598 (Ack:76736137)] (S) Msg RX from 1:038D91DEAB9EE1C1 [8F71] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (26062) chip[EM]: >>> [E:46473r S:42275 M:162606599] (S) Msg RX from 1:038D91DEAB9EE1C1 [8F71] --- Type 0001:02 (IM:ReadRequest)
I (26062) chip[EM]: <<< [E:46473r S:42275 M:76736138 (Ack:162606599)] (S) Msg TX to 1:038D91DEAB9EE1C1 [8F71] [UDP:192.168.1.116:5541] --- Type 0001:05 (IM:ReportData)
I (26082) chip[EM]: >>> [E:46473r S:42275 M:162606600 (Ack:76736138)] (S) Msg RX from 1:038D91DEAB9EE1C1 [8F71] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (26092) chip[EM]: >>> [E:46474r S:42275 M:162606601] (S) Msg RX from 1:038D91DEAB9EE1C1 [8F71] --- Type 0001:02 (IM:ReadRequest)
I (26102) chip[EM]: <<< [E:46474r S:42275 M:76736139 (Ack:162606601)] (S) Msg TX to 1:038D91DEAB9EE1C1 [8F71] [UDP:192.168.1.116:5541] --- Type 0001:05 (IM:ReportData)
I (26132) chip[EM]: >>> [E:46474r S:42275 M:162606602 (Ack:76736139)] (S) Msg RX from 1:038D91DEAB9EE1C1 [8F71] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (28202) chip[EM]: >>> [E:46475r S:42275 M:162606603] (S) Msg RX from 1:038D91DEAB9EE1C1 [8F71] --- Type 0001:06 (IM:WriteRequest)
I (28212) chip[EM]: <<< [E:46475r S:42275 M:76736140 (Ack:162606603)] (S) Msg TX to 1:038D91DEAB9EE1C1 [8F71] [UDP:192.168.1.116:5541] --- Type 0001:07 (IM:WriteResponse)
I (28232) chip[EM]: >>> [E:46475r S:42275 M:162606604 (Ack:76736140)] (S) Msg RX from 1:038D91DEAB9EE1C1 [8F71] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (28232) chip[EM]: >>> [E:46476r S:42275 M:162606605] (S) Msg RX from 1:038D91DEAB9EE1C1 [8F71] --- Type 0001:03 (IM:SubscribeRequest)
I (28242) chip[DMG]: Final negotiated min/max parameters: Min = 1s, Max = 300s
I (28252) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x0000001F's Attribute 0x0000FFFC is 0 **********
I (28252) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x00000028's Attribute 0x0000FFFC is 0 **********
I (28262) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x00000028's Attribute 0x0000FFFD is 3 **********
I (28262) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x00000028's Attribute 0x00000005 is (empty) **********
**E (28262) chip[DMG]: Error retrieving data from clusterId: 0x0000_0028, err = b**
I (28272) chip[EM]: <<< [E:46476r S:42275 M:76736141 (Ack:162606605)] (S) Msg TX to 1:038D91DEAB9EE1C1 [8F71] [UDP:192.168.1.116:5541] --- Type 0001:05 (IM:ReportData)
I (28472) chip[EM]: >>> [E:46476r S:42275 M:162606606 (Ack:76736141)] (S) Msg RX from 1:038D91DEAB9EE1C1 [8F71] --- Type 0001:01 (IM:StatusResponse)
I (28472) chip[IM]: Received status response, status is 0x00
I (28472) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x0000FFFC is 0 **********
I (28472) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x0000FFFD is 1 **********
I (28482) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 0 **********
**E (28482) chip[DMG]: Error retrieving data from clusterId: 0x0000_0033, err = b**
I (28492) chip[EM]: <<< [E:46476r S:42275 M:76736142 (Ack:162606606)] (S) Msg TX to 1:038D91DEAB9EE1C1 [8F71] [UDP:192.168.1.116:5541] --- Type 0001:05 (IM:ReportData)
I (28532) chip[EM]: >>> [E:46476r S:42275 M:162606607 (Ack:76736142)] (S) Msg RX from 1:038D91DEAB9EE1C1 [8F71] --- Type 0001:01 (IM:StatusResponse)
I (28542) chip[IM]: Received status response, status is 0x00
I (28542) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x0000003C's Attribute 0x0000FFFC is 0 **********
I (28542) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x0000003C's Attribute 0x0000FFFD is 1 **********
I (28552) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x0000003E's Attribute 0x0000FFFC is 0 **********
**E (28552) chip[DMG]: Error retrieving data from clusterId: 0x0000_003E, err = b**
I (28562) chip[EM]: <<< [E:46476r S:42275 M:76736143 (Ack:162606607)] (S) Msg TX to 1:038D91DEAB9EE1C1 [8F71] [UDP:192.168.1.116:5541] --- Type 0001:05 (IM:ReportData)
I (28772) chip[EM]: >>> [E:46476r S:42275 M:162606608 (Ack:76736143)] (S) Msg RX from 1:038D91DEAB9EE1C1 [8F71] --- Type 0001:01 (IM:StatusResponse)
I (28772) chip[IM]: Received status response, status is 0x00
I (28782) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x0000003E's Attribute 0x0000FFFD is 1 **********
**E (28792) chip[DMG]: Error retrieving data from clusterId: 0x0000_003E, err = b**
[... cut ...]
I (28962) chip[IM]: Received status response, status is 0x00
I (28962) chip[EM]: <<< [E:46476r S:42275 M:76736146 (Ack:162606610)] (S) Msg TX to 1:038D91DEAB9EE1C1 [8F71] [UDP:192.168.1.116:5541] --- Type 0001:04 (IM:SubscribeResponse)
I (28972) chip[DMG]: Registered a ReadHandler that will schedule a report between system Timestamp: 0x000000000000739A and system Timestamp 0x0000000000050392.
I (28982) chip[EM]: >>> [E:46476r S:42275 M:162606611 (Ack:76736146)] (S) Msg RX from 1:038D91DEAB9EE1C1 [8F71] --- Type 0000:10 (SecureChannel:StandaloneAck)
`

wqx6, 2024-08-12:
You should setup a light bulb on ESP32-H2 following https://github.com/espressif/esp-matter/tree/main/examples/zigbee_bridge#22-setup-zigbee-bulb-on-esp32-h2. If the light bulb is in the same channel as the Zigbee bridge, it will joining the Zigbee network and an endpoint will be added to the bridge.

Shivi-6, 2024-08-13:
@wqx6 Is it not possible to add some other end devices ? I have some Vetaar zigbee devices from Anchor (smart plug, dimmer).

wqx6, 2024-08-13:
Firstly, you should figure out what pair method your zigebee end-devices are using to join a Zigbee network. Are the devices using steering/install-code/touch-link to join the network? Our Zigbee bridge are using steering to add a Zigbee device by default. If the end-devices are using other pair mode, you should also change the pairing mode on the Zigbee bridge. Please raise issue on https://github.com/espressif/esp-zigbee-sdk if you have any issues on adding Zigbee device to Zigbee gateway.

Note that currently the zigbee bridge supports only on-off light zigbee end devices, if you want to add more device types, you need to add the device types support in [zigbee_bridge.cpp](https://github.com/espressif/esp-matter/blob/main/examples/zigbee_bridge/main/zigbee_bridge.cpp)

once you have added the device to the network, you can read the data model with the Bridge and create the dynamic endpoint for the zigbee device.

Shivi-6, 2024-08-13:
Thanks for this information. I will try to add the zigbee light to the gateway and also find the pairing method being used.

closed 2024-08-13 by Shivi-6