espressif/esp-matter #1046: How to modify the name of the scanning device

MaplestoryAlen, 2024-08-08:
I currently have multiple matter devices, but their names are all "Matter Accessory". How can I change this name so that I can distinguish between different devices.

shubhamdp, 2024-08-08:
Renaming the device name when adding the device to home app should work.

MaplestoryAlen, 2024-08-09:
> Renaming the device name when adding the device to home app should work.

If I have 10 devices connected to the network at the same time, it will display 10 'Matter Accessory', so I cannot be sure which one is my own device.

shubhamdp, 2024-08-09:
Identifying your device would work. In esp-matter there is callback when someone tried to identify your device: https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L122.

Also, for mode details, please check the identify cluster.

ronny-antoon, 2024-08-09:
Is their any wat to change the name from the code or configurations in esp_matter or mfg_tool ?

Like we can change the Pairing code/Manufacturer name/...  in mfg_tool, is their any way to change the Display Name when pairing ?

MaplestoryAlen, 2024-08-12:
> Identifying your device would work. In esp-matter there is callback when someone tried to identify your device: https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L122.
> 
> Also, for mode details, please check the identify cluster.
What I mean is to use the Apple app - "Home" to configure the network. For 10 devices, all of them display "Matter Accessory" on the interface. I cannot distinguish which device I want to configure, so is there any way to modify this name? For example, I want to customize the device name, such as Device 1, Device 2, Device 3

shubhamdp, 2024-08-23:
@MaplestoryAlen @ronny-antoon what you have requested may not be possible right now. Even if an ecosystem names its devices say based on some attribute, other ecosystem may not do the same and could lead to inconsistent behaviour across ecosystems.

MaplestoryAlen, 2024-08-23:
> @MaplestoryAlen @ronny-antoon what you have requested may not be possible right now. Even if an ecosystem names its devices say based on some attribute, other ecosystem may not do the same and could lead to inconsistent behaviour across ecosystems.

Thank you for your answer

closed 2024-08-23 by MaplestoryAlen