espressif/esp-matter #1047: One ESP32 with 64 Endpoints.

ronny-antoon, 2024-08-11:
I have a system DIY, with 64 input and 64 output, and serial connection.

i want to make the esp32 to communicate with it by serial and control the 64 "endpoint OnOffLight" that control 64 button with 64 relay.

the communication done and working perfect.
but their is no enough fee heap space. i want it in a one device not more.

Solutions ?

pimpalemahesh, 2024-08-13:
@ronny-antoon You can add up to 16 endpoints. However, if you disable Bluetooth, you can add up to 40 endpoints. In that case, you'll need to find another way to do the commissioning, as all memory will be consumed with 40 endpoints.

ronny-antoon, 2024-08-14:
> @ronny-antoon You can add up to 16 endpoints. However, if you disable Bluetooth, you can add up to 40 endpoints. In that case, you'll need to find another way to do the commissioning, as all memory will be consumed with 40 endpoints.

Thank you,

But thats not enough, i need 64 endpoints

pimpalemahesh, 2024-08-14:
@ronny-antoon If you can afford PSRAM, you can use it by connecting it externally. You can refer to the documentation here: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/external-ram.html. Use the `Memory allocation strategy` configuration option from menuconfig.

pimpalemahesh, 2024-08-14:
@ronny-antoon You can use PSRAM. Here is reference from esp-idf documentation [https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/external-ram.html](url).

Configure below options from menuconfig. Use `idf.py menuconfig` command to open menuconfig gui.

Select Component config -> ESP PSRAM
Enable `Support for external, SPI-connected RAM`.
Check if `SPIRAM_USE_MALLOC` **OR** `SPIRAM_USE_CAPS_ALLOC` options are enabled.

Select Component config -> ESP Matter -> Memory allocation strategy
Enable option `External SPIRAM`

dhrishi, 2024-08-27:
As mentioned, for large number of endpoints (like 64), using PSRAM and moving the allocations to it is a solution. Closing this now

closed 2024-08-27 by dhrishi