espressif/esp-matter #1048: Energy Management Application

svn-matterize, 2024-08-11:
Hi,

I would like to put together an application which measure the power consumption and provide the forecast with the previous data.

I've started with [light](https://github.com/espressif/esp-matter/tree/release/v1.3/examples/light) and mostly assembled the pieces.

For convenience, I wanted to work zap, so tried with [zap_light](https://github.com/espressif/esp-matter/tree/release/v1.3/examples/zap_light) and unable to write to the attributes (fields are greyed out) and reading the attribute from the `chip-tool` lead to failure even when it's enabled.

What might be the issue?

jadhavrohit924, 2024-08-12:
> I would like to put together an application which measure the power consumption and provide the forecast with the previous data.

Try `ElectricSensor` device?

> What might be the issue?

Please share what you have done, in detailed steps. and which device type are you building using zap-light?

svn-matterize, 2024-08-12:
Yeah, `electrical sensor` is used.

But for some reason, the fields are greyed out as you see  in the below image.

[image]

`electricalpowermeasurement read feature-map 1 2 ` leads to failure

```
[1723452510.052000][4094:4096] CHIP:EM: Found matching exchange: 47328i, Delegate: 0x73cdbc013940
[1723452510.052010][4094:4096] CHIP:EM: Rxd Ack; Removing MessageCounter:178751674 from Retrans Table on exchange 47328i
[1723452510.052029][4094:4096] CHIP:DMG: ReportDataMessage =
[1723452510.052036][4094:4096] CHIP:DMG: {
[1723452510.052057][4094:4096] CHIP:DMG: 	AttributeReportIBs =
[1723452510.052066][4094:4096] CHIP:DMG: 	[
[1723452510.052072][4094:4096] CHIP:DMG: 		AttributeReportIB =
[1723452510.052093][4094:4096] CHIP:DMG: 		{
[1723452510.052098][4094:4096] CHIP:DMG: 			AttributeStatusIB =
[1723452510.052104][4094:4096] CHIP:DMG: 			{
[1723452510.052110][4094:4096] CHIP:DMG: 				AttributePathIB =
[1723452510.052116][4094:4096] CHIP:DMG: 				{
[1723452510.052123][4094:4096] CHIP:DMG: 					Endpoint = 0x2,
[1723452510.052130][4094:4096] CHIP:DMG: 					Cluster = 0x90,
[1723452510.052136][4094:4096] CHIP:DMG: 					Attribute = 0x0000_FFFC,
[1723452510.052143][4094:4096] CHIP:DMG: 				}
[1723452510.052151][4094:4096] CHIP:DMG: 					
[1723452510.052157][4094:4096] CHIP:DMG: 				StatusIB =
[1723452510.052164][4094:4096] CHIP:DMG: 				{
[1723452510.052171][4094:4096] CHIP:DMG: 					status = 0x01 (FAILURE),
[1723452510.052177][4094:4096] CHIP:DMG: 				},
[1723452510.052184][4094:4096] CHIP:DMG: 				
[1723452510.052190][4094:4096] CHIP:DMG: 			},
[1723452510.052197][4094:4096] CHIP:DMG: 			
[1723452510.052203][4094:4096] CHIP:DMG: 		},
[1723452510.052211][4094:4096] CHIP:DMG: 		
[1723452510.052216][4094:4096] CHIP:DMG: 	],
[1723452510.052224][4094:4096] CHIP:DMG: 	
[1723452510.052230][4094:4096] CHIP:DMG: 	SuppressResponse = true, 
[1723452510.052236][4094:4096] CHIP:DMG: 	InteractionModelRevision = 11
[1723452510.052241][4094:4096] CHIP:DMG: }
[1723452510.052311][4094:4096] CHIP:TOO: Response Failure: IM Error 0x00000501: General error: 0x01 (FAILURE)
```

jadhavrohit924, 2024-08-14:
It looks like it's an error with connectedhomeip SDK.

svn-matterize, 2024-08-15:
@jadhavrohit924 - For the zap?

jadhavrohit924, 2024-08-16:
No for the `Failure` you got. 

For zap, can you please share with me which attributes you are trying to update?

dhrishi, 2024-08-27:
@svn-matterize Can you please provide the requested information

svn-matterize, 2024-08-27:
Hey, thanks for the nudge. Here is the [zap](https://github.com/user-attachments/files/16760979/lighting-app.zip) file I'm working with.

env:
idf: [ v5.2.1](https://github.com/espressif/esp-idf/tree/v5.2.1)
matter: [v1.3](https://github.com/espressif/esp-matter/tree/release/v1.3)

jadhavrohit924, 2024-08-28:
@svn-matterize This can be the issue with the zap-tool. I failed you open your .zap on release/v1.3 branch in the zap-tool as there is some problem with your zap. Can you please tell me which zap-tool you are using I mean the connectedhomeip commit id and the zap release.

Does all the attributes of `electricalpowermeasurement` can not be modified in the zap-tool?

svn-matterize, 2024-08-29:
@jadhavrohit924 

I am using 
matter: [v1.3](https://github.com/espressif/esp-matter/tree/release/v1.3) ,
[Connectedhomeip](https://github.com/espressif/connectedhomeip/tree/ae52f4e597cd79e956142009b82eb700a0c916f1)
[image]

Issue
I am  adding  the `Matter Electrical Sensor (0x0510)` as Endpoint 2 , In which  (Electrical Power Measurement,Electrical Energy Measurement, Power Topology clusters) 
where only `ClusterRevision (0xFFFD)` is editable and rest of all attributes are  not editable 
but in Endpoint 1 `Matter Dimmable Light (0x0101)` (On/Off, Level control , color control clusters ) where all  attributes are editable 
[image]

jadhavrohit924, 2024-08-29:
@svn-matterize Thanks for the detailed explanation. If you see the storage type for these attributes is `External` because it is expected that the attribute data will be provided by the Application and SDK has delegated this action to the Application. That's why you cannot set these attributes in the zap-tool. You have to provide a delegate implementation in your application to use this cluster. Here is the [reference implementation](https://github.com/project-chip/connectedhomeip/blob/c14a370535c4a3f30277761cc4ce277f84ba8949/examples/energy-management-app/energy-management-common/energy-reporting/include/ElectricalPowerMeasurementDelegate.h#L29).

svn-matterize, 2024-08-29:
Is it the case for all `External` type of the attributes?

jadhavrohit924, 2024-08-29:
Yes. External means it is configured externally.

svn-matterize, 2024-08-29:
Gotcha, let me give it a try and let you know.

svn-matterize, 2024-09-03:
Alright. [This](https://github.com/project-chip/connectedhomeip/blob/c14a370535c4a3f30277761cc4ce277f84ba8949/examples/energy-management-app/energy-management-common/energy-reporting/include/ElectricalPowerMeasurementDelegate.h#L29) worked.

However, in a meantime, I've leveraged something without the help of the .zap implementation. On that when I try to write to `ElectricalEnergyMeasurement::Attributes::Accuracy::Id` with

```c
ElectricalEnergyMeasurement::Structs::MeasurementAccuracyRangeStruct::Type energyAccuracyRanges[] = {
		{ .rangeMin = 0,
		  .rangeMax = 1'000'000'000'000'000, // 1 million Mwh
		  .percentMax = chip::MakeOptional(static_cast<chip::Percent100ths>(500)),
		  .percentMin = chip::MakeOptional(static_cast<chip::Percent100ths>(50)) }
	};

	ElectricalEnergyMeasurement::Structs::MeasurementAccuracyStruct::Type accuracy = {
		.measurementType = chip::app::Clusters::detail::MeasurementTypeEnum::kElectricalEnergy,
		.measured = true,
		.minMeasuredValue = 0,
		.maxMeasuredValue = 1'000'000'000'000'000, // 1 million Mwh
		.accuracyRanges = chip::app::DataModel::List<
			const ElectricalEnergyMeasurement::Structs::MeasurementAccuracyRangeStruct::Type>(
			energyAccuracyRanges)
	};
	arr_val = esp_matter_array((uint8_t *)&accuracy, sizeof(arr_val), 1);
	err = attribute::set_val(attribute, &arr_val);
	if (err != ESP_OK) {
		ESP_LOGE(TAG, "Updated EnergyAccuracy failed");
	}
```
It gets success in writing, however when reading the same from the `chip-tool`, it returns,

```sh
[1725372167.731751][7535:7537] CHIP:DMG: 					Endpoint = 0x1,
[1725372167.732217][7535:7537] CHIP:DMG: 					Cluster = 0x91,
[1725372167.732742][7535:7537] CHIP:DMG: 					Attribute = 0x0000_0000,
[1725372167.733015][7535:7537] CHIP:DMG: 				}
[1725372167.733128][7535:7537] CHIP:DMG: 					
[1725372167.733303][7535:7537] CHIP:DMG: 				StatusIB =
[1725372167.733342][7535:7537] CHIP:DMG: 				{
[1725372167.733367][7535:7537] CHIP:DMG: 					status = 0x89 (RESOURCE_EXHAUSTED),
[1725372167.733502][7535:7537] CHIP:DMG: 				},
```
Is there something not right in writing to the attributes?

If you want me to open this as a separate issue, I'm happy to.

jadhavrohit924, 2024-09-04:
@svn-matterize Please open a separate issue with all the code that you have added, I mean your delegate implementation. Also, add device and chip-tool logs in separate files to analyze. Please close this issue as your original problem is resolved.

svn-matterize, 2024-09-04:
Okay, I have created as a new issue [refer](https://github.com/espressif/esp-matter/issues/1075)
Thanks for your support.

closed 2024-09-04 by svn-matterize