espressif/esp-matter #1049: Unable to commission the matter light example in esp32c6 dev board

Mr-Techtron, 2024-08-11:
I was trying to commission the matter light with esp-idf v5.2.2 tag and esp-matter release/v1.3 branch. I am unable to commission the matter light device. However if I flash the firmware from the launch pad, I am able to commission the device but it doesn't control the light used in esp32c6 dev board and doesn't respond to the boot button.

**Environment**
- ESP-Matter Commit Id: 48751df484932b269b03b7a1432d5e2fb68b95cf (release/v1.3)
- ESP-IDF Commit Id: 3b8741b172dc951e18509698dee938304bcf1523 (v5.2.2 tag)
- SoC (eg: ESP32 or ESP32-C3): ESP32C6 dev board
- Device Logs (Please attach the log file):
[esp32c6_light_log.txt](https://github.com/user-attachments/files/16575659/esp32c6_light_log.txt)

- Host Machine OS: Ubuntu 24.04 LTS
- Host Machine Python version: python 3.12.3

- Commissioner app and versions if present: 3.4.1-50658f9 (Android esp-rainmaker app)
- Home hub app and versions if present: matter_controller (example from esp-rainmaker matter folder)
- Commissioner's logs if present:
[matter_controller_log.txt](https://github.com/user-attachments/files/16575660/matter_controller_log.txt)

**Any additional details**
Note: In the rainmaker app, the UI of the matter controller doesn't show "Local" or "Remote"
[image]

shubhamdp, 2024-08-12:
@Mr-Techtron 

> I am unable to commission the matter light device

What are you trying to commission? A WiFi end device or thread end device?

> However if I flash the firmware from the launch pad, I am able to commission the device

Which example did you use on C6, thread_matter_light or wifi_matter_light?

Also, can you elaborate on what exactly you are trying to do? Are you trying to commission the Thread end device or Wi-Fi end device? For commissioning thread end device you should have to setup the TBR on the RainMaker controller. For this, click on the controller and you should see the `Setup Thread Border Router` option. And after that you will be able to commission the thread end device.

For building the C6 WiFi light is straight forward: `idf.py build`.
For building the C6 Thread light:
```
rm sdkconfig.defaults.esp32c6
idf.py -D SDKCONFIG_DEFAULTS="sdkconfig.defaults;sdkconfig.defaults.c6_thread"
```

Please give it a try, and let me know if this works.

Mr-Techtron, 2024-08-12:
@shubhamdp 
> What are you trying to commission? A WiFi end device or thread end device?
    I am trying to commission the thread end device. As you can see, I have attached the log files of matter thread device and matter controller log above.

> Which example did you use on C6, thread_matter_light or wifi_matter_light?
    thread_matter_light
> Also, can you elaborate on what exactly you are trying to do? Are you trying to commission the Thread end device or Wi-Fi end device? For commissioning thread end device you should have to setup the TBR on the RainMaker controller. For this, click on the controller and you should see the `Setup Thread Border Router` option. And after that you will be able to commission the thread end device.
> The issue is that the UI doesn't prompt me the page that display the 'Setup Thread Border Router'. I have mentioned above and attached the screenshot, it neither shows "Local" nor "Remote". In order to show the 'Setup Thread Border Router' option, the matter controller UI tile should show "Local". This isn't happening in my case considering that the matter controller and my phone are both connected to same WiFi.
> For building the C6 WiFi light is straight forward: `idf.py build`. For building the C6 Thread light:
> 
> ```
> rm sdkconfig.defaults.esp32c6
> idf.py -D SDKCONFIG_DEFAULTS="sdkconfig.defaults;sdkconfig.defaults.c6_thread"
> ```
> 
> I had did the same method as you mentioned, but still I can't commission the matter thread light.

shubhamdp, 2024-08-17:
@Mr-Techtron steps that you have mentioned are in line, If the device is in the same wifi network it should show the `local` as well on the controller.

In attached log file, esp32c6_light_log.txt, I was/am seeing some wifi logs in there, so I thought may be some problem building the firmware. Since its a thread only firmware, it should not initialize the WiFi. Can you please share the sdkconfig of the project?

Regarding launchpad thread light unable to control the on board LED and push button, will check and fix this, could be some problem with GPIO numbers.

LiangZugeng, 2024-08-17:
@shubhamdp I encountered the exact same problem, in my case, I flashed the thread-matter-light example from the launch pad, the flash process finished without any errors, and I reset the esp32-c6, after that I can add the light in my Apple Home, but when I turn on/off the light in Apple Home, the physical LED didn't respond. From the console I can tell esp32-c6 worked well with matter/thread because when I tried to turn on/off the light, the console showed the communication logs without error, it's just that the LED didn't work, I guess it had something to do with the LED GPIO control logics in the example.

To further proving the guess, I tried to flash wifi-matter-light example on the same esp32-c6 board, everything works as expected.

Hope this will provide more information for troubleshooting the example and look forward having it fixed ASAP.

Mr-Techtron, 2024-08-18:
@shubhamdp I resolved the issue related to esp32c6 matter light. Now, I am facing the issue related to the matter controller that doesn't show "Local" or "Remote" which I mentioned above. Due to this, I am unable to setup the thread border router setup. It seems there is issue in the rainmaker cloud. When will this be fixed? It's already been like 2 weeks, I am facing this issue.

shubhamdp, 2024-08-23:
@Mr-Techtron figured out why the LED and Boot button wasn't working, raised a fixed internally, it will be released soon.

Mr-Techtron, 2024-08-28:
@shubhamdp what about the "Local / Remote" matter controller UI not showing up issue?

Mr-Techtron, 2024-09-07:
> @shubhamdp what about the "Local / Remote" matter controller UI not showing up issue?

@shubhamdp  any update on this?

shubhamdp, 2024-09-08:
@Mr-Techtron we are still working on fixing this problem. I'll update once this is fixed.

dhrishi, 2024-09-11:
@Mr-Techtron The fix is pushed to production. Can you please verify the same?

Mr-Techtron, 2024-09-11:
@dhrishi Do I have to update anything on the matter controller firmware as well? The UI status is still the same on my side. It is neither "Local" nor "Remote".

shubhamdp, 2024-09-12:
@Mr-Techtron  can you please remove and re-commission the controller in rainmaker android app and check.

Mr-Techtron, 2024-09-16:
@dhrishi @shubhamdp It seems the issue related to the Local / Remote UI has been resolved.

shubhamdp, 2025-03-18:
@Mr-Techtron Thread Border Router custom cluster is replaced by the official Thread BR Management cluster in release/v1.4 branch so the Phone app cannot configure the Thread network on the Controller+TBR.

For this test, you should switch to older revision (i.e. release/v1.3 and respective one for rainmaker). Please find the well tested revisions below. It is expected to work with current rainmaker android app.

- esp-idf: v5.2.3 tag
- esp-matter: ba10f4942b753699cd898e5f17ce0af8384cce44 on release/v1.3 branch
- esp-rainmaker 296e71dc3ab97cada3bb12c77e31fc2087c52d54