espressif/esp-matter #1051: Factory data : BLE Adv Data failed

MaplestoryAlen, 2024-08-12:
I have encountered a problem where Bluetooth encountered an error while using factory data, preventing me from opening the commissioning window. I did not encounter this issue with Matter 1.2 and IDF 5.1.2 versions, but now switching to IDF5.2.1 has resulted in this situation. Below is my setup process

1.Log:

[image]

2.sdkconfig:
[sdkconfig.txt](https://github.com/user-attachments/files/16578291/sdkconfig.txt)

3.Generated and converted files

[image]

4.menuconfig：
# Disable the DS Peripheral support
CONFIG_ESP_SECURE_CERT_DS_PERIPHERAL=n
# Use DAC Provider implementation which reads attestation data from secure cert partition
CONFIG_SEC_CERT_DAC_PROVIDER=y  
# Enable some options which reads CD and other basic info from the factory partition
CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER=y
CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER=y
CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL="fctry"

CONFIG_FACTORY_DEVICE_INSTANCE_INFO_PROVIDER=y

5.generate_esp32_chip_factory_bin command:

./generate_esp32_chip_factory_bin.py -d 1083 -p 10831125 \
--product-name 81ma1 --product-id 0x8000 \
--vendor-name Testvendor --vendor-id 0xFFF1 \
--hw-ver 1009 --hw-ver-str 1.0.0.9  --serial-num 202408091125 \
--dac-cert /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-DAC-01.der \
--dac-key /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-DAC-Key-01.der \
--pai-cert /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-PAI-Cert.der \
--cd /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/esp_dac_fff1_8000.der --dac-in-secure-cert --target esp32s3 

6.flash esp_secure_cert command:

esptool.py -p  /dev/ttyUSB0  write_flash 0xd000   /home/alen/esp-matter/connectedhomeip/connectedhomeip/scripts/tools/bin/esp_secure_cert_partititon.bin

esptool.py -p  /dev/ttyUSB0  write_flash   0xc20000   /home/alen/esp-matter/connectedhomeip/connectedhomeip/scripts/tools/bin/factory_partition.bin

7.partition table:
[image]

MaplestoryAlen, 2024-08-13:
> > /tmp/nvs.txt

1. chip ESP32S3 log:
[log.txt](https://github.com/user-attachments/files/16593461/log.txt)

2.matter version： release/v1.2 branch
SDK  README.md （/esp-matter/README.md）
[README.md](https://github.com/user-attachments/files/16593474/README.md)

 I can only generate the 'nvs. bin' file using esptool.exe. The analyze_nvs.p command will give me an error if I cannot use it， 
 Please use your tools to help analyze it

3.esptool.py -p /dev/ttyUSB0 read_flash 0xc20000 0x6000   /home/alen/esp-matter/connectedhomeip/connectedhomeip/scripts/tools/bin/nvs.bin

[nvs.zip](https://github.com/user-attachments/files/16593452/nvs.zip)

4.  ./analyze_nvs.py  /home/alen/esp-matter/connectedhomeip/connectedhomeip/scripts/tools/bin/nvs.bin  > /home/alen/esp-matter/connectedhomeip/connectedhomeip/scripts/tools/bin/nvs.txt

error log:

 ./analyze_nvs.py  /home/alen/esp-matter/connectedhomeip/connectedhomeip/scripts/tools/bin/nvs.bin  
./analyze_nvs.py: line 2: import: command not found
./analyze_nvs.py: line 3: import: command not found
./analyze_nvs.py: line 4: import: command not found
from: can't read /var/mail/enum
./analyze_nvs.py: line 11: PAGE_SIZE: command not found
./analyze_nvs.py: line 15: PSB_INIT: command not found
./analyze_nvs.py: line 16: PSB_FULL: command not found
./analyze_nvs.py: line 17: PSB_FREE: command not found
./analyze_nvs.py: line 18: PSB_CORRUPT: command not found
./analyze_nvs.py: line 22: ESB_WRITTEN: command not found
./analyze_nvs.py: line 23: ESB_ERASED: command not found
./analyze_nvs.py: line 26: PAGE_STATE_OFFSET: command not found
./analyze_nvs.py: line 27: PAGES_SEQ_NO_OFFSET: command not found
./analyze_nvs.py: line 28: ESB_OFFSET: command not found
./analyze_nvs.py: line 29: ESB_LENGTH: command not found
./analyze_nvs.py: line 30: ENTRY_0_OFFSET: command not found
./analyze_nvs.py: line 31: ENTRY_LENGTH: command not found
./analyze_nvs.py: line 34: NS_OFFSET: command not found
./analyze_nvs.py: line 35: TYPE_OFFSET: command not found
./analyze_nvs.py: line 36: SPAN_OFFSET: command not found
./analyze_nvs.py: line 37: CHUNK_IDX_OFFSET: command not found
./analyze_nvs.py: line 39: KEY_OFFSET: command not found
./analyze_nvs.py: line 40: KEY_LENGTH: command not found
./analyze_nvs.py: line 41: DATA_OFFSET: command not found
./analyze_nvs.py: line 45: syntax error near unexpected token `('
./analyze_nvs.py: line 45: `class EntryState(Enum):'

MaplestoryAlen, 2024-08-15:
> @MaplestoryAlen the steps and `sdkconfig` looks pretty sane, still wondering why it failed. Can you please share the complete logs from bootup (please attach them in the file).
> 
> Are you using esp-matter release/v1.2 branch?
> 
> Also, I see rainmaker configurations in your sdkconfig file. Any chance `fctry` was overwritten by rainmaker configuration? Can you please read back the `fctry` and analyze the nvs contents? Personally I use [Espressif-NVS-Analyzer](https://github.com/AFontaine79/Espressif-NVS-Analyzer) for this.
> 
> ```
> esptool.py -p (PORT) read_flash 0xc20000 0x6000 /tmp/nvs.bin
> analyze_nvs.py /tmp/nvs.bin > /tmp/nvs.txt
> ```
> 
> PS: Coincidentally I tried the same thing, but on H2 in the morning today and I was able to commission my H2. (esp-matter: main, esp-idf: master)

After deleting the build manageability components file, I reconfigured the compilation to avoid BLE errors

shubhamdp, 2024-08-15:
@MaplestoryAlen is the error fixed?

MaplestoryAlen, 2024-08-15:
> @MaplestoryAlen is the error fixed?

Yes, it has been fixed

closed 2024-08-15 by MaplestoryAlen