espressif/esp-matter #1052: Questions about OTA with ESP32

Eliostrim, 2024-08-13:
I'm able to perform OTA with ESP32 and i understand how it's work, there is no issue and everything works fine.

But, i have a little question about it because i can't find an information anywere.

I know for the OTA firmware there is at least two main partitions (OTA_0 and OTA_1).

I guess, when the OTA update proceed, the provider write into a cluster and the firmware on OTA_0 (assuming we are on OTA_0 at the begining) is writing into OTA_1.

 **So, I have two questions :** 
- The thing i guess before is it right ?
- And in which cluster is the firmware written?

(I guess the endpoint is root but i don't know about the cluster and attribute)

I have some logs but it's just the percentage of the loading of the new firmware 

And the cluster+attribute does not appear in the matter documentation.

Endpoint 0x0 is root
Cluster 0x2A does not appear into the matter doc for the root endpoint
Attribute 0x3 is the same as the cluster 

[image]

If you have any information and want to teach me about it, feel free to share ! 👍

wqx6, 2024-08-14:
There are two roles in Matter OTA. One is OTA Requestor which is the device to be updated and another is OTA Provider which is the device transferring the image to the requestor. The OTA requestor has an OTA Requestor cluster(cluster ID 0x2A) which helps managing the OTA, and OTA provider has an OTA provider cluster. The requestor cluster is always on Endpoint 0(root node). you can find more details about OTA in SPEC 11.21.7.

Eliostrim, 2024-08-14:
Thank you for your answer.

Yes, i did the OTA two terminal on my Raspi. One with Chip-tool and the other with the ota-provider-app builded with Build_example.py script

I had an old spec and that's why i didn't find the information

[image]

(Maybe that wasn't what you were talking about with the SPEC 11.21.7 👀)

Thank you again !

closed 2024-08-14 by dhrishi