espressif/esp-matter #1053: Light example takes a long time to set brightness when controlled by Home Assistant

dheckl, 2024-08-14:
**Describe the bug**
Light example tested on ESP32-C6, ESP32-H2 and ESP32-C3 takes a long time to set a new brightness when controlled by Home Assistant via Matter. When setting a new brightness, the brightness starts fading and takes up to 20 seconds to finish setting the desired brightness. Turning on / off works as expected. Also setting brightness with some other Client, e.g. Home App from Apple, works as expected. Please see this discussion https://github.com/home-assistant/core/issues/113775#issuecomment-2037800750. As mentioned [here](https://github.com/home-assistant/core/issues/113775#issuecomment-2139288918), Home Assistant has a default transition time and esp-matter seems to not handle it correctly.

There is a workaround available. Setting 0 for transition in the Home Assistant light profile config fixes the issue, but then no transition time is applied for changing e.g. brightness.

sayondeep, 2024-10-08:
@dheckl 
We carried out some tests and it was found that executing each iteration of [this](https://github.com/project-chip/connectedhomeip/blob/4c4970a48f79575487928e927e123df722712f37/src/app/clusters/level-control/level-control.cpp#L415) callback takes around 1 ms on the Linux platform and on MCU-based platform (like ESP32), it takes around 40ms - 70ms. This eventually causes the delay in level control response from the end-devices. Even after disabling all the console logs, it came down to around 6ms for each execution, which somewhat improved the response time.
Example -
To change level by say 40 in 200ms, this callback is scheduled every 5ms ([see here](https://github.com/project-chip/connectedhomeip/blob/4c4970a48f79575487928e927e123df722712f37/src/app/clusters/level-control/level-control.cpp#L999)) to reduce brightness by 1 ([see here](https://github.com/project-chip/connectedhomeip/blob/4c4970a48f79575487928e927e123df722712f37/src/app/clusters/level-control/level-control.cpp#L461)). When callback itself takes around 50ms for each execution, overall it takes quite some time for the device to respond.
It has been fixed on the Home Assistant side in this PR https://github.com/home-assistant/core/pull/126220, which changes the default transisiton time from 2 to 0.

dhrishi, 2024-10-16:
@dheckl Please give it a try with the latest Home Assistant code and close this if the issue is resolved

dheckl, 2024-10-16:
Thanks for your analysis. I can confirm that it works with transition time 0. However, very sad that we don't get it fixed with a defined transition time.

closed 2024-10-24 by dhrishi